platformio/platformio-core #3149: Home: Libraries: Could not update library

hectroleo, 2019-10-18:
PIO Core Call Error: "Library Storage: /home/kowalski/.platformio/lib\nUpdating SD                              @ 1.2.3          [1.2.4]\nUninstalling SD @ 1.2.3: \t[OK]\nLooking for SD library in registry\nSD\n==\n#ID: 161\nThis library aims to expose a subset of SD card functionality in the form of a higher level 'wrapper' object\n\nKeywords: sd, card, file, system\nCompatible frameworks: Arduino\nCompatible platforms: Atmel AVR\nAuthors: Adafruit Industries\n\nSD\n==\n#ID: 868\nEnables reading and writing on SD cards.\n\nKeywords: storage, data\nCompatible frameworks: Arduino\nCompatible platforms: Atmel AVR, Atmel SAM, Espressif 32, Espressif 8266, GigaDevice GD32V, Infineon XMC, Intel ARC32, Kendryte K210, Microchip PIC32, Nordic nRF51, Nordic nRF52, ST STM32, ST STM8, Teensy, TI MSP430\nAuthors: Arduino, SparkFun\n\nSD\n==\n#ID: 79\nThis library aims to expose a subset of SD card functionality in the form of a higher level 'wrapper' object (adaptation for LaunchPad LM4F120 / TM4C123)\n\nKeywords: sd, card, file, system\nCompatible frameworks: Energia\nCompatible platforms: TI MSP430, TI TIVA\nAuthors: Rei Vilo, William Greiman, SparkFun Electronics\n\nSD\n==\n#ID: 5210\nSD\n\nKeywords: sd\nCompatible frameworks: mbed\nCompatible platforms: Atmel SAM, Freescale Kinetis, Maxim Integrated MAX32, Nordic nRF51, Nordic nRF52, NXP LPC, Silicon Labs EFM32, ST STM32, Teensy, WIZNet W7500\nAuthors: Jackson Lv\n\nFound: https://platformio.org/lib/show/161/SD\nLibraryManager: Installing id=161 @ 1.2.4\n\n\nConflict: More than one library has been found by request {\"requirements\": \"1.2.4\", \"name\": \"SD\"}:\nAutomatically chose the first available library (use `--interactive` option to make a choice)\nError: Could not find a version that satisfies the requirement '1.2.4' for your system 'linux_x86_64'"

ivankravets, 2019-10-20:
Please open PlatformIO IDE Terminal and run:
```
pio lib -g uninstall SD
pio lib -g install 868
```

closed 2019-10-20 by ivankravets

hectroleo, 2019-10-20:
> [
> {
> "@type": "EmailMessage",
> "@type": "ViewAction",
> "target": "
> https://github.com/platformio/platformio-core/issues/3149?email_source=notifications\u0026email_token=AEZTXEIVXHESJMNOTICUQT3QPRJLLA5CNFSM4JCJUWQ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEBYJT5Y#issuecomment-544250359
> ",
> "url": "
> https://github.com/platformio/platformio-core/issues/3149?email_source=notifications\u0026email_token=AEZTXEIVXHESJMNOTICUQT3QPRJLLA5CNFSM4JCJUWQ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEBYJT5Y#issuecomment-544250359
> ",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "@type": "Organization",
> "name": "GitHub",
> "url": "https://github.com"
> }
> }
> ]

> Please open PlatformIO IDE Terminal and run:
> 
> pio lib -g uninstall SDpio lib -g install 868

WOOWWW, Thank's.
This issue was resolve..
Regards.