platformio/platformio-core #3158: Home: Could not load serial devices

MathZone, 2019-10-19:
PIO Core Call Error: "cannot import name 'threadpool' from 'twisted.python' (unknown location)"

ivankravets, 2019-10-20:
What is your OS?

MathZone, 2019-10-22:
Windows

ivankravets, 2019-10-23:
Could you try this solution?
- https://github.com/platformio/platformio-vscode-ide/issues/1259#issuecomment-545377055

ivankravets, 2019-10-23:
Please reopen if you still need help

MathZone, 2019-10-23:
%23 Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

%23 Configuration

VSCode: 1.39.2
PIO IDE: v1.9.0
System: Windows_NT, 10.0.17134, x64

%23 Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv 
 Error: Virtualenv Create: Traceback (most recent call last):
  File "C:\Users\MathZone\.platformio\.cache\tmp-13144pGiWAj3htHyO\virtualenv-16.7.5\virtualenv.py", line 2632, in <module>
    main()
  File "C:\Users\MathZone\.platformio\.cache\tmp-13144pGiWAj3htHyO\virtualenv-16.7.5\virtualenv.py", line 860, in main
    create_environment(
  File "C:\Users\MathZone\.platformio\.cache\tmp-13144pGiWAj3htHyO\virtualenv-16.7.5\virtualenv.py", line 1156, in create_environment
    install_python(home_dir, lib_dir, inc_dir, bin_dir, site_packages=site_packages, clear=clear, symlink=symlink)
  File "C:\Users\MathZone\.platformio\.cache\tmp-13144pGiWAj3htHyO\virtualenv-16.7.5\virtualenv.py", line 1526, in install_python
    shutil.copyfile(executable, py_executable)
  File "C:\Users\MathZone\AppData\Local\Programs\Python\Python38-32\lib\shutil.py", line 259, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\MathZone\\.platformio\\penv\\Scripts\\python.exe'

Using base prefix 'C:\\Users\\MathZone\\AppData\\Local\\Programs\\Python\\Python38-32'
New python executable in C:\Users\MathZone\.platformio\penv\Scripts\python.exe

	at t.value (c:\Users\MathZone\.vscode\extensions\platformio.platformio-ide-1.9.0\node_modules\platformio-node-helpers\dist\index.js:1:31671)
```

ivankravets, 2019-10-23:
See https://github.com/platformio/platformio-vscode-ide/issues/1213

closed 2019-10-23 by ivankravets