platformio/platformio-core #3183: Invalid encoding for project files for Visual Studio

antipin00, 2019-10-23:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 x64

**PlatformIO Version** (`platformio --version`): 4.0.3

### Description of problem
When generating project files for Visual Studio, PIO adds 6 bytes `C3 AF C2 BB C2 BF` at the beginning of `platformio.vcxproj` and `platformio.vcxproj.filters` files. These 6 bytes, if decoded as UTF-8, become characters `ï»¿`, which look exactly like UTF-8 Byte Order Mark [decoded using CP1252](https://en.wikipedia.org/wiki/Byte_order_mark#Byte_order_marks_by_encoding).
Trying to open the project file in Visual Studio 2019 as is gives the following error:
[image]
Removing these 6 bytes or replacing them with `EF BB BF`, a correct value for UTF-8 BOM, fixes it.

#### Steps to Reproduce

1. `mkdir test`
2. `cd test`
3. `platformio init --ide visualstudio -b uno`

### Actual Results
platformio.vcxproj (as UTF-8)
```
ï»¿<?xml version="1.0" encoding="utf-8"?>
<Project DefaultTargets="Build" ToolsVersion="12.0" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
  <PropertyGroup>
   ...
```

### Expected Results
platformio.vcxproj (as UTF-8)
```
<?xml version="1.0" encoding="utf-8"?>
<Project DefaultTargets="Build" ToolsVersion="12.0" xmlns="http://schemas.microsoft.com/developer/msbuild/2003">
  <PropertyGroup>
   ...
```

### Additional info
Python 3.7.3
Visual Studio Community 16.3.6

ivankravets, 2019-10-24:
@valeros could you reproduce it?

valeros, 2019-10-25:
Many thanks for reporting! Fixed in the development branch.

commit 087a8f6d (2019-10-25): Fix Visual Studio template files encoding // Resolve #3183
closed 2019-10-25 by valeros