platformio/platformio-core #3188: Option to make include path parsing more strict (Arduino IDE compatible)

TD-er, 2019-10-25:
What kind of issue is this?
- [x] **Feature Request**.
- [x] **PlatformIO Core**.

------------------------------------------------------------------

### Configuration

**Operating system**:  Windows

**PlatformIO Version** (`platformio --version`):  4.0.3

### Description of problem
I have a lot of includes in my code and to keep it compatible with ArduinoIDE I already have most of the .h files in `src/<somedir>/foo.h`
When I need something included from another file, I need to have it defined relative to the file from where I do the `#include`.  (Arduino IDE apparently demands this)
But I often mix up the relative paths and PlatformIO does build just fine with it.
Somehow the tab-completion in VS-code is also mixing up the paths, so even when using tab-completion the path is not relative to the file location.

I do have some users still working with Arduino IDE and they do report the include errors ([example](https://github.com/letscontrolit/ESPEasy/issues/2681#issuecomment-546306710)), but it would be great if I also had the same errors reported by the compiler.

Is there an option (and if not, then this is a feature request for it :) ) to force PlatformIO to check the paths in `#include` statements more strict like Arduino IDE does?

ivankravets, 2019-10-30:
Is this PlatformIO Core issue? Could you check `c_cpp_properties.json` file in `.vscode` folder? PlatformIO Core generate it here https://github.com/platformio/platformio-core/tree/develop/platformio/ide/tpls/vscode/.vscode

Does it contain valid values?

TD-er, 2019-10-30:
`compilerPath` you mean?

closed 2019-10-31 by ivankravets