platformio/platformio-core #3193: Failed to install Platformio IDE

nodiz, 2019-10-26:
Impossible to install again Platformio Ide on Visual studio code after upgrading OsX to Catilina

How can I overcome this?

Error: Conda Virtualenv: [Errno 30] Read-only file system: '/.cph_tmp5ntbcxll' at o.runCommand (/Users/nodiz/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:30256) at ChildProcess.h (/Users/nodiz/.vscode/extensions/platformio.platformio-ide-1.9.2/node_modules/platformio-node-helpers/dist/index.js:1:6292) at ChildProcess.emit (events.js:182:13) at maybeClose (internal/child_process.js:962:16) at Process.ChildProcess._handle.onexit (internal/child_process.js:251:5)

nodiz, 2019-10-26:
Conda info:

active environment : base
    active env location : /Users/nodiz/anaconda3
            shell level : 1
       user config file : /Users/nodiz/.condarc
 populated config files :
          conda version : 4.7.12
    conda-build version : not installed
         python version : 3.7.3.final.0
       virtual packages :
       base environment : /Users/nodiz/anaconda3  (writable)
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/nodiz/anaconda3/pkgs
                          /Users/nodiz/.conda/pkgs
       envs directories : /Users/nodiz/anaconda3/envs
                          /Users/nodiz/.conda/envs
               platform : osx-64
             user-agent : conda/4.7.12 requests/2.22.0 CPython/3.7.3 Darwin/19.0.0 OSX/10.15
                UID:GID : 501:20
             netrc file : None
           offline mode : False

(base) ➜  ~

ivankravets, 2019-10-28:
Yes, please remove conda from system and use official stable Python 3 from https://www.python.org/

closed 2019-10-28 by ivankravets