platformio/platformio-core #3213: platformio.ini lost semicolon separated comments support

dontsovcmc, 2019-10-31:
Visual studio code platformio extension lost semicolon separeted comments support.

### Configuration
Home 2.3.3 Core 4.0.3
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
**Operating system**: MacOS

### Description of problem

Bad content of `platformio.ini`:**
```[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev

upload_port = /dev/cu.wchusbserial1420 
upload_speed = 115200  ; раньше это компилировалось
```

Good:**
```[env:esp32dev]
platform = espressif32
framework = arduino
board = esp32dev

upload_port = /dev/cu.wchusbserial1420 
; теперь комментарии должны быть на отдельной строке
upload_speed = 115200  
```

ivankravets, 2019-10-31:
Thanks! Interesting bug. Python 3 has different behavior for RAW config parsing and does not handle inline comments. We will work on a fix for this.

ivankravets, 2019-10-31:
You can re-test with `pio upgrade --dev`.

ivankravets, 2019-10-31:
Resolved in https://github.com/platformio/platformio-core/commit/1d5d09feab2640d372e42744e12356605d84e697

closed 2019-10-31 by ivankravets