platformio/platformio-core #3219: Home: Could not load PIO Account information

takanotume24, 2019-10-31:
PIO Core Call Error: "the JSON object must be str, not 'bytes'"
```
Version: 1.39.2
Commit: 6ab598523be7a800d7f3eb4d92d7ab9a66069390
Date: 2019-10-15T15:33:40.634Z
Electron: 4.2.10
Chrome: 69.0.3497.128
Node.js: 10.11.0
V8: 6.9.427.31-electron.0
OS: Linux x64 4.4.0-166-generic
```
```
takuya@takuya-ThinkPad-X240:~$ uname -a
Linux takuya-ThinkPad-X240 4.4.0-166-generic #195-Ubuntu SMP Tue Oct 1 09:35:25 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

ivankravets, 2019-10-31:
Could you re-test with the latest PIO Core? `pio upgrade --dev`. Does it work now?

takanotume24, 2019-11-01:
Problems continue to occur. I can log in from CLI (pio account login), and I can confirm that I can display normally by "pio account show". However, the same error has still occurred from “PIO Account” of GUI on VSCode.

ivankravets, 2019-11-01:
What is your Python version?

takanotume24, 2019-11-06:
```
takuya@takuya-ThinkPad-X240 ~> python3
Python 3.5.2 (default, Oct  8 2019, 13:06:37) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()
takuya@takuya-ThinkPad-X240 ~> python
Python 2.7.12 (default, Oct  8 2019, 14:14:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()

```

ivankravets, 2019-11-06:
Could you try the next?
1. Open PlatformIO IDe Terminal (see icon on the bottom toolbar) and type `pio home`. It should open PIO Home in browser window
2. Try to do login. 
3. Do you see errors in terminal?

ivankravets, 2019-11-06:
Also, which Python does PIO Core use? Please run
```
~/.platformio/penv/bin/python --version
```

closed 2019-11-08 by ivankravets