platformio/platformio-core #3231: Home: Libraries: Could not update library

rmaltais, 2019-11-02:
PIO Core Call Error: "Library Storage: D:\\Users\\Réjean Maltais\\Documents\\Ender_3_mat\\Marlin-bugfix-2.0.x\\.pio\\libdeps\\megaatmega2560\n\n\nError: Traceback (most recent call last):\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 90, in main\n    cli(None, None, None)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 37, in invoke\n    return super(PlatformioCLI, self).invoke(ctx)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\click\\decorators.py\", line 17, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\lib.py\", line 253, in lib_update\n    lm.update(library, only_check=only_check)\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 773, in update\n    pkg_dir = self.get_package_dir(*self.parse_pkg_uri(package))\n  File \"c:\\.platformio\\penv\\lib\\site-packages\\platformio\\managers\\package.py\", line 237, in parse_pkg_uri\n    text = str(text)\nUnicodeEncodeError: 'ascii' codec can't encode character u'\\xe9' in position 10: ordinal not in range(128)\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2019-11-12:
Please reopen if you still need help

closed 2019-11-12 by ivankravets

gorrioncillo, 2020-12-10:
Sorry for opening this threat
I have the same problem when I search for `pio lib search arduinojson`
`UnicodeEncodeError: 'charmap' codec can't encode character '\u2714' in position 75: character maps to <undefined>`

I use platformio installed from recommend script and on `~/.platformio` as and venv on windows 10 SO.
python --version
`python 3.7.2`
pip --version
`pip 20.3.1`

ivankravets, 2020-12-22:
Could you provide any links to this library?

gorrioncillo, 2020-12-22:
> Could you provide any links to this library?

Of course
https://libraries.io/platformio/ArduinoJson

ivankravets, 2020-12-23:
Sorry, I can't reproduce it in the Windows virtual machine. It seems you have multiple Pythons in the system and Python 2 is used. I  recommend to UNINSTALL all Pythons, remove `~/.platformio/penv` folder and restart VSCode.

[image]

[image]

gorrioncillo, 2020-12-25:
I use only python 3.7 in windows 10 with platformio core on a terminal (the terminal taht comes with git for windows, mintty i thing)
Do not use vsCode
Using ``pio home`` do not show any error, as you show on that pictures