platformio/platformio-core #3239: [Feature] Select platform based on pull request

TD-er, 2019-11-04:
- [x] **Feature Request**.

### Description of problem

Every now and then I would like to test a new fix or feature in the platform libraries of my project.
For example, for ESP8266, I use this one as a platform:
```
platform = https://github.com/platformio/platform-espressif8266.git#feature/stage
```
But this does only fetch the `feature/stage` branch of that repo.

What I would like is to be able to select a specific pull request of that (esp8266/Arduino) repo to set as my platform.

Since this is more of a generic Git/Github issue, I hoped it would be a PlatformIO-core issue, but if it does not belong here, please let me know where it is more appropriate.

ivankravets, 2019-11-06:
You can do this with http://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages

```
platform_packages = 
    framework-arduinoespressif8266 @ https://github.com/d-a-v/Arduino/tree/weakfsend
```

In this case, I used this PR https://github.com/esp8266/Arduino/pull/6699

ivankravets, 2019-11-06:
P.S: You don't need to use `platform = https://github.com/platformio/platform-espressif8266.git#feature/stage` with PIO Core 4. It's legacy solution. Please use `platform_packages` directly.

closed 2019-11-06 by ivankravets

TD-er, 2019-11-06:
Oh that's great.
Will test it later when I'm back home.

Jason2866, 2019-11-07:
@ivankravets  Mhh, i have this in platformio.ini 
```
[core_cstage]
; *** Arduino Esp8266 -> will be released as core 2.6.0 
platform                  =  espressif8266
platform_packages         =  
    framework-arduinoespressif8266 @ https://github.com/Jason2866/Arduino/tree/core_260
```
I get installed:
```
> Executing task: C:\Users\Hans\.platformio\penv\Scripts\platformio.exe run --target clean <

Processing tasmota (platform: espressif8266; framework: arduino; board: esp01_1m)
-------------------------------------------------------------------------------------------------------------- 
PlatformManager: Installing espressif8266
espressif8266 @ 2.2.3 has been successfully installed!
The platform 'espressif8266' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-arduinoespressif8266 @ ~2.20502.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
framework-arduinoespressif8266 @ 2.20502.0 has been successfully installed!
Done cleaning
```
I have just the files from framework-arduinoespressif8266 @ 2.20502.0 
There is no file from https://github.com/Jason2866/Arduino/tree/core_260

ivankravets, 2019-11-12:
@Jason2866 it should be
```
[core_cstage]
platform                  =  espressif8266
platform_packages         =  
    framework-arduinoespressif8266 @ https://github.com/Jason2866/Arduino.git#core_2_6_0
```

Jason2866, 2019-11-12:
@ivankravets  still doesnt work. i have now 
```
[core_cstage]
platform                  =  espressif8266
platform_packages         =  
    framework-arduinoespressif8266 @ https://github.com/Jason2866/Arduino.git#core_2_6_0                     
build_flags               = ${esp82xx_defaults.build_flags}
                            -Wl,-Tesp8266.flash.1m.ld
```
and it installs
```
Processing tasmota (platform: espressif8266; framework: arduino; board: esp01_1m)
--------------------------------------------------------------------------------------------------------------- 
PlatformManager: Installing espressif8266
espressif8266 @ 2.2.3 has been successfully installed!
The platform 'espressif8266' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing framework-arduinoespressif8266 @ ~2.20502.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
framework-arduinoespressif8266 @ 2.20502.0 has been successfully installed!
Done cleaning
```

ivankravets, 2019-11-12:
@Jason2866 Yes, it installs directly from your custom branch. Try to build a project and you will see 
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/nodemcuv2.html
PLATFORM: Espressif 8266 2.2.3 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), framework-arduinoespressif8266 c245d88, tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
```

See `c245d88` => https://github.com/Jason2866/Arduino/tree/core_2_6_0

P.S: Please note that your example about is bad. `platform_packages` should be included in `[env]` section. See https://github.com/platformio/platformio-core/issues/3239#issuecomment-553003544

Jason2866, 2019-11-12:
Thx! found my error your P.S. was the hint!
 forgot to define in `[env]` something like
`platform_packages         = ${core_stage.platform_packages}`

ivankravets, 2019-11-12:
Happy coding ;)

TD-er, 2019-11-16:
@Jason2866 Can you give me the relevant part of your ini file where we can build using core 2.6.1?
When using this 'new' syntax of providing the `platform` and `platform_packages`, I get loads of warnings regarding redefines and build which cannot finish.

I would really like to start building with `platform_packages` set to a PR (branch), but I cannot get it to work with the right tool chain.

Jason2866, 2019-11-16:
@TD-er  You have to fork https://github.com/platformio/platform-espressif8266/tree/feature/stage
Make a changed platform.json.  Example `https://github.com/Jason2866/platform-espressif8266/blob/core_2_6_1/platform.json` . Change number of needed build chain and esptool
add a new entry in `packageRepositories` (mine `"https://github.com/Jason2866/platform-espressif8266/releases/download/2.20601.0/manifest.json",`) in this manifest.json add your new package. See https://github.com/Jason2866/platform-espressif8266/releases
This packages are repacked ones of the original Arduino releases with **added** the needed `package.json`

The new syntax providing `platform` and `platform_packages` DOES NOT work because it still uses the old build chain!!

TD-er, 2019-11-25:
@Jason2866 I'm looking into the instructions you provided, but it is quite a lot of work to do any single time I want to test something in the Arduino core libs, like a pull request.

Isn't there a more simple way?

If I follow the instructions mentioned here: https://github.com/platformio/platformio-core/issues/3239#issuecomment-550261931
I get an error like this:

```
Processing spec_wifimesh_ESP8266_4M1M (platform: espressif8266; board: esp12e; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PackageManager: Installing toolchain-xtensa @ ~1.40802.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
toolchain-xtensa @ 1.40802.0 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266
git version 2.21.0.windows.1
Cloning into 'C:\Users\gijs\.platformio\packages\_tmp_installing-txchszk2-package'...
fatal: repository 'https://github.com/aerlon/Arduino/tree/wifi_mesh_update_2.2/' not found
Error: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/aerlon/Arduino/tree/wifi_mesh_update_2.2', 'C:\\Users\\gijs\\.platformio\\packages\\_tmp_installing-txchszk2-package']
The terminal process terminated with exit code: 1
```

With them defined in my platformio.ini like this:
```
[core_2_6_1_wifimesh]
platform                  = espressif8266
platform_packages         = 
    framework-arduinoespressif8266 @ https://github.com/aerlon/Arduino/tree/wifi_mesh_update_2.2
build_flags               = ${esp82xx_2_6_x.build_flags} 
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703
```

ivankravets, 2019-11-25:
> https://github.com/aerlon/Arduino/tree/wifi_mesh_update_2.2/

This is not a git URL. This is the Github Web URL. See my comment here https://github.com/platformio/platformio-core/issues/3239#issuecomment-553003544

TD-er, 2019-11-25:
@ivankravets You're right.
I do need extra coffee I guess.
Now using `framework-arduinoespressif8266 @ https://github.com/aerlon/Arduino.git#wifi_mesh_update_2.2` and it does seem to be cloning.

TD-er, 2019-11-25:
Problem remains with this one (as I reported in another issue yesterday)
Submodule 'lwip2-src' (https://git.savannah.nongnu.org/git/lwip.git) registered for path 'tools/sdk/lwip2/builder/lwip2-src'
Cloning into 'C:/Users/gijs/.platformio/packages/_tmp_installing-gdd2zpwp-package/tools/sdk/lwip2/builder/lwip2-src'...

Any idea how to overcome this problem?

TD-er, 2019-11-25:
Hmm, I keep having issues with this platform_packages configuration.
Tried several of these but they all result in the same, a build which is a lot smaller and also doesn't have my code in it.
It is just like there is no call to setup() or loop() in the compile run.

```
[core_2_6_1_wifimesh]
platform                  = espressif8266
;platform_packages         = 
;    framework-arduinoespressif8266 @ https://github.com/TD-er/Arduino.git#wifi_mesh_update_2.2
platform_packages         =  
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.1
build_flags               = ${esp82xx_2_6_x.build_flags} 
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703

[env:spec_wifimesh_ESP8266_4M1M]
extends                   = esp8266_4M1M
framework                 = arduino
build_unflags             =
platform                  = ${core_2_6_1_wifimesh.platform}
platform_packages         = ${core_2_6_1_wifimesh.platform_packages}
build_flags               = ${core_2_6_1_wifimesh.build_flags} 
                            ${esp8266_4M1M.build_flags} 
                            -DPLUGIN_BUILD_CUSTOM
lib_ignore                = ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS
extra_scripts             = pre:pre_wifimesh_esp82xx.py
```

The rest is the same as in my other env. configurations.
Even the Python script is the same as my 'custom' builds, just with a different name.

I really have no clue how to make this work here.

TD-er, 2019-11-26:
This is definitely a bug, so I files a bug report: #3288