platformio/platformio-core #3249: PlatfortmIO not compile anything

polachz, 2019-11-05:
Unable to compile any project by pio. 

Always get Error 1 on Windows. 
I have tried:

- Remove my other python dev environments, 
- Deleted PIO folder content,
- Reinstalled python by PlatformIO setup (by VS Code),  

still same result: **Error 1.**

Even if try to compile empty sketch, it failed. 

------------------------------------------------------------------

All platform builds seems to be broken, but checked on

Platform: btatmega328  Framework: Arduino
Platform: esp01 Framework: Arduino

Same result, unable to compile main.cpp with Error 1:

------------------------------------------------------------------

### Configuration

**Operating system**:
WIndows 10 version 1903

**PlatformIO Version** (`platformio --version`):
PlatformIO, version 4.0.3

### Description of problem

#### Steps to Reproduce

1. pio run -v -j 1

### Actual Results

```
S:\POkusnak>pio run -v -j 1
Processing btatmega328 (platform: atmelavr; board: btatmega328; framework: arduino)
------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelavr/btatmega328.html
PLATFORM: Atmel AVR 1.15.0 > Arduino BT ATmega328
HARDWARE: ATMEGA328P 16MHz, 2KB RAM, 28KB Flash
PACKAGES: toolchain-atmelavr 1.50400.190710 (5.4.0), framework-arduinoavr 4.1.2
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 5 compatible libraries
Scanning dependencies...
No dependencies
avr-g++ -o .pio\build\btatmega328\src\main.cpp.o -c -fno-exceptions -fno-threadsafe-statics -fpermissive -std=gnu++11 -Os -Wall -ffunction-sections -fdata-sections -flto -mmcu=atmega328p -DPLATFORMIO=40003 -DARDUINO_AVR_BT -DF_CPU=16000000L -DARDUINO_ARCH_AVR -DARDUINO=10805 -Iinclude -Isrc -IC:\Users\polach\.platformio\packages\framework-arduinoavr\cores\arduino -IC:\Users\polach\.platformio\packages\framework-arduinoavr\variants\eightanaloginputs src\main.cpp
*** [.pio\build\btatmega328\src\main.cpp.o] Error 1
============================================== [FAILED] Took 0.92 seconds ==============================================
```
-------------------------
```
S:\ESP_Temp_Sensor>pio run -v -j 1
Processing esp01 (platform: espressif8266; board: esp01_1m; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 2.2.3 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), framework-arduinoespressif8266 2.20502.0 (2.5.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
No dependencies
xtensa-lx106-elf-g++ -o .pio\build\esp01\src\main.cpp.o -c -fno-rtti -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=40003 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_QIO -DLWIP_OPEN_SRC -DNONOSDK221=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -Iinclude -Isrc -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\variants\generic src\main.cpp
*** [.pio\build\esp01\src\main.cpp.o] Error 1
============================================================ [FAILED] Took 1.11 seconds ============================================================
```

### Expected Results

Correct sketch build

### If problems with PlatformIO Build System:

[env:btatmega328]
platform = atmelavr
board = btatmega328
framework = arduino

--------------------------------------------------

[env:esp01]
platform = espressif8266
board = esp01_1m
framework = arduino

-----------------------------------------------------------

**Source file to reproduce issue:**

#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
}

ivankravets, 2019-11-06:
Do you have antivirus tools in system? It seems that some tool blocks compiler.

polachz, 2019-11-06:
No, AV is completely disabled and no detection has been provided

Additionally the command pio run -v -j 1

creates 

c:\ESP_Temp_Sensor\.pio\build\esp01\src\main.cpp.o 

Then compiler is running

If I remove job count limit to 1, (run without -j) i can see more output files:

.pio\build\esp01\libFrameworkArduinoVariant.a (size is 8 bytes)
.pio\build\esp01\FrameworkArduino\Esp-frag.cpp.o  (size 2608)

And output is:

```
ESP_Temp_Sensor>pio run -v
Processing esp01 (platform: espressif8266; board: esp01_1m; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp01_1m.html
PLATFORM: Espressif 8266 2.2.3 > Espressif Generic ESP8266 ESP-01 1M
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), framework-arduinoespressif8266 2.20502.0 (2.5.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 28 compatible libraries
Scanning dependencies...
No dependencies
xtensa-lx106-elf-g++ -o .pio\build\esp01\src\main.cpp.o -c -fno-rtti -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=40003 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_QIO -DLWIP_OPEN_SRC -DNONOSDK221=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -Iinclude -Isrc -IC:\Users\polach\.platformio\packages\framework-arduinoespr*** [.pio\build\esp01\libFrameworkArduinoVariant.a] Error 1
essif8266\tools\sdk\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Use*** [.pio\build\esp01\ld\local.eagle.app.v6.common.ld] Error 1
rs\polach\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\variants\generic src\main.cpp
xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH C:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\ld\eagle.app.v6.common.ld.h -o .pio\build\esp01\ld\local.eagle.app.v6.common.ld
xtensa-lx106-elf-ar rc .pio\build\esp01\libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pio\build\esp01\FrameworkArduino\Esp-frag.cpp.o -c -fno-rtti -std=c++11 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPLATFORMIO=40003 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP01 -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP01_1M\" -DFLASHMODE_QIO -DLWIP_OPEN_SRC -DNONOSDK221=1 -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DVTABLES_IN_FLASH -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\polach\.platformio\packages\framework-arduinoespressif8266\variants\generic C:\Users\polach\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Esp-frag.cpp
*** [.pio\build\esp01\src\main.cpp.o] Error 1
*** [.pio\build\esp01\FrameworkArduino\Esp-frag.cpp.o] Error 1
============================================================== [FAILED] Took 1.19 seconds ==============================================================

```

Then compiler is creating files....

polachz, 2019-11-06:
Just update.. I have installed PIO on my laptop. SAme AV solution, config very near to main computer. Here everything works. And files generated her looks same. By my opinion, first step of build works OK, and then something failed... What PIO does after successfull source code compilation? Is possible to enable an debug logging for build process. To see what failed ?

ivankravets, 2019-11-07:
> Is possible to enable an debug logging for build process. To see what failed ?

Could you try `pio run -v`?

polachz, 2019-11-07:
It's always run with -v option. please take look deeply onto my posts.

polachz, 2019-11-10:
I have had reinstalled computer to make things ready. Now not able to analyse anything, computer state has been lost.

closed 2019-11-10 by polachz