platformio/platformio-core #3258: Sugest for Clion template

Mikolaj98p, 2019-11-08:
Hello,
I have a suggest for Clion IDE, can you change default template for `CMakeListsPrivate.txt` file from `set(PLATFORMIO_CMD "D://programs//devel//Python//Python37-32//Scripts//platformio.exe")` to `set(PLATFORMIO_CMD "platformio.exe")` or universal `set(PLATFORMIO_CMD "platformio")`, because after upgrade python and move project to another PC/OS Clion displays error:
```
System nie moľe odnale«† okre�lonej �cieľki.
CMakeFiles\PLATFORMIO_BUILD.dir\build.make:56: recipe for target 'CMakeFiles/PLATFORMIO_BUILD' failed
mingw32-make.exe[3]: *** [CMakeFiles/PLATFORMIO_BUILD] Error 1
mingw32-make.exe[2]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/all] Error 2
CMakeFiles\Makefile2:108: recipe for target 'CMakeFiles/PLATFORMIO_BUILD.dir/all' failed
CMakeFiles\Makefile2:120: recipe for target 'CMakeFiles/PLATFORMIO_BUILD.dir/rule' failed
Makefile:130: recipe for target 'PLATFORMIO_BUILD' failed
mingw32-make.exe[1]: *** [CMakeFiles/PLATFORMIO_BUILD.dir/rule] Error 2
mingw32-make.exe: *** [PLATFORMIO_BUILD] Error 2
```
Thanks.

ivankravets, 2019-11-11:
We can't because it will not work if user generates project from virtual environemnt. In this case, `platformio` is not visible in global PATH.

ivankravets, 2019-11-11:
This is a `CMakeListsPrivate.txt` file, you should share it to VCS. Every user should open your project in CLion and re-run "rebuild project index" or re-init project. 

This is a temporary workaround while we don't have an official PlatformIO plugin for CLion. See the progress here https://github.com/platformio/platformio-core/issues/2201#issuecomment-550506274

closed 2019-11-11 by ivankravets