platformio/platformio-core #3259: Home: Could not initialize project

Fret23, 2019-11-08:
PIO Core Call Error: "The current working directory /Users/steve/Documents/PlatformIO/Projects/Nano Play will be used for the project.\n\nThe next files/directories have been created in /Users/steve/Documents/PlatformIO/Projects/Nano Play\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Traceback (most recent call last):\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py\", line 90, in main\n    click.echo = lambda *args, **kwargs: _safe_echo(0, *args, **kwargs)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 764, in __call__\n    return self.main(*args, **kwargs)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 717, in main\n    rv = self.invoke(ctx)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py\", line 37, in invoke\n    ]\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 1137, in invoke\n    return _process_result(sub_ctx.command.invoke(sub_ctx))\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 956, in invoke\n    return ctx.invoke(self.callback, **ctx.params)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/click/core.py\", line 555, in invoke\n    return callback(*args, **kwargs)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/click/decorators.py\", line 17, in new_func\n    return f(get_current_context(), *args, **kwargs)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py\", line 92, in cli\n    init_base_project(project_dir)\n  File \"/Users/steve/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py\", line 131, in init_base_project\n    config = ProjectConfig()\nTypeError: __init__() takes at least 2 arguments (1 given)\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n  `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n  https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n  https://github.com/platformio/platformio-core/issues\n\n============================================================"

ivankravets, 2019-11-11:
What is your PlatformIO Core version? `pio --version`. Could you reproduce this issue again? How to reproduce it?

ivankravets, 2019-11-11:
Duplicate of https://github.com/platformio/platformio-home/issues/49

closed 2019-11-11 by ivankravets