platformio/platformio-core #3264: env.BoardConfig() does not work for custom boards in extra scripts of libraries.

JelleRoets, 2019-11-11:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: OSX 10.14.6 

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.1.0

### Description of problem
I have a library with an extraScript. This script wants to read the current BoardConfig using `env.BoardConfig()`. However when I use a custom board, which' definition lives in the 'boards' folder of my project root folder, env.BoardConfig() in my library script does not work anymore and throws an error: `UnknownBoard: Unknown board ID 'myboard'`.

#### Steps to Reproduce

1. Create a custom board
2. Create a library with an extraScript, in that script use env.BoardConfig()
3. try to run platformio

### Actual Results
throws error:  `UnknownBoard: Unknown board ID 'myboard'`.

### Expected Results
returns the correct current board config

### Additional info
I think the culprit is https://github.com/platformio/platformio-core/blob/3c796ca7c87212b437cd6cbf30bed1821d72f97d/platformio/project/config.py#L376 when the script for the custom library is executed, the cwd changes to the folder of the library, hence the `get_optional_dir("boards")` (from https://github.com/platformio/platformio-core/blob/develop/platformio/managers/platform.py#L619) will return a path that contains the path of the library instead of the project path, hence it can't find the custom boards folder and throws an error it can't find the custom board.

ivankravets, 2019-11-12:
Please re-test with `pio upgrade --dev`.

JelleRoets, 2019-11-12:
unfortunately it didn't solve the problem, I still get the same error. 
In the mean time I pushed my example, so in order to reproduce
1) check out https://github.com/JelleRoets/AtmelStart_PlatformIO
2) running `pio run` in the ArduinoZeroASF example (https://github.com/JelleRoets/AtmelStart_PlatformIO/tree/master/examples/ArduinoZeroASF) results in the following error:

```
UnknownBoard: Unknown board ID 'myboard':
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/jelle/.platformio/platforms/atmelsam/builder/main.py", line 120:
    target_elf = env.BuildProgram()
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/platformio.py", line 60:
    env.ProcessProjectDeps()
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/platformio.py", line 129:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 1037:
    project.install_dependencies()
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 853:
    if _is_builtin(uri):
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 845:
    for lb in self.env.GetLibBuilders():
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 964:
    lb = LibBuilderFactory.new(env, lib_dir)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 57:
    obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 132:
    self.process_extra_options()
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py", line 271:
    exports={"env": self.env, "pio_lib_builder": self},
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/jelle/testPio/.pio/libdeps/testPio/AtmelStart_PlatformIO/library.py", line 188:
    linker_script = adjust_linker_offset(env.subst('$PIOENV'), linker_script)
  File "/Users/jelle/testPio/.pio/libdeps/testPio/AtmelStart_PlatformIO/library.py", line 166:
    offset_address = env.BoardConfig().get("upload.offset_address", "0")
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/pioplatform.py", line 43:
    p = env.PioPlatform()
  File "/Users/jelle/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/util.py", line 59:
    self.cache[key] = (time.time(), func(*args, **kwargs))
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/pioplatform.py", line 38:
    p.configure_default_packages(variables, COMMAND_LINE_TARGETS)
  File "/Users/jelle/.platformio/platforms/atmelsam/platform.py", line 26:
    board = self.board_config(variables.get("board"))
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/managers/platform.py", line 647:
    return self.get_boards(id_)
  File "/Users/jelle/.platformio/platforms/atmelsam/platform.py", line 61:
    result = PlatformBase.get_boards(self, id_)
  File "/Users/jelle/.platformio/penv/lib/python3.6/site-packages/platformio/managers/platform.py", line 643:
    raise exception.UnknownBoard(id_)
```

ivankravets, 2019-11-12:
Please re-run `pio upgrade --dev`

Thanks for the report!

closed 2019-11-12 by ivankravets

JelleRoets, 2019-11-12:
yep that fixed it, thanks for the quick respons!

commit c6a8e033 (2019-11-12): Fixed an issue when ``env.BoardConfig()`` does not work for custom boards in extra scripts of libraries // Resolve #3264
commit a9f9f4ef (2019-11-12): Fixed an issue when ``env.BoardConfig()`` does not work for custom boards in extra scripts of libraries // Resolve #3264