platformio/platformio-core #3267: PROJECTPACKAGES_DIR empty

MarkJeronimus, 2019-11-12:
The platformio.ini variable PROJECTPACKAGES_DIR doesn't resolve anymore sinve last night's update.

According to the manual, it should resolve to “core_dir/packages”. In my experience this is "~/.platformio/packages". Previously it resolved to this, but now it resolves to "".

This prevents me from using custom avrdude parameters.

Excerpt from my platformio.ini:

```prefs
upload_flags =
    -C$PROJECTPACKAGES_DIR/tool-avrdude/avrdude.conf
    -P$UPLOAD_PORT
    -c$UPLOAD_PROTOCOL
    -b$UPLOAD_SPEED
    -patmega328p
    -V
upload_command = avrdude $UPLOAD_FLAGS -U flash:w:$SOURCE:i
```

Error I'm getting:

> avrdude: can't open config file "/tool-avrdude/avrdude.conf": No such file or directory
> avrdude: error reading system wide configuration file "/tool-avrdude/avrdude.conf"

ivankravets, 2019-11-12:
It should be `PROJECT_PACKAGES_DIR`. See https://github.com/platformio/platformio-core/blob/develop/platformio/builder/main.py#L104

closed 2019-11-12 by ivankravets

MarkJeronimus, 2019-11-12:
Thanks, it works.

Changes like these should be put in the release notes and the manual updated in tandem with the update rollout.

ivankravets, 2019-11-12:
It was changed in PIO Core 4.0. We totally refactored project configuration. We don't plan to change these variables in future. Sorry for the issue.

MarkJeronimus, 2019-11-12:
If the update is that big, it should be able to migrate project configuration to the new format (after user confirmation)

ivankravets, 2019-11-12:
We have it, maybe, not so much clear http://docs.platformio.org/en/latest/migration.html

MarkJeronimus, 2019-11-12:
- I can't find PROJECTPACKAGES_DIR or even "PROJECT_PACKAGES_DIR" on that page
- When I click on the platformio.ini item it takes me to the old ('latest') manual with PROJECTPACKAGES_DIR
- When I go to the release notes, I already read that and it doesn't seem to contain anything relevant to this

MarkJeronimus, 2019-11-12:
Sorry if I sound angry, I'm not. It's just that with great projects (and this is one heck of a great project) comes great responsibility.

ivankravets, 2019-11-12:
I've just fixed docs https://github.com/platformio/platformio-docs/commit/a6c6b4a80591388e3b76c04bec7920aed13ed851

> Sorry if I sound angry

Everything is OK, this open-source project. We would be thankful if you help us to improve current documentation https://github.com/platformio/platformio-docs