platformio/platformio-core #3271: Could not import Arduino Due project when importing Arduino libraries

fredlcore, 2019-11-12:
I have a project written in the Arduino IDE for the Arduino Due. When I try to import this project within PlatformIO in VS Code **without** checking the "Use libraries installed by Arduino IDE", then the import goes through without and error.
However, if I check this box, I get a strange mix of status messages and error codes after the import process is running for a few seconds (see below). Versions are Home 3.0.0 and Core 4.1.0

`PIO Core Call Error: "The current working directory /Users/frederik/Documents/PlatformIO/Projects/191113-003700-due will be used for the project.\n\nThe next files/directories have been created in /Users/frederik/Documents/PlatformIO/Projects/191113-003700-due\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing due (platform: atmelsam; board: due; framework: arduino)\n--------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\nCONFIGURATION: https://docs.platformio.org/page/boards/atmelsam/due.html\nPLATFORM: Atmel SAM 3.9.0 > Arduino Due (Programming Port)\nHARDWARE: AT91SAM3X8E 84MHz, 96KB RAM, 512KB Flash\nDEBUG: Current (atmel-ice) External (atmel-ice, blackmagic, jlink, stlink)\nPACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinosam 4.4.191002\nLDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf\nLDF Modes: Finder ~ chain, Compatibility ~ soft\nManifestParserError: Could not parse manifest -> Expecting ',' delimiter: line 12 column 9 (char 600):\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/main.py\", line 151:\n env.SConscript(\"$BUILD_SCRIPT\")\n File \"/Users/frederik/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 605:\n return _SConscript(self.fs, *files, **subst_kw)\n File \"/Users/frederik/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py\", line 286:\n exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)\n File \"/Users/frederik/.platformio/platforms/atmelsam/builder/main.py\", line 120:\n target_elf = env.BuildProgram()\n File \"/Users/frederik/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\n return self.method(*nargs, **kwargs)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/platformio.py\", line 140:\n _build_project_deps(env)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/platformio.py\", line 48:\n project_lib_builder = env.ConfigureProjectLibBuilder()\n File \"/Users/frederik/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\n return self.method(*nargs, **kwargs)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py\", line 1029:\n lib_builders = env.GetLibBuilders()\n File \"/Users/frederik/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py\", line 224:\n return self.method(*nargs, **kwargs)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py\", line 954:\n lb = LibBuilderFactory.new(env, lib_dir)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py\", line 54:\n obj = getattr(sys.modules[__name__], clsname)(env, path, verbose=verbose)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py\", line 111:\n self._manifest = manifest if manifest else self.load_manifest()\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/builder/tools/piolib.py\", line 662:\n return ManifestParserFactory.new_from_file(manifest_path).as_dict()\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/package/manifest/parser.py\", line 67:\n return ManifestParserFactory.new(get_file_contents(path), t, remote_url)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/package/manifest/parser.py\", line 117:\n return globals()[clsname](contents, remote_url, package_dir)\n File \"/Users/frederik/.platformio/penv/lib/python3.6/site-packages/platformio/package/manifest/parser.py\", line 127:\n raise ManifestParserError(\"Could not parse manifest -> %s\" % e)\n========================== [FAILED] Took 3.09 seconds =========================="`

ivankravets, 2019-11-16:
You have broken manifest in libraries. We added workaround for this issue. Please switch PlatformIO IDE to the development version of PlatformIO Core and re-test. See http://docs.platformio.org/en/latest/ide/vscode.html#platformio-ide-usedevelopmentpiocore

closed 2019-11-16 by ivankravets

fredlcore, 2019-12-12:
Thank you, with the development version it now works like charm!