platformio/platformio-core #3276: Request for more debugging options for Arduino family

MarkyBoy7, 2019-11-17:
Although PlatformIO boasts remote debugging with both Atom and  Visual Studio Code , the options available for Arduino seem few till now. Any news / plans in the pipeline?

ivankravets, 2019-11-19:
Which boards do you mean?

MarkyBoy7, 2019-11-20:
I tend to use Arduino Nanos at the moment but have used Uno and Mega too. I think for debugging it's not so important which ones are supported, at least to start with.
Another route could be to use a simulator to avoid the need for extra kit for the debugging link. 
All debugging which is non-hardware specific should then work.
Do you support a particular simulator?
If so, are there instructions for setting it up.

ivankravets, 2019-11-20:
Duplicate of https://github.com/platformio/platform-atmelavr/issues/95

closed 2019-11-20 by ivankravets