platformio/platformio-core #3279: No error when referred parameter is missing (using extends)

TD-er, 2019-11-19:
PIO version:
- Home 3.1.0-beta.1
- Core 4.1.1b1

I tried to get my platformio.ini file a lot more structured by using `extends`.

Just a section of my ini file:

```ini
[common]
lib_ldf_mode              = chain
lib_archive               = false
framework                 = arduino
upload_speed              = 115200
monitor_speed             = 115200
;targets                   = size, checkprogsize
targets                   =

[regular_platform]
build_unflags             =
build_flags               = ${core_2_6_1.build_flags} ${common.build_flags}
platform                  = ${core_2_6_1.platform}

[hard_esp82xx]
platform                  = ${regular_platform.platform}
build_flags               = ${regular_platform.build_flags}
                            -DBUILD_NO_DEBUG
                            -DPLUGIN_BUILD_CUSTOM
```

As you can see, the `${regular_platform.build_flags}` refers `${common.build_flags}`, which doesn't exist.
The result is no error, but it has the `build_flags` empty.

Part of the verbose output:
```
Processing hard_SONOFF_POW_4M1M (extends: esp8266_4M1M, hard_esp82xx; platform: https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1; build_flags: ; board: esp12e; board_build.flash_mode: dio; board_upload.maximum_size: 1044464; board_build.f_cpu: 80000000L; build_unflags: -DDEBUG_ESP_PORT; lib_deps: https://github.com/TD-er/ESPEasySerial.git; lib_ignore: ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS; lib_ldf_mode: chain; lib_archive: False; framework: arduino; upload_speed: 115200; monitor_speed: 115200; targets: )
```

TD-er, 2019-12-03:
Another thing what has taken me quite a while to find is that for example the `build_flags` will be empty when some build flags are referred twice.

For example:
```
[regular_platform]
build_flags               = ${core_2_6_1.build_flags} ${common.build_flags}

[esp82xx_common]
extends                   = common

[esp8266_4M1M]
extends                   = esp82xx_common

[env:normal_ESP8266_4M1M]
extends                   = esp8266_4M1M
platform                  = ${regular_platform.platform}
```

As can be seen, my mistake of including ` ${common.build_flags}` twice results in empty `build_flags`

ivankravets, 2020-02-08:
Thanks! Please re-test with `pio upgrade --dev`.

closed 2020-02-08 by ivankravets

TD-er, 2020-02-27:
I forgot to reply (and test), but now I have tested it (by accident ;) ) and it works :)

Thanks.

commit 2763853d (2020-02-08): Fixed an issue when no error is raised if referred parameter (interpolation) is missing in a project configuration file // Resolve #3279