platformio/platformio-core #3280: [Feature] Build flags per directory

TD-er, 2019-11-20:
- [x] **Feature Request**.

Not sure if it also touches this issue: https://github.com/platformio/platformio-core/issues/2144

What I would like to have, is the option to define some build flags for my own source code and some flags for the platform code and maybe for the libraries.

For example, I would like to have extra warnings on my own code, but when I add these I get flooded with warnings on the code which is a bit out of my control.

Maybe we can have 3 levels of build flags:
- platform/framework related code (e.g. framework-arduinoespressif8266) Maybe the `core_dir` ?
- Libraries (quite well defined concept in PIO)
- Project code (everything in the `src_dir`)

ivankravets, 2019-11-20:
You can do this with [Build Middlewares](http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#build-middlewares).

We don't plan to make more complicated `platformio.ini`. So, just use extra script and out API.

closed 2019-11-20 by ivankravets

positron96, 2021-12-27:
By the way, there are `build_flags` and `src_build_flags`. I use it for `-Wall -Werror` for my code, but not for framework code.