platformio/platformio-core #3281: 4.1.0 can no longer use local library for the native platform on the arduino framework

nomis, 2019-11-20:
### Description of problem
The library in `lib/` is not found for the native platform. This works on 4.0.3 and it still works for other platforms.

It complains that the library is a "Framework incompatible library" but the library is compatible with the "arduino" framework.

#### Steps to Reproduce

1. git clone https://github.com/nomis/mcu-uuid-common
2. cd mcu-uuid-common/test
3. platformio run -d build/example-Uptime -e native -v

(at commit 1d9997c1b5d7a4b22b66e6ed06517a336d1cc9be)

### Actual Results

Build cannot find the library and fails:
```
Processing native (platform: native; build_flags: -std=c++11 -Os -Wall -Wextra -I../native; src_build_flags: ${build_flags} -flto -Werror -I../native -DENV_NATIVE; lib_ldf_mode: off; lib_deps: uuid-common)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PACKAGES: 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Framework incompatible library /home/simon/src/mcu-uuid-common/test/build/example-Uptime/lib/uuid-common
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
g++ -o .pio/build/native/src/Uptime.o -c -std=c++11 -Os -Wall -Wextra -flto -Werror -DPLATFORMIO=40100 -DENV_NATIVE -Isrc -I/home/simon/src/mcu-uuid-common/test/build/native -I/home/simon/src/mcu-uuid-common/test/build/native src/Uptime.cpp
src/Uptime.cpp:2:10: fatal error: uuid/common.h: No such file or directory
 #include <uuid/common.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/native/src/Uptime.o] Error 1
```

### Expected Results

Build finds the library and is successful:
```
Processing native (platform: native; build_flags: -std=c++11 -Os -Wall -Wextra -I../native; src_build_flags: ${build_flags} -flto -Werror -I../native -DENV_NATIVE; lib_ldf_mode: off; lib_deps: uuid-common)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PACKAGES: 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 1 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <uuid-common> 1.1.0 (/home/simon/src/mcu-uuid-common/test/build/example-Uptime/lib/uuid-common)
...
============================================================================================================================ [SUCCESS] Took 2.08 seconds ============================================================================================================================
```

valeros, 2019-11-20:
Hi @nomis! Thanks for the report. Indeed, v4.1.0 now takes framework compatibility into account so this library is discarded because there is no framework specified when you use native platform. I see two options here:
1. Add `lib_compat_mode = off` to your environment. 
2. Allow any framework in `library.json`.

nomis, 2019-11-20:
Disabling the compatibility check will work since the native environment is only used to fake execution of the examples.

closed 2019-11-20 by nomis