platformio/platformio-core #3283: Does not work anymore on ArchLinux due dependency issues

TheAifam5, 2019-11-21:
What kind of issue is this?

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: ArchLinux

**PlatformIO Version** (`platformio --version`): 4.0.0

### Description of problem
Currently ArchLinux repo has `python-marshmallow 2.20.1-3` and in testing `python-marshmallow 3.2.2-1`, which can be merged in the main repo, everything will obviously not work anymore.

#### Steps to Reproduce

1. Install platformio from AUR and edit for 4.0.0 version
2. Run platformio

### Actual Results
```
The 'marshmallow<3,>=2.20.5' distribution was not found and is required by platformio
```

### Expected Results
More Linux distro friendly.

TheAifam5, 2019-11-26:
For ArchLinux users I made a workaround, please look here:
https://aur.archlinux.org/packages/platformio/#comment-717511

ivankravets, 2019-11-26:
Just create a Python virtual environment and forget about all problems. See http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2019-11-26 by ivankravets

greyltc, 2020-01-30:
This doesn't seem like an actual solution that warrants the issue being closed. It's a great workaround. Shouldn't this project strive to work with the latest stable versions of its dependencies? What is it about about marshmallow beyond version 3 that breaks?

ivankravets, 2020-01-30:
We have fixed this issue in the latest development version. The PIO Core 4.1.1 is planned for release next week.

You can check it in https://github.com/platformio/platformio-core/blob/develop/setup.py#L36

greyltc, 2020-01-30:
Thanks! Didn't realize it was solved and apologies for the noise

ivankravets, 2020-02-12:
PlatformIO Core 4.2 is out! https://community.platformio.org/t/platformio-core-4-2/11980/2