platformio/platformio-core #3286: Multiple bugs on PIO Plus

lackdaz, 2019-11-25:
I've raised this in the forums before but it seems like there are serialisation errors for core functions of PIO Plus on Mac OSX. I've upgraded to 4.1.0 and these problems persist with the addition of a new 'sequence item 1` error.

Host machine: MBP 13" 10.14.6
Agent: Raspberry Pi 3B+
PIO Plus: 2.6.0
PIO: 4.1.0

### Description of problem
running `pio remote agent list` or `pio remote device list` results in 
```PIO Plus (https://pioplus.com) v2.6.0
sequence item 1: expected str instance, bytes found```

running `remote serial monitor outputs
```
PIO Plus (https://pioplus.com) v2.6.0
'hwid'
```
however, if I specify the specific agent name and use the appropriate upload flags it runs ok. So I can imagine it would be passing your upload tests.

So to summarize:
1) Agent seems to be up and running
2) Uploading code works fine as before
3) Remote monitor, or listing agents do not work

ivankravets, 2019-11-26:
Do you use the same Python 3 on agent and client machines? You can't mix Python 2 vs Python 3.

lackdaz, 2019-12-11:
Agent: 3.5.3
Client: 3.7.1

Does it matter if the agent machine defaults to python 2.7 when using `python` instead of python3 `bindings`?

lackdaz, 2019-12-16:
It doesn't work as well when I had both agent and client machines on the same python version.

On agent:
```
pi@prusauno:~$ python
Python 2.7.13 (default, Sep 26 2018, 18:42: 22) 
[GCC 6.3.0 20170516] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()
pi@prusauno:~$ pio remote agent start
PIO Plus (https://pioplus.com) v2.6.1
2019-12-16 12:42:  43 [info] Name: prusauno
2019-12-16 12:42:  43 [info] Connecting to PIO Remote Cloud
2019-12-16 12:42:  49 [info] Successfully connected
2019-12-16 12:42:  49 [info] Authenticating
2019-12-16 12:42: 50 [info] Successfully authorized
```

On client:
```
$ python
Python 2.7.13 (default, Jan  4 2019, 00:20: 41) 
[GCC 4.2.1 Compatible Apple LLVM 9.0.0 (clang-900.0.39.2)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()
seth ~
$ pio remote agent list
PIO Plus (https://pioplus.com) v2.6.1
sequence item 1: expected str instance, bytes found
```

radimsejk, 2020-02-03:
Any progress on this? I have exactly the same problem with same terminal outputs (except my computer name of course) on my machines (both Windows 10 and Python 2.7.14).

ivankravets, 2020-02-03:
@radimsejk Are you sure that you run PlatformIO with the same Python major version? Du you have Python 3 isntalled on these machines? We recommend using Python 3 instead of 2.

lackdaz, 2020-02-06:
@ivankravets What python major version would you recommend running on for both machines? Is there some known setup that works right now?

ivankravets, 2020-02-06:
I tested with Python 2 and Python 3. How did you install PIO Core? 
```sh
pip install platformio # will install for Python 2
pip3 install platformio # will install for Python 3
```

lackdaz, 2020-02-06:
@ivankravets It was a while back, but I think I may have used your super-quick script. Should I just reinstall it?

ivankravets, 2020-02-06:
I never install PlatformIO Core globally into the system. We work on he new `get-platformio.py` script/installer which will install PIO Core into isolated environment `~/.platformio/penv` and you will be able to select a python version.

I recommend do the next:

```
cd /tmp
python3 -m venv pio
source pio/bin/activate
python --version
pip install platformio
pio remote agent start
```

lackdaz, 2020-02-06:
Thanks again @ivankravets , I'll write this into a setup script to simplify the process. So I'm daemonising the agent start process I would also need to activate the virtual environment?

ivankravets, 2020-02-06:
> So I'm daemonising the agent start process I would also need to activate the virtual environment?

Python's `virtualenv` will do this automatically for you when you call binary from env:
```
/tmp/pio/bin/platformio --help
```

lackdaz, 2020-02-06:
Alright I'll try to work it out on both plus and core and let you know the results. Definitely my oversight when installing to global.

lackdaz, 2020-02-06:
Hi @ivankravets ,

I can confirm that from some of the functions that had problems has now recovered after running the remote agent in a virtual environment. My client remains unchanged. I'm running Python 3.7.3 on the remote agent and 3.5 on my client machine.

My issue is solved but I'll close it after @radimsejk solves his issues too?

radimsejk, 2020-02-06:
Thank you for waiting for me, but I do not have time to play with it now. So please feel free to close it.

closed 2020-02-06 by lackdaz

lackdaz, 2020-02-07:
@ivankravets Do you know how do I uninstall platformio if I had used your script? I'm trying to get PIO Core out of my global environment

ivankravets, 2020-02-07:
What is your OS? How did you install it? You can check PIO Core location wuth `which pio` command on Unix.

lackdaz, 2020-02-09:
MacOSX, I'm pretty sure I used your super quick script too.

`which pio` returned `/Users/lackdaz/.platformio/penv/bin/pio`

It looks like its already in a virtual environment actually. Hmm I guess it wouldn't be a problem then.