platformio/platformio-core #3288: [Bug] Defines added in extra_script are lost when using platform_packages

TD-er, 2019-11-26:
- [x] **PlatformIO Core**.

### Configuration

**Operating system**: Windows (VS code)

**PlatformIO Version**  4.1.1b3

### Description of problem

I use a Python script to set/add defines before building the project.

When using the "old" way to define a platform, the defines set in the Python script are present at compile time.
When using `platform` + `platform_packages`, they are lost.

This is the bug report of what I noticed here: https://github.com/platformio/platformio-core/issues/3239#issuecomment-558358080

Working:
```
platform                  = https://github.com/platformio/platform-espressif8266.git#feature/stage
```

Not working:
```
platform                  = espressif8266
platform_packages         =  
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.1
```

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[platformio]

[esp82xx_defaults]
build_flags               = -D BUILD_GIT='"${sysenv.TRAVIS_TAG}"'
                            -D NDEBUG
                            -lstdc++ -lsupc++
                            -mtarget-align
                            -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
                            -DVTABLES_IN_FLASH
                            -DPUYA_SUPPORT=1

[core_2_6_0]
platform                  = https://github.com/Jason2866/platform-espressif8266.git#core_2_6_0
build_flags               = ${esp82xx_2_6_x.build_flags} 
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703

[core_2_6_1]
platform                  = https://github.com/Jason2866/platform-espressif8266.git#core_2_6_1
build_flags               = ${esp82xx_2_6_x.build_flags} 
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703

[core_stage]
platform                  = https://github.com/platformio/platform-espressif8266.git#feature/stage
build_flags               = ${esp82xx_2_6_x.build_flags} -DARDUINO_ESP8266_RELEASE='"2.6.0-dev stage"'

[core_2_6_1_wifimesh]
platform                  = espressif8266
;platform_packages         = 
;    framework-arduinoespressif8266 @ https://github.com/TD-er/Arduino.git#wifi_mesh_update_2.2
platform_packages         =  
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.1
build_flags               = ${esp82xx_2_6_x.build_flags} 
                            -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703

[common]
board_build.f_cpu         = 80000000L
build_flags               = -DHTTPCLIENT_1_1_COMPATIBLE=0
build_unflags             = -DDEBUG_ESP_PORT
lib_deps                  = https://github.com/TD-er/ESPEasySerial.git
lib_ignore                = ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS
;lib_ignore                = ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR
lib_ldf_mode              = chain
lib_archive               = false
upload_speed              = 115200
framework                 = arduino
board                     = esp12e
monitor_speed             = 115200
;targets                   = size, checkprogsize
targets                   =

[env]
framework                 = ${common.framework}
lib_ldf_mode              = ${common.lib_ldf_mode}
lib_archive               = ${common.lib_archive}
lib_deps                  = ${common.lib_deps}
lib_ignore                = ${common.lib_ignore}
upload_speed              = ${common.upload_speed}
monitor_speed             = ${common.monitor_speed}
targets                   = ${common.targets}

[regular_platform]
build_unflags             =
build_flags               = ${core_2_5_2.build_flags} ${common.build_flags}
platform                  = ${core_2_5_2.platform}

[beta_platform]
build_unflags             =
build_flags               = ${core_stage.build_flags} ${common.build_flags}
platform                  = ${core_stage.platform}

[esp8266_4M1M]
board                     = esp12e
board_build.flash_mode    = dio
board_upload.maximum_size = 1044464
board_build.f_cpu         = ${common.board_build.f_cpu}
build_unflags             = ${regular_platform.build_unflags}
build_flags               = -Wl,-Tesp8266.flash.4m1m.ld

; Environment for very simple build tests to assist in reporting issues.
[env:MCVE_ESP8266_4M1M]
platform                  = ${beta_platform.platform}
board                     = ${esp8266_4M1M.board}
board_build.f_cpu         = ${esp8266_4M1M.board_build.f_cpu}
board_build.flash_mode    = ${esp8266_4M1M.board_build.flash_mode}
build_unflags             = ${esp8266_4M1M.build_unflags}
build_flags               = ${esp8266_4M1M.build_flags} ${beta_platform.build_flags} -DPLUGIN_BUILD_CUSTOM
lib_ignore                = ESP32_ping, ESP32WebServer

extra_scripts             = pre:pre_extra_script_mcve.py

[env:spec_wifimesh_ESP8266_4M1M]
extends                   = esp8266_4M1M
framework                 = arduino
build_unflags             =
platform                  = ${core_2_6_1_wifimesh.platform}
platform_packages         = ${core_2_6_1_wifimesh.platform_packages}
build_flags               = ${core_2_6_1_wifimesh.build_flags} 
                            ${esp8266_4M1M.build_flags} 
                            -DPLUGIN_BUILD_CUSTOM
lib_ignore                = ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS
extra_scripts             = pre:pre_extra_script_mcve.py

```

**The content of `pre_extra_script_mcve.py`:**
```python
Import("env")
import os

env.Append(CPPDEFINES=[
  "PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703",
  "ESPEASY_MCVE_BUILD"
  # ,"NO_HTTP_UPDATER"
  # ,("WEBSERVER_RULES_DEBUG", "0")
])
if os.path.isfile('src/Custom.h'):
  env['CPPDEFINES'].append("USE_CUSTOM_H")
else:
  env['CPPDEFINES'].extend([
        "USES_P001",  # Switch
        "USE_SETTINGS_ARCHIVE"
  ])

print(env['CPPDEFINES'])
```

**Source file to reproduce issue:**
```cpp
#include <Arduino.h>

void setup() {

  #ifdef USES_P001
  #warning "USES_P001 defined"
  #else
    #warning "USES_P001 NOT defined"
  #endif

}

void loop() {

}
```

### Additional info
Build output:
```
Compiling .pio\build\MCVE_ESP8266_4M1M\src\main.cpp.o
src\main.cpp:14:4: warning: #warning "USES_P001 defined" [-Wcpp]
   #warning "USES_P001 defined"
    ^
```

```
Compiling .pio\build\spec_wifimesh_ESP8266_4M1M\FrameworkArduino\MD5Builder.cpp.o
src\main.cpp:16:6: warning: #warning "USES_P001 NOT defined" [-Wcpp]
     #warning "USES_P001 NOT defined"
      ^
```

Jason2866, 2019-11-26:
@TD-er  Tried your test sketch (in my fast adapted Tasmota platformio.ini). It does work for me. with script enabled
[image]
script disabled
[image]

[platformio.ini.txt](https://github.com/platformio/platformio-core/files/3891624/platformio.ini.txt)

TD-er, 2019-11-26:
@Jason2866 What PIO version do you use?

TD-er, 2019-11-26:
@Jason2866 In the platformio.ini file you linked, you have the `[core_stage]` defined after it is used for the first time. 
The same for `[core_active]`
Can you swap those, to see if that makes a difference?

Jason2866, 2019-11-26:
@TD-er  Sorry i do not understand what you mean.

TD-er, 2019-11-26:
In your platformio.ini:

Line 49:
```
[common]
framework                 = arduino
board                     = esp01_1m
board_build.flash_mode    = dout

platform                  = ${core_active.platform}
platform_packages         = ${core_active.platform_packages}
build_flags               = ${core_active.build_flags}
```

Line 102:
```
[core_active]
...
platform                  = ${core_stage.platform}
build_flags               = ${core_stage.build_flags}
platform_packages         = ${core_stage.platform_packages}
```

Line 194:
```
[core_stage]
```

Meaning, the definitions of the scope is after it is being referred to.

TD-er, 2019-11-26:
I made a [repository](https://github.com/TD-er/MCVE_ESPxx) for this, so it is easier to do more MCVE tests.

Can you check if this one does fail on your system too?

Jason2866, 2019-11-26:
The test repo does work too for me

[image]

TD-er, 2019-11-26:
And the other build, which is supposed to fail?

Jason2866, 2019-11-26:
Does fail with installing with ESPEasySerial
[image]

TD-er, 2019-11-26:
Well for the MCVE it is not needed, so that one can be removed from the build definitions

Jason2866, 2019-11-26:
Works too
[image]

TD-er, 2019-11-26:
What Python version do you run?
I have Python 3.8 here for my PIO builds.

Jason2866, 2019-11-26:
Still on Python 2.7 for PlatformIO

TD-er, 2019-11-26:
Built it on a clean install in Linux with Python 2.7 and that one also gives the same issue:
```
Compiling .pio/build/MCVE_ESP8266_4M1M/src/main.cpp.o
src/main.cpp:14:4: warning: #warning "USES_P001 defined" [-Wcpp]
   #warning "USES_P001 defined"
    ^
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/Esp.cpp.o
```

```
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/Tone.cpp.o
src/main.cpp:16:6: warning: #warning "USES_P001 NOT defined" [-Wcpp]
     #warning "USES_P001 NOT defined"
      ^
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/Stream.cpp.o
```

ivankravets, 2019-11-26:
Could someone prepare a simple "Blink/Hello" project with scripts to reproduce this issue?

Jason2866, 2019-11-26:
I am using Windows. Maybe that makes the difference?

TD-er, 2019-11-26:
> Could someone prepare a simple "Blink/Hello" project with scripts to reproduce this issue?

See: https://github.com/TD-er/MCVE_ESPxx
It has the same content as in this issue.
I could git clone, activate python venv and run `pio run` to see it fail.

TD-er, 2019-11-26:
> I am using Windows. Maybe that makes the difference?

Me too.
- Windows 10 & Python 3.8
- Ubuntu 18.04 & venv running Python 2.7

ivankravets, 2019-11-28:
@valeros could you reproduce this issue?

valeros, 2019-11-28:
Unfortunately, I cannot reproduce the issue. I cloned @TD-er repository then and compiled `spec_wifimesh_ESP8266_4M1M` environment. PlatformIO successfully cloned the specified the framework and I can see:
```
src\main.cpp:14:4: warning: #warning "USES_P001 defined" [-Wcpp]
   #warning "USES_P001 defined"
```
Tried on Windows with the following combinations: Python: `2.7`, `3.7`, `3.8`, `PlatformIO 4.1.0`, `4.1.1b4`

@TD-er  Am I missing something?

TD-er, 2019-11-28:
Hmm that's very strange as I really cannot build it here.
I will try it later on a completely different computer.

TD-er, 2019-11-28:
Just to show you the entire output, on a computer (Ubuntu 18.04), which was never used to build this test environment.

```
gijs@htpc-nuc:~/GitHub/TD-er$ git clone https://github.com/TD-er/MCVE_ESPxx.git
Cloning into 'MCVE_ESPxx'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 10 (delta 0), reused 7 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), done.
gijs@htpc-nuc:~/GitHub/TD-er$ cd MCVE_ESPxx/
gijs@htpc-nuc:~/GitHub/TD-er/MCVE_ESPxx$ ll
total 36
drwxrwxr-x 4 gijs gijs 4096 nov 28 23:13 ./
drwxrwxr-x 4 gijs gijs 4096 nov 28 23:13 ../
drwxrwxr-x 8 gijs gijs 4096 nov 28 23:13 .git/
-rw-rw-r-- 1 gijs gijs  104 nov 28 23:13 .gitignore
-rw-rw-r-- 1 gijs gijs 5600 nov 28 23:13 platformio.ini
-rw-rw-r-- 1 gijs gijs  667 nov 28 23:13 pre_extra_script_mcve.py
-rw-rw-r-- 1 gijs gijs   75 nov 28 23:13 README.md
drwxrwxr-x 2 gijs gijs 4096 nov 28 23:13 src/
gijs@htpc-nuc:~/GitHub/TD-er/MCVE_ESPxx$ source ~/.venv/python2.7/
bin/   lib/   local/ share/
gijs@htpc-nuc:~/GitHub/TD-er/MCVE_ESPxx$ source ~/.venv/python2.7/bin/activate
(python2.7) gijs@htpc-nuc:~/GitHub/TD-er/MCVE_ESPxx$ pio --version
PlatformIO, version 4.1.0
(python2.7) gijs@htpc-nuc:~/GitHub/TD-er/MCVE_ESPxx$ pio run
Processing MCVE_ESP8266_4M1M (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: esp12e; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
['PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', 'ESPEASY_MCVE_BUILD', 'USES_P001', 'USE_SETTINGS_ARCHIVE']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (Stage) 2.3.0-alpha.2 #3500fb2 > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: toolchain-xtensa 2.50400.170719 (5.4.0), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0), framework-arduinoespressif8266 8b3b600
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing ESPEasySerial
git version 2.17.1
Cloning into '/home/gijs/GitHub/TD-er/MCVE_ESPxx/.pio/libdeps/MCVE_ESP8266_4M1M/_tmp_installing-JDTHfs-package'...
ESPeasySerial @ 7ddbf6c has been successfully installed!
Found 28 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESPeasySerial> 1.0.13 #7ddbf6c
Building in release mode
Generating .pio/build/MCVE_ESP8266_4M1M/core/core_version.h
Generating LD script .pio/build/MCVE_ESP8266_4M1M/ld/local.eagle.app.v6.common.ld
Archiving .pio/build/MCVE_ESP8266_4M1M/libFrameworkArduinoVariant.a
Indexing .pio/build/MCVE_ESP8266_4M1M/libFrameworkArduinoVariant.a
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/StackThunk.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/libfe5/ESPeasySerial/ESPeasySerial_ESP8266.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/FunctionalInterrupt.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/src/main.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/HardwareSerial.cpp.o
src/main.cpp:14:4: warning: #warning "USES_P001 defined" [-Wcpp]
   #warning "USES_P001 defined"
    ^
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/Print.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/MD5Builder.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/libfe5/ESPeasySerial/ESPEasySerial_ESP8266_noSWserial.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/Esp-frag.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/libfe5/ESPeasySerial/ESPEasySoftwareSerial.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/FSnoop.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/Schedule.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/Esp-version.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/libfe5/ESPeasySerial/ESPEasySerial_ESP32.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/Esp.cpp.o
Compiling .pio/build/MCVE_ESP8266_4M1M/FrameworkArduino/FS.cpp.o
[... 255 lines cut ...]
=============================================================================== [SUCCESS] Took 45.19 seconds ===============================================================================

Environment                 Status    Duration
--------------------------  --------  ------------
MCVE_ESP8266_4M1M           SUCCESS   00:00:16.163
spec_wifimesh_ESP8266_4M1M  SUCCESS   00:00:45.195
================================================================================ 2 succeeded in 00:01:01.358 ================================================================================

*********************************************************************************************************************************************************************************************
There are the new updates for platforms (espressif8266)
Please update them via `platformio platform update` command.

If you want to manually check for the new versions without updating, please use `platformio platform update --dry-run` command.
*********************************************************************************************************************************************************************************************

(python2.7) gijs@htpc-nuc:~/GitHub/TD-er/MCVE_ESPxx$
```

The "success build":
[image]

The "fail build":
[image]

So it really baffles me why anybody else is reporting it does build and I tried on 4 systems here and all fail.

TD-er, 2019-11-28:
Just to let you know, since core 2.6.2 was just released.
I also tested with that core, so I just added this line:
```
platform_packages = 
	framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.2
```

And the same result:
```
src\main.cpp:16:6: warning: #warning "USES_P001 NOT defined" [-Wcpp]
     #warning "USES_P001 NOT defined"
Compiling .pio\build\spec_wifimesh_ESP8266_4M1M\FrameworkArduino\Schedule.cpp.o
      ^
```

So just to make sure it is not in my own forked repo or anything.

valeros, 2019-11-29:
That's really strange, but I still cannot reproduce the issue. I tried all the steps you described above on clean Ubuntu in virtual box and here is my output:
```
(pio410) valeros@valeros-VirtualBox:/tmp/MCVE_ESPxx$ python --version
Python 2.7.12
(pio410) valeros@valeros-VirtualBox:/tmp/MCVE_ESPxx$ pio --version
PlatformIO, version 4.1.0
(pio410) valeros@valeros-VirtualBox:/tmp$ git clone https://github.com/TD-er/MCVE_ESPxx.git
Cloning into 'MCVE_ESPxx'...
remote: Enumerating objects: 10, done.
remote: Counting objects: 100% (10/10), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 10 (delta 0), reused 7 (delta 0), pack-reused 0
Unpacking objects: 100% (10/10), done.
Checking connectivity... done.
(pio410) valeros@valeros-VirtualBox:/tmp$ cd MCVE_ESPxx/
(pio410) valeros@valeros-VirtualBox:/tmp/MCVE_ESPxx$ ll
total 36
drwxrwxr-x  4 valeros valeros 4096 лис 29 11:56 ./
drwxrwxrwt 11 root    root    4096 лис 29 11:57 ../
drwxrwxr-x  8 valeros valeros 4096 лис 29 11:56 .git/
-rw-rw-r--  1 valeros valeros  104 лис 29 11:56 .gitignore
-rw-rw-r--  1 valeros valeros 5600 лис 29 11:56 platformio.ini
-rw-rw-r--  1 valeros valeros  667 лис 29 11:56 pre_extra_script_mcve.py
-rw-rw-r--  1 valeros valeros   75 лис 29 11:56 README.md
drwxrwxr-x  2 valeros valeros 4096 лис 29 11:56 src/
(pio410) valeros@valeros-VirtualBox:/tmp/MCVE_ESPxx$ pio run
************************************************************************************************************************************************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
************************************************************************************************************************************************************************************************************

Processing MCVE_ESP8266_4M1M (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; board: esp12e; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing platform-espressif8266
git version 2.7.4
Cloning into '/home/valeros/.platformio/platforms/_tmp_installing-knq42C-package'...
remote: Enumerating objects: 103, done.
remote: Counting objects: 100% (103/103), done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 103 (delta 55), reused 40 (delta 11), pack-reused 0
Receiving objects: 100% (103/103), 30.10 KiB | 0 bytes/s, done.
Resolving deltas: 100% (55/55), done.
Checking connectivity... done.
espressif8266 @ dc6771b has been successfully installed!
The platform 'https://github.com/platformio/platform-espressif8266.git#feature/stage' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa @ ~2.40802.191122
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
toolchain-xtensa @ 2.40802.191122 has been successfully installed!
PackageManager: Installing tool-esptool @ <2
tool-esptool @ 1.413.0 has been successfully installed!
PackageManager: Installing tool-esptoolpy @ ~1.20800.0
tool-esptoolpy @ 1.20800.0 has been successfully installed!
PackageManager: Installing framework-arduinoespressif8266
git version 2.7.4
Cloning into '/home/valeros/.platformio/packages/_tmp_installing-I9DqcZ-package'...
remote: Enumerating objects: 1636, done.
remote: Counting objects: 100% (1636/1636), done.
remote: Compressing objects: 100% (1395/1395), done.
remote: Total 1636 (delta 198), reused 833 (delta 75), pack-reused 0
Receiving objects: 100% (1636/1636), 23.23 MiB | 5.92 MiB/s, done.
Resolving deltas: 100% (198/198), done.
Checking connectivity... done.
Submodule 'libraries/ESP8266SdFat' (https://github.com/earlephilhower/ESP8266SdFat.git) registered for path 'libraries/ESP8266SdFat'
Submodule 'libraries/LittleFS/lib/littlefs' (https://github.com/ARMmbed/littlefs.git) registered for path 'libraries/LittleFS/lib/littlefs'
Submodule 'libraries/SoftwareSerial' (https://github.com/plerup/espsoftwareserial.git) registered for path 'libraries/SoftwareSerial'
[... 368 lines cut ...]
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/time.cpp.o
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/uart.cpp.o
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/umm_malloc/umm_info.c.o
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/umm_malloc/umm_integrity.c.o
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/umm_malloc/umm_local.c.o
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/umm_malloc/umm_malloc.cpp.o
Compiling .pio/build/spec_wifimesh_ESP8266_4M1M/FrameworkArduino/umm_malloc/umm_poison.c.o
Archiving .pio/build/spec_wifimesh_ESP8266_4M1M/libFrameworkArduino.a
Indexing .pio/build/spec_wifimesh_ESP8266_4M1M/libFrameworkArduino.a
Linking .pio/build/spec_wifimesh_ESP8266_4M1M/firmware.elf
[... cut ...]
======================================================================================= [SUCCESS] Took 47.03 seconds =======================================================================================

Environment                 Status    Duration
--------------------------  --------  ------------
MCVE_ESP8266_4M1M           SUCCESS   00:01:06.539
spec_wifimesh_ESP8266_4M1M  SUCCESS   00:00:47.029
======================================================================================= 2 succeeded in 00:01:53.568 =======================================================================================
```

TD-er, 2019-11-29:
That's really strange.

So now we're at the level of comparing pip packages installed:

```
gijs@htpc-nuc:~$ source ~/.venv/python2.7/bin/activate
(python2.7) gijs@htpc-nuc:~$ pip freeze
alabaster==0.7.12
Babel==2.6.0
bottle==0.12.15
certifi==2018.11.29
chardet==3.0.4
click==5.1
colorama==0.4.1
CommonMark==0.5.4
docutils==0.14
idna==2.7
imagesize==1.1.0
Jinja2==2.10
MarkupSafe==1.1.0
marshmallow==2.20.5
packaging==18.0
pkg-resources==0.0.0
platformio==4.1.0
pyelftools==0.25
Pygments==2.3.0
pyparsing==2.3.0
pyserial==3.4
pytz==2018.7
recommonmark==0.4.0
requests==2.20.1
semantic-version==2.8.1
six==1.11.0
snowballstemmer==1.2.1
Sphinx==1.8.4
sphinx-bootstrap-theme==0.6.5
sphinxcontrib-websupport==1.1.0
tabulate==0.8.3
typing==3.6.6
urllib3==1.24.1
```
Can you cut/paste the pip freeze output and copy it to a file like `requirements.txt` and then run (from your Python env) `pip install -r requirements.txt` and run the build again?

valeros, 2019-11-29:
@TD-er Here is the packages installed in my environment:
```
bottle==0.12.17
certifi==2019.11.28
chardet==3.0.4
Click==7.0
colorama==0.4.1
idna==2.8
marshmallow==2.20.5
platformio==4.1.0
pyelftools==0.25
pyserial==3.4
requests==2.22.0
semantic-version==2.8.3
tabulate==0.8.6
urllib3==1.25.7
```
But despite the huge difference between our environments, I still cannot reproduce this issue  even with all packages from your `requirements.txt` (pip wasn't able to install `pkg-resources` so I removed it).

TD-er, 2019-11-29:
What else can be of influence for these kind of bugs?
I do see it on Ubuntu 18.04 and Windows.
Linux has Python 2.7 and Windows Python 3.8
Tested with PIO 4.1 and 4.1.1b4 (and b5)

valeros, 2019-11-29:
I'll try other systems maybe something will come up. Did your try a clean virtualenv only with the packages from my environment?

TD-er, 2019-11-29:
No not yet, but I will try it using Vagrant.

TD-er, 2019-11-30:
It must have something to do with the Python libraries installed.

I did a clean install on Vagrant, which only installs the platformio packages and then runs the build.

```
Compiling .pio/build/Failing_platform_ESP8266_4M1M/src/main.cpp.o
src/main.cpp:14:4: warning: #warning "USES_P001 defined" [-Wcpp]
   #warning "USES_P001 defined"
    ^
Generating LD script .pio/build/Failing_platform_ESP8266_4M1M/ld/local.eagle.app.v6.common.ld
```

See the vagrant script here: https://github.com/TD-er/MCVE_ESPxx

I will clear the Python virtual env here and see what packages should be upgraded to make it work.

TD-er, 2019-11-30:
Hmm, I made all packages in the virtual env in Windows the same as in my Vagrant and removed all installed frameworks in platformio.
But still:

```
Compiling .pio\build\Failing_platform_ESP8266_4M1M\lib17a\ESPeasySerial\ESPEasySerial_ESP32.cpp.o
src\main.cpp:16:6: warning: #warning "USES_P001 NOT defined" [-Wcpp]
     #warning "USES_P001 NOT defined"
      ^
```

This is really frustrating, as it already has taken at least 30 hours already to get this to work. :(

TD-er, 2019-11-30:
See also the debug output on Windows:

### Working build:
```
> Executing task: C:\Users\gijs\.platformio\penv\Scripts\platformio.exe run --verbose --environment MCVE_ESP8266_4M1M <

Processing MCVE_ESP8266_4M1M (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; extends: esp8266_4M1M; build_unflags: ; build_flags: -Wl,-Tesp8266.flash.4m1m.ld -D BUILD_GIT='""', -DNDEBUG, -mtarget-align, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++, -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH, -DPUYA_SUPPORT=1, -DCORE_POST_2_5_0, -O2, -DBEARSSL_SSL_BASIC, -DCORE_POST_2_6_0 -DARDUINO_ESP8266_RELEASE='"2.6.0-dev stage"' -DHTTPCLIENT_1_1_COMPATIBLE=0 -DPLUGIN_BUILD_CUSTOM; lib_ignore: ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS; extra_scripts: pre:pre_extra_script_mcve.py; board: esp12e; board_build.flash_mode: dio; board_upload.maximum_size: 1044464; board_build.f_cpu: 80000000L; framework: arduino; lib_ldf_mode: chain; lib_archive: False; lib_deps: https://github.com/TD-er/ESPEasySerial.git; upload_speed: 115200; monitor_speed: 115200; targets: )
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

['PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', 'ESPEASY_MCVE_BUILD', 'USES_P001', 'USE_SETTINGS_ARCHIVE']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (Stage) 2.3.0-alpha.2 #3500fb2 (git+https://github.com/platformio/platform-espressif8266.git#feature/stage) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 22b2687 (git+https://github.com/esp8266/Arduino.git), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Ignored library C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\libraries\SD
Ignored library C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\libraries\SDFS
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Found 28 compatible libraries
Scanning dependencies...     
Dependency Graph
|-- <ESPeasySerial> 1.0.13 #7ddbf6c [git+https://github.com/TD-er/ESPEasySerial.git] (C:\GitHub\TD-er\MCVE_ESPxx\.pio\libdeps\MCVE_ESP8266_4M1M\ESPeasySerial)
Building in release mode
"c:\users\gijs\.platformio\penv\scripts\python.exe" "C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\makecorever.py" -b "C:\GitHub\TD-er\MCVE_ESPxx\.pio\build\MCVE_ESP8266_4M1M" -p "C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910" -v 2.7.0-dev
xtensa-lx106-elf-gcc -CC -E -P -DVTABLES_IN_FLASH C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\ld\eagle.app.v6.common.ld.h -o .pio\build\MCVE_ESP8266_4M1M\ld\local.eagle.app.v6.common.ld
xtensa-lx106-elf-ar rc .pio\build\MCVE_ESP8266_4M1M\libFrameworkArduinoVariant.a
xtensa-lx106-elf-g++ -o .pio\build\MCVE_ESP8266_4M1M\FrameworkArduino\StackThunk.cpp.o -c -fno-rtti -std=c++11 -mtarget-align -fno-exceptions -O2 -Os -mlongcalls -mtext-section-literals -falign-functions=4 -U__STRICT_ANSI__ -ffunction-sections -fdata-sections -fno-exceptions -Wall -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DESPEASY_MCVE_BUILD -DUSES_P001 -DUSE_SETTINGS_ARCHIVE -DPLATFORMIO=40101 -DESP8266 -DARDUINO_ARCH_ESP8266 -DARDUINO_ESP8266_ESP12 -DBUILD_GIT=\"\" -DNDEBUG -DVTABLES_IN_FLASH -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH -DPUYA_SUPPORT=1 -DCORE_POST_2_5_0 -DBEARSSL_SSL_BASIC -DCORE_POST_2_6_0 "-DARDUINO_ESP8266_RELEASE=\"2.6.0-dev stage\"" -DHTTPCLIENT_1_1_COMPATIBLE=0 -DPLUGIN_BUILD_CUSTOM -DF_CPU=80000000L -D__ets__ -DICACHE_FLASH -DARDUINO=10805 -DARDUINO_BOARD=\"PLATFORMIO_ESP12E\" -DFLASHMODE_DIO -DLWIP_OPEN_SRC -DNONOSDK22x_190703=1 -DTCP_MSS=536 -DLWIP_FEATURES=0 -DLWIP_IPV6=0 -I.pio\build\MCVE_ESP8266_4M1M\core -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\include -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266 -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\tools\sdk\lwip2\include -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\variants\nodemcu C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266\StackThunk.cpp
```
### Failing build:
```
> Executing task: C:\Users\gijs\.platformio\penv\Scripts\platformio.exe run --verbose --environment Failing_platform_ESP8266_4M1M <

[... cut ...] inoespressif8266\tools\sdk\include -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266\tools\sdk\libc\xtensa-lx106-elf\include -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266 -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266\tools\sdk\lwip2\include -IC:\Users\gijs\.platformio\packages\framework-arduinoespressif8266\variants\nodemcu C:\Users\gijs\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\StackThunk.cpp
```

TD-er, 2019-11-30:
Hmm I found something that looks different:

Vagrant (working) build:
```
['PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', 'ESPEASY_MCVE_BUILD', 'USES_P001', 'USE_SETTINGS_ARCHIVE']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 (Stage) 2.4.0-alpha.1 #dc6771b (git+https://github.com/platformio/platform-espressif8266.git#feature/stage) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), framework-arduinoespressif8266 bc204a9 (git+https://github.com/esp8266/Arduino.git#2.6.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0)
```

Failing build:
```
['PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703', 'ESPEASY_MCVE_BUILD', 'USES_P001', 'USE_SETTINGS_ARCHIVE']
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp12e.html
PLATFORM: Espressif 8266 1.7.3 #7507605 (git+https://github.com/TD-er/platform-espressif8266.git#patch/v1.7.3_Puya) > Espressif ESP8266 ESP-12E
HARDWARE: ESP8266 80MHz, 80KB RAM, 1019.98KB Flash
PACKAGES: toolchain-xtensa 1.40802.0 (4.8.2), framework-arduinoespressif8266 bc204a9 (git+https://github.com/esp8266/Arduino.git#2.6.2), tool-esptool 1.413.0 (4.13)
```

Any idea where the toolchain-xtensa is defined?

TD-er, 2019-11-30:
OK, I think I may have found what's wrong here....

I copy/paste the output of the `Processing...` line to Notepad++ and noticed something strange:

[image]

For whatever reason, the lines on the failing platform are split into several lines.
On the Vagrant environment they are also single line.

Windows:
```
Processing Failing_platform_ESP8266_4M1M (platform: espressif8266; platform_packages: framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.2; extends: esp8266_4M1M; build_unflags: ; build_flags: -Wl,-Tesp8266.flash.4m1m.ld -D BUILD_GIT='""', -DNDEBUG, -mtarget-align, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++, 
-DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH, -DPUYA_SUPPORT=1, -DCORE_POST_2_5_0, -O2, -DBEARSSL_SSL_BASIC, -DCORE_POST_2_6_0, -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DHTTPCLIENT_1_1_COMPATIBLE=0 -DPLUGIN_BUILD_CUSTOM; lib_ignore: ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS; extra_scripts: 
pre:pre_extra_script_mcve.py; board: esp12e; board_build.flash_mode: dio; board_upload.maximum_size: 1044464; board_build.f_cpu: 80000000L; framework: arduino; lib_ldf_mode: chain; lib_archive: False; lib_deps: https://github.com/TD-er/ESPEasySerial.git; upload_speed: 115200; monitor_speed: 115200; targets: )
Processing MCVE_ESP8266_4M1M (platform: https://github.com/platformio/platform-espressif8266.git#feature/stage; extends: esp8266_4M1M; build_unflags: ; build_flags: -Wl,-Tesp8266.flash.4m1m.ld -D BUILD_GIT='""', -DNDEBUG, -mtarget-align, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++, -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH, -DPUYA_SUPPORT=1, -DCORE_POST_2_5_0, -O2, -DBEARSSL_SSL_BASIC, -DCORE_POST_2_6_0 -DARDUINO_ESP8266_RELEASE='"2.6.0-dev stage"' -DHTTPCLIENT_1_1_COMPATIBLE=0 -DPLUGIN_BUILD_CUSTOM; lib_ignore: ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS; extra_scripts: pre:pre_extra_script_mcve.py; board: esp12e; board_build.flash_mode: dio; board_upload.maximum_size: 1044464; board_build.f_cpu: 80000000L; framework: arduino; lib_ldf_mode: chain; lib_archive: False; lib_deps: https://github.com/TD-er/ESPEasySerial.git; upload_speed: 115200; monitor_speed: 115200; targets: )
```

Vagrant:
```
Processing Failing_platform_ESP8266_4M1M (platform: espressif8266; platform_packages: framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.2; extends: esp8266_4M1M; build_unflags: ; build_flags: -Wl,-Tesp8266.flash.4m1m.ld -D BUILD_GIT='""', -DNDEBUG, -mtarget-align, -DVTABLES_IN_FLASH, -fno-exceptions, -lstdc++, -DPIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY_LOW_FLASH, -DPUYA_SUPPORT=1, -DCORE_POST_2_5_0, -O2, -DBEARSSL_SSL_BASIC, -DCORE_POST_2_6_0, -DPIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703 -DHTTPCLIENT_1_1_COMPATIBLE=0 -DPLUGIN_BUILD_CUSTOM; lib_ignore: ESP32_ping, ESP32WebServer, IRremoteESP8266, HeatpumpIR, SD(esp8266), SDFS; extra_scripts: pre:pre_extra_script_mcve.py; board: esp12e; board_build.flash_mode: dio; board_upload.maximum_size: 1044464; board_build.f_cpu: 80000000L; framework: arduino; lib_ldf_mode: chain; lib_archive: False; lib_deps: https://github.com/TD-er/ESPEasySerial.git; upload_speed: 115200; monitor_speed: 115200; targets: )
```

ivankravets, 2019-11-30:
I've just finally found a time and tested on my machine. Everything works as expected. What do I recommend to try?

1) Avoid direct using Python's list in SCons environment. See https://github.com/TD-er/MCVE_ESPxx/blob/master/pre_extra_script_mcve.py#L22 . It's correct to use `env.Append(CPPDEFINES=["MACRO1", ("MARCO_NAME", 34)])`
2) Avoid using nono-official dev/platforms. They can REPLACE PRE-defined variables. That's why you do not see custom PRE changes.

closed 2019-11-30 by ivankravets

Jason2866, 2019-11-30:
@TD-er  you are using and define here https://github.com/TD-er/platform-espressif8266/blob/develop/platform.json the (old) extensa build chain.
You have to update!

TD-er, 2019-11-30:
> 2\. Avoid using nono-official dev/platforms. They can REPLACE PRE-defined variables. That's why you do not see custom PRE changes.

Hmm not sure what you mean by this.
The whole idea of the `platform_packages` parameter was to be able to refer to a branch or tag right?
So my latest platform definition in the MCVE repo has:

```
[core_2_6_2_packages]
platform                  = espressif8266
platform_packages         =
	framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.2
```
Is this a "non-official platform" ?

This is also a reply to @Jason2866 who replied while I was typing ;)

Jason2866, 2019-11-30:
This points to old stuff `(git+https://github.com/TD-er/platform-espressif8266.git#patch/v1.7.3_Puya)`

TD-er, 2019-11-30:
> This points to old stuff `(git+https://github.com/TD-er/platform-espressif8266.git#patch/v1.7.3_Puya)`

But that's not used in this way of addressing the platform.
As mentioned, the last tests I ran all today were using the https://github.com/esp8266/Arduino.git#2.6.2

Jason2866, 2019-11-30:
`framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git#2.6.2` does NOT install or change the extensa build chain. It will use what you have in your setup!

TD-er, 2019-11-30:
And where is that defined?

ivankravets, 2019-11-30:
@TD-er could you try to debug this issue on your machine step-by-step? It's very difficult to catch errors if you did a lot of changes.

1) Try using official dev/platform and framework from our registry
2. They try to use `platform_packages = ...`
3. etc, etc

Jason2866, 2019-11-30:
`platform                  = espressif8266@2.3.1`

TD-er, 2019-11-30:
> `platform = espressif8266@2.3.1`

This does seem to do the trick on my Windows PC.
Tried several other `platform_packages =` things and it now does state every time my `USES_P001` is defined.

@ivankravets 
I did change the pre_extra_script.py to do the following:
```Python
cppdefines=[
  "PIO_FRAMEWORK_ARDUINO_ESPRESSIF_SDK22x_190703",
  "ESPEASY_MCVE_BUILD"
  # ,"NO_HTTP_UPDATER"
  # ,("WEBSERVER_RULES_DEBUG", "0")
]

if os.path.isfile('src/Custom.h'):
  cppdefines.append("USE_CUSTOM_H")
else:
  cppdefines.extend([
        "USES_P001",  # Switch
        "USE_SETTINGS_ARCHIVE"
  ])

env.Append(CPPDEFINES=cppdefines)
```
Is this how you meant your point nr. 1?

ivankravets, 2019-11-30:
> Is this how you meant your point nr. 1?

Yes, this is the correct way. Please note that you can call `env.Append()` as many times as you need. See other useful API, such as Append/Prepend, AppendUnique, etc. https://scons.org/doc/production/HTML/scons-user.html#idm1682