platformio/platformio-core #3290: pio lib hangs

424778940z, 2019-11-27:
It just hangs, both install and search are not working.
It shows nothing until I press ctrl+c, for install it will abort, for search it will start showing first page of result.

Windows 10 1903 x64
VS code 1.40.2
fresh installed pio
esp32 devkit v1 board

`
pio lib install 1513
Library Storage: D:\works\Arduino\platformio\esp32_test\.pio\libdeps\esp32doit-devkit-v1
LibraryManager: Installing id=1513

Aborted!
PS D:\works\Arduino\platformio\esp32_test> ^C
PS D:\works\Arduino\platformio\esp32_test> ^C
PS D:\works\Arduino\platformio\esp32_test> ^C
PS D:\works\Arduino\platformio\esp32_test> ^C
PS D:\works\Arduino\platformio\esp32_test> ^C
`

`
pio lib search a    
Nothing has been found by your request
Try a less-specific search or use truncation (or wildcard) operator *
For example: DS*, PCA*, DHT* and etc.

For more examples and advanced search syntax, please use documentation:
https://docs.platformio.org/page/userguide/lib/cmd_search.html
`

ivankravets, 2019-11-28:
Do you use antivirus tools?

ivankravets, 2019-11-28:
Also, could you try to kill all `python.exe` processes and try again?

424778940z, 2019-11-29:
> Also, could you try to kill all `python.exe` processes and try again?

did try that, but didin't help

How ever the issue disappeared after I play with those files.
Seems like a file locking issue cause the hang
you may close this issue.

closed 2019-11-29 by ivankravets