platformio/platformio-core #3291: Support for Logic Green LGT8F328P based boards

IndianTinker, 2019-11-29:
Hey, I recently came across [these](https://www.aliexpress.com/item/33058909260.html) cool Nano clones that run at 32Mhz and support the arduino core and libraries. They also have a few extra features like 12bit ADC and a DAC. You can program them using the arduino IDE using [this](https://github.com/dbuezas/lgt8fx) super cool resource. 

Since, I moving most of my development to PlatformIO it would be great to know how it can be added.

Edzelf, 2019-11-29:
I am also very interested.

darkautism, 2020-04-17:
Hi all,
I create a platform which can running program in platformIO.
The code is here.
https://github.com/darkautism/pio-lgt8fx

Just clone this repository into your %USER%\.platformio\platforms\

An example setting
``` ini
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:LGT8F328P]
platform = lgt8f
board = LGT8F328P
framework = arduino

board_build.f_cpu=16000000L
;internal clock
board_build.clock_source=1
```