platformio/platformio-core #3301: PlatformIO remote test doesn't find sdkconfig for esp32thing

marcovr, 2019-12-09:
### Configuration

**Operating system**: Debian GNU/Linux 10 (latest 'debian' image from Docker) / Ubuntu 18.04.3 LTS (my machine)

**PlatformIO Version**: 4.1.0

### Description of problem
When using the `platformio remote test` command to run tests on a remote esp32thing, the test fails to run because apparently the sdkconfig is not found.
However, if the test is run locally with `platformio test` it works as expected and the test passes.

#### Steps to Reproduce
> **Note**: The error can be reproduced without running PlatformIO inside docker, this just illustrates that the issue is not caused by a misconfiguration of my system.

1. Spin up a machine with an esp32thing attached to it and run the PlatformIO remote agent
```bash
docker run -it --rm --device=/dev/ttyUSB0 debian bash

apt-get update
apt-get -y install git python python-pip
pip install -U platformio
platformio account login  # login with your account
platformio remote agent start
```
2. Spin up another machine with platformio and the project source
```bash
docker run -it --rm debian bash

apt-get update
apt-get -y install git python python-pip curl
pip install -U platformio
mkdir example && cd example
platformio init --board esp32thing
curl -sS http://pastebin.com/raw/LuJwd8NR > test/main.c  # get source file (content included below)
sed -i -e "s/arduino/espidf/" platformio.ini  # switch to ESP-IDF
echo 'monitor_speed = 115200' >> platformio.ini  # set the correct baud rate for my device
platformio run  # This is just to generate the default sdkconfig.h needed for the next line
sed -i -e "s/XTAL_FREQ 40/XTAL_FREQ 26/" src/sdkconfig.h  # set the correct frequency for my device
platformio account login  # login with same account
platformio remote test
```
3. Wait until it builds and observe the results

### Actual Results
```
[...]

Processing * in esp32thing environment
--------------------------------------------------------------------------------------------------------------------------------------
Uploading...
IOError: [Errno 2] No such file or directory: '/root/.platformio/remote/projects/example-ca31ba1401d6e2e20908bb8aa09b6756d7df0deb/src/                                                                                                                                          sdkconfig':
  File "/usr/local/lib/python2.7/dist-packages/platformio/builder/main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/root/.platformio/platforms/espressif32/builder/main.py", line 210:
    target_elf = env.BuildProgram()
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 115:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/builder/tools/platformio.py", line 319:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 604:
    return method(*args, **kw)
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Script/SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "/root/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 745:
    process_project_configs("sdkconfig")
  File "/root/.platformio/platforms/espressif32/builder/frameworks/espidf.py", line 726:
    copy(find_valid_example_file(filename), config)
  File "/usr/lib/python2.7/shutil.py", line 139:
    copyfile(src, dst)
  File "/usr/lib/python2.7/shutil.py", line 97:
    with open(dst, 'wb') as fdst:
===================================================== [FAILED] Took 0.37 seconds =====================================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       esp32thing     FAILED    00:00:00.372
=============================================== 1 failed, 0 succeeded in 00:00:00.372 ===============================================

```

### Expected Results
I expect the test to successfully run and pass as it does when run locally with `platformio test`
```
[...]

Hello world!
test/main.c:15:hello    [PASSED]
-----------------------
1 Tests 0 Failures 0 Ignored
==================================================== [PASSED] Took 204.91 seconds ====================================================

Test    Environment    Status    Duration
------  -------------  --------  ------------
*       esp32thing     PASSED    00:03:24.913
==================================================== 1 succeeded in 00:03:24.913 ====================================================

```

### If problems with PlatformIO Build System:

**The (final) content of `platformio.ini`:**
```ini
[env:esp32thing]
platform = espressif32
board = esp32thing
framework = espidf
monitor_speed = 115200
```

**[Source file](http://pastebin.com/raw/LuJwd8NR) to reproduce issue:**
```c
#include <stdio.h>
#include <unity.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"

void hello() {
    printf("Hello world!\n");
    TEST_ASSERT(1);
}

void app_main(void)
{
    UNITY_BEGIN();
    RUN_TEST(hello);
    UNITY_END();
}
```

### Additional info
The device I used is a SparkFun ESP32 Thing with XTAL frequency of 26 MHz.
I am not sure whether the same issue is present for other devices as well.

ivankravets, 2019-12-21:
Are you sure that the issue is linked with sdkconfig?

Could you try to put it to https://docs.platformio.org/en/latest/projectconf/section_platformio.html#shared-dir

In `platformio.ini` please add `build_flags = -Ishared`

marcovr, 2019-12-21:
> Are you sure that the issue is linked with sdkconfig?

Oh, I just noticed that due to a copy-paste mistake (which I now fixed) the relevant part was hidden way off to the right:
```
IOError: [Errno 2] No such file or directory: '/root/.platformio/[...]/src/sdkconfig':
```
So yes, I do think it's related to sdkconfig or at least that's how I interpret the error message.

> Could you try to put it to https://docs.platformio.org/en/latest/projectconf/section_platformio.html#shared-dir
> In platformio.ini please add build_flags = -Ishared

Do you mean I should copy `sdkconfig` into a new folder `shared` within the project directory?
```
example/
├── include
│   └── README
├── lib
│   └── README
├── platformio.ini
├── shared
│   └── sdkconfig
├── src
│   ├── sdkconfig
│   └── sdkconfig.h
└── test
    ├── README
    └── main.c
```
I just tried that, it made no difference.

ivankravets, 2020-02-14:
This is a default list of files which we sync from `src` and `includes` folders
```
    PSYNC_SRC_EXTS = [
        "c", "cpp", "S", "spp", "SPP", "sx", "s", "asm", "ASM", "h", "hpp",
        "ipp", "ino", "pde", "json", "properties"
    ]
```

The files from `shared` folders are synced without any limitation. Please put files to `shared` and refer from a code.

closed 2020-02-14 by ivankravets