platformio/platformio-core #3302: pio remote run -t nobuild -t upload not working

hpssjellis, 2019-12-12:
# -t nobuild not working on a remote run but works fine on a local run

**Operating system**:  Tested on Windows 10, Ubuntu and Gitpod cloud

**PlatformIO Version** (`platformio --version`):

PlatformIO, version 4.1.0

### Description of problem

"pio remote run -t nobuild -t upload"  not working but 
"pio  run -t nobuild -t upload"   works fine

Community post related to the issue at 

[https://community.platformio.org/t/upload-latest-build-without-a-compile-link/9520/9](https://community.platformio.org/t/upload-latest-build-without-a-compile-link/9520/9)

------------------------------------------------------------------

### Configuration

Normal UNO Arduino setup
Serial proof of upload

#### Steps to Reproduce

1. So running VS-Code local on a windows machine with

pio account login
pio remote agent start

2. Then on a ubuntu laptop running VS-Code after changing your code to version 2! and saving the file. 

pio account login
pio remote run -t upload

3. Now change your code to version 2! and save the file. 

pio remote run -t nobuild -t upload

### Actual Results

Version 2!

### Expected Results

Version 1!

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini

[env:uno]
platform = atmelavr
board = uno
framework = arduino

```

**Source file to reproduce issue:**
```cpp

#include <Arduino.h>

void setup() {
	Serial.begin(9600);
}

void loop() {
	Serial.println("Version 1! ");
	delay(1000);
}

```

### Additional info

Being able to use "-t nobuild" can be very useful when you just want to install a pre-tested working program for that board. I can give my students the pre-tested program and not worry if they have made any environment changes that effect the build.

I found this bug when testing with a cloud method using gitpod.io but the bug seems to happen even when using VS Code on 2 local machines.

Gitpod Cloud docker  github of it [my-gitpod-of-openthread](https://github.com/hpssjellis/my-gitpod-of-openthread) 
[image]

ivankravets, 2020-02-14:
Did you try `pio remote run -t nobuild -r`?

hpssjellis, 2020-02-17:
Thanks @ivankravets I tried pio remote run -t nobuild -r? on the cloud server and got this result.
*** [nobuild] Source `.pio/build/nano_33_iot/firmware.bin' not found, needed by target `nobuild'.

The file firmware.bin is definitely in the folder location.

ark02, 2021-06-19:
I can build with -r 

but without -r, I get

Verbose mode can be enabled via `-v, --verbose` option
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Warning! Please install `99-platformio-udev.rules`.
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules
Auto-detected: /dev/ttyUSB0
Uploading .pio/build/esp32dev/firmware.bin
*** [upload] sh: No such file or directory: 'sh'