platformio/platformio-core #3303: Esp-prog programmer, start debugging ESP32

Misterion96, 2019-12-13:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 10 1803

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.1.0

I'm trying to learn how to debug ESP32 via esp-prog
### Sketch:
#include <Arduino.h>

void setup()
{
    Serial.begin(9600);
}

void loop()
{
    Serial.println("Hello world!");
    delay(1000);
}

### platform.ini:
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
debug_tool = esp-prog
upload_protocol = esp-prog

### After clicking the "Start debugging" button nothing happens
Console contents
undefinedError: Traceback (most recent call last):
  File "c:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 717, in main
    rv = self.invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\click\decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\debug\command.py", line 101, in cli
    debug_options, configuration
  File "c:\.platformio\penv\lib\site-packages\platformio\commands\debug\helpers.py", line 208, in configure_esp32_load_cmds
    for item in configuration.get("flash_extra_images")
UnicodeEncodeError: 'ascii' codec can't encode characters in position 3-8: ordinal not in range(128)

ivankravets, 2019-12-21:
Do you have non-ascii chars in your full project path?

Misterion96, 2019-12-22:
> Есть ли у вас не ascii символы в вашем полном пути проекта?

The presence of non-ASCII does not affect the operation. The solution was to reinstall Python to version 3.8.0.

closed 2019-12-22 by Misterion96

ivankravets, 2019-12-22:
What was your previous Python version?

Misterion96, 2019-12-22:
> Какой была ваша предыдущая версия Python?

Python 2.7.13