platformio/platformio-core #3304: Randomly rebuilding mbed library

HeathPetersen, 2019-12-13:
The mbed library is being rebuilt at random times WITHOUT ANY file changes.  It is easily reproducible as follows:

```
mkdir x; cd x
pio init --board nucleo_l432kc --project-option "framework=mbed"
cat >src/main.cpp <<END
#include <mbed.h>
int main() {}
END
pio run
```

After that, consecutive `pio run` commands may or may not recompile the mbed libraries.

There appear to be multiple mentions of this bug from other people.  For examples, see the following:
* [Always full rebuild allegedly after V4.0.0 - is my platformio.ini file deficient?](https://community.platformio.org/t/always-full-rebuild-allegedly-after-v4-0-0-is-my-platformio-ini-file-deficient/9327)
* [PIO/VSCode recompiles everything, every time](https://community.platformio.org/t/pio-vscode-recompiles-everything-every-time/10135)
* [Entire MBED recompiles every time](https://community.platformio.org/t/entire-mbed-recompiles-every-time/6766)
* [Randomly rebuilding all mbed libraries](https://community.platformio.org/t/randomly-rebuilding-all-mbed-libraries/9628/3)

Now, I admit that some of these may be something else.  However, the fact that multiple people are mentioning this, it only seems to be mentioned related to the mbed framework, and it is easily reproduced as shown above leads me to believe there is a genuine bug.

I am running platformio on Ubuntu 19.04 - fully upgraded.  The `platformio` / `pio` commands start with `#!/usr/bin/python3` so it should be using python 3.  `python3 --version` returns `Python 3.7.3`.  I have tested using platformio v4.1.0 and v4.1.1b5.

FYI - I did a very lengthy / detailed comparison of the whole project and .platformio directorys' timestamps and file sizes to see if something was changing without my knowledge.  I was unable to find anything changing outside of the generated files during runs.

nmathon, 2019-12-30:
Hi, I published a topic here on the same subject :
https://community.platformio.org/t/build-compiles-all-source-files-every-time/11262

And I found this anomaly:
https://gavv.github.io/articles/sconsign-bug/

I tried to modify the line 144 of the file .platformio/penv/lib/python3.7/site-packages/platformio/builder/main.py
And I replaced:
`
env.SConsignFile(
    join("$BUILD_DIR", ".sconsign.py%d%d" % (sys.version_info[0], sys.version_info[1]))
)
`
by
`
env.SConsignFile(
    join("$BUILD_DIR", ".sconsign.dblite")
)
`

And it looks like if it's working now. Can you try ?

HeathPetersen, 2019-12-30:
I have platformio installed using pip3 at the system level so I tried editing `/usr/local/lib/python3.7/dist-packages/platformio/builder/main.py` as per your request / suggestion.  Then I exactly followed my test case as above.  It still randomly rebuilds the mbed sources.  It does NOT do it every time or in a specific pattern.  It is about a 50/50 shot as to whether or not it does it.

ivankravets, 2019-12-30:
> And it looks like if it's working now. Can you try ?

I got you. You have multiple PIO Cores in a system or use them with different Pythons. This is why PlatformIO rebuilds project.

HeathPetersen, 2019-12-31:
I do not have multiple PIOs installed.  Only one at the system level installed using `sudo pip3 install platformio`.  I always run the same command (see example above).

In response to your comment I have done the following:
```
sudo pip3 uninstall platformio
sudo rm -rf ~/.cache/pip
sudo rm -rf ~/.local
which platformio # didn't show one
which pio # didn't show one
pip3 install --user platformio
pio update
pio upgrade
```

The mbed code still rebuilds at random times when I run the example above.

ivankravets, 2019-12-31:
I still do not understand do you have problem with PIO Core or just only with mbed framework. Could you try to build with `framework=arduino`. Does it rebuild too?

HeathPetersen, 2019-12-31:
At @ivankravets request I tried the following:
```
mkdir x; cd x
pio init --board uno --project-option "framework=arduino"
cat >src/main.cpp <<END
#include "Arduino.h"
void setup() {}
void loop() {}
END
pio run
```
After that, consecutive pio run commands never recompile the arduino libraries.  This is how I'd expect the mbed framework to work as well, but it doesn't.