platformio/platformio-core #3306: Handling dependencies that use the same header file names

mlewand, 2019-12-15:
What kind of issue is this?

- [x] **Feature Request**.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

Any

**PlatformIO Version** (`platformio --version`):

```
PlatformIO, version 4.1.0
```

### Description of problem

Currently you have no control of dependencies that use unambiguous file names.

For instance, consider you have 4 dependencies, each of the dependency have file "Base.h".

Then if I use:

```cpp
#include <Base.h>
```

In my project I have no way to tell what file exactly should be loaded.

For instance npm (node package manager) requires you to first provide library name and only then inserted file (e.g. `const foo = require( '@org/library-name/src/foo'`) - this gets around such issues.

Similarly I would expect that I can use:

```cpp
#include <Color/Base.h> // Where "Color" is the library name.
```

And now the compiler knows that it is Color lib that should be used, and not any of three other libraries that are dependencies of my project.

### Additional info

Initially reported here: https://community.platformio.org/t/handling-dependencies-that-use-the-same-header-file-names/9037

ivankravets, 2019-12-21:
See https://docs.platformio.org/en/latest/librarymanager/ldf.html

You can control of libraries via `lib_deps` and `lib_ignore`.

closed 2019-12-21 by ivankravets

mattbaker-digital, 2024-10-23:
Is it not possible to improve this further to meet the requirements of the original post?
This would make it clearer on which header the developer intended to use. Just making it include the headers during the build process doesn't address this and is still ambiguous when reading the code.