platformio/platformio-core #3313: PIO Home not working anymore (Raspberry Pi 3 and Termux)

StephanBeer, 2019-12-23:
Pio Home stopped working on Raspberry Pi3 installation. Reinstalling, updating or upgrading doesn´t help.
Error log on Raspberry Pi :

~~~
Error: Traceback (most recent call last):
  File "/home/openhabian/.local/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/openhabian/.local/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/openhabian/.local/lib/python2.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/home/openhabian/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "/home/openhabian/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/home/openhabian/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
File "/home/openhabian/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/home/openhabian/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: /home/openhabian/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.so: cannot open shared object file: No such file or directory

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
~~~

StephanBeer, 2019-12-23:
the same inside my Termux installation :
~~~Error: Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter          File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)                        File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 717, in main                     rv = self.invoke(ctx)                                    File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke                                                             return super(PlatformioCLI, self).invoke(ctx)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/platformio/commands/home/command.py", line 52, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession     File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: dlopen failed: library "libpthread.so.0" not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
~~~

ivankravets, 2019-12-24:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

StephanBeer, 2019-12-25:
echo $PATH
~~~
/data/data/com.termux/files/home/.yarn/bin /data/data/com.termux/files/home/.fzf/bin /data/data/com.termux/files/usr/bin /data/data/com.termux/files/usr/bin/applets
~~~

StephanBeer, 2019-12-25:
echo $PYTHONPATH returns nothing but 'which python' returns ~~~/data/data/com.termux/files/usr/bin/python~~~

StephanBeer, 2019-12-25:
python --version
~~~
Python 3.8.1
~~~

StephanBeer, 2019-12-25:
pip --version
~~~
pip 19.3.1 from /data/data/com.termux/files/usr/lib/python3.8/site-packages/pip (python 3.8)
~~~

StephanBeer, 2019-12-25:
pip list
~~~
Package                           Version
--------------------------------- -----------
argcomplete                       1.10.3
astroid                           2.3.3
attrs                             19.3.0
backcall                          0.1.0
bleach                            3.1.0
bottle                            0.12.18
bqplot                            0.12.1
certifi                           2019.11.28
chardet                           3.0.4
Click                             7.0
colorama                          0.4.3
cycler                            0.10.0
decorator                         4.4.1
defusedxml                        0.6.0
dot-kernel                        0.1.8
emu86                             0.0.9.3
entrypoints                       0.3
et-xmlfile                        1.0.1
future                            0.18.2
gnuplot-kernel                    0.3.0
graphviz                          0.13.2
greenlet                          0.4.15
hdl-checker                       0.6.12
iarm                              0.1.0
idna                              2.7
ipydatawidgets                    4.0.1
ipykernel                         5.1.3
ipyparallel                       6.2.4
ipython                           7.10.2
ipython-genutils                  0.2.0
ipyvolume                         0.5.2
ipywebrtc                         0.5.0
ipywidgets                        7.5.1
ipykernel                         5.1.3
ipyparallel                       6.2.4
ipython                           7.10.2
ipython-genutils                  0.2.0
ipyvolume                         0.5.2
ipywebrtc                         0.5.0
ipywidgets                        7.5.1
isort                             4.3.21
jdcal                             1.4.1
jedi                              0.15.2
Jinja2                            2.10.3
json5                             0.8.5
jsonschema                        3.2.0
jupyter                           1.0.0
jupyter-c-kernel                  1.2.2
jupyter-client                    5.3.4
jupyter-console                   6.0.0
jupyter-contrib-core              0.3.3
jupyter-contrib-nbextensions      0.5.1
jupyter-core                      4.6.1
jupyter-highlight-selected-word   0.2.0
jupyter-latex-envs                1.4.6
jupyter-nbextensions-configurator 0.4.1
jupyterlab                        1.2.4
jupyterlab-server                 1.0.6
jupyterthemes                     0.20.0
jupytext                          1.3.0
kiwisolver                        1.1.0
lab                               5.1
lazy-object-proxy                 1.4.3
lesscpy                           0.13.0
lxml                              4.4.2
MarkupSafe                        1.1.1
[... 67 lines cut ...]
simplejson                        3.17.0
six                               1.13.0
stevedore                         1.31.0
tabulate                          0.8.6
terminado                         0.8.3
testpath                          0.4.4
tornado                           6.0.3
traitlets                         4.3.3
traittypes                        0.2.1
typed-ast                         1.4.0
typing                            3.7.4.1
typing-extensions                 3.7.4.1
ujson                             1.35
urllib3                           1.24.3
virtualenv                        16.7.9
virtualenv-clone                  0.5.3
virtualenvwrapper                 4.8.4
waitress                          1.4.0
wcwidth                           0.1.7
webencodings                      0.5.1
widgetsnbextension                3.5.1
wrapt                             1.11.2
xonsh                             0.9.13.dev1
zsh-jupyter-kernel                3.2
~~~

StephanBeer, 2019-12-25:
That are all outputs from the Termux installation. The following are from the Raspberry Pi 3

StephanBeer, 2019-12-25:
echo $PATH
~~~
/usr/local/bin:/usr/bin:/bin:/usr/games:/home/openhabian/.fzf/bin
~~~

StephanBeer, 2019-12-25:
echo $PYTHONPATH returns nothing but 'which python' returns
~~~
/usr/bin/python
~~~

StephanBeer, 2019-12-25:
python --version
~~~
Python 2.7.13
~~~

StephanBeer, 2019-12-25:
pip --version
~~~
pip 19.3.1 from /usr/local/lib/python2.7/dist-packages/pip (python 2.7)
~~~

StephanBeer, 2019-12-25:
pip list
~~~
openhabian@openHABianPi  ~  pip list
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Package          Version
---------------- ----------
bottle           0.12.18
certifi          2019.11.28
chardet          3.0.4
Click            7.0
colorama         0.4.3
cryptography     1.7.1
dnspython        1.16.0
enum34           1.1.6
futures          3.3.0
greenlet         0.4.15
idna             2.8
ipaddress        1.0.23
keyring          10.1
keyrings.alt     1.3
marshmallow      2.20.5
msgpack          0.6.1
Pillow           6.2.1
pip              19.3.1
pip-manager      1.0.3
platformio       4.1.0
pyasn1           0.4.8
PyBluez          0.18
pycrypto         2.6.1
pyelftools       0.26
pygobject        3.22.0
pynvim           0.3.2
pyserial         3.4
python-lzo       1.8
pyxdg            0.25
ranger           1.7.1
requests         2.22.0
RPi.GPIO         0.6.5
SecretStorage    2.3.1
semantic-version 2.8.4
setuptools       42.0.2
six              1.12.0
tabulate         0.8.6
trollius         2.2.post1
tzupdate         1.5.0
urllib3          1.25.7
wheel            0.33.6
~~~

ivankravets, 2019-12-25:
How did you install PlatformIO Core?

StephanBeer, 2019-12-25:
I installed platformio on both systems with 'pip install -U platformio', without an virtual environment.

dpippenger, 2019-12-27:
Please take all of this with the consideration I don't do much python development. The issue seems the be that the shared library assets under contrib-pysite for cryptography's bindings seem to be incompatible with the raspberry pi runtime (at least while running raspbian). Examining the objects they seemed to be arm compatible, but I didn't dig into this much further beyond the fact they wouldn't load.

Someone else can probably clean up my workaround, but to circumvent the issue for now inside the venv created by platformio then did a pip install cryptography, finally copied the binary artifacts created from that install over the files inside contrib-pysite.

`cd ~/.platformio`
`source penv/bin/activate`
`pip install cryptography`
`cp -r penv/lib/python3.7/site-packages/cryptography/hazmat/bindings/* packages/contrib-pysite/cryptography/hazmat/bindings/`

Again, sorry for the crude and ugly workaround, I'm sure a more experienced python developer can provide better guidance.

ivankravets, 2019-12-28:
Cloud you try to create virtualenv with Python 2?

```
virtualenv -p python2 pio
source pio/bin/activate
pip install platformio
pio home
```

StephanBeer, 2019-12-28:
~~~
(pio) $ pio home                                           Error: Traceback (most recent call last):                    File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main       cli()  # pylint: disable=no-value-for-parameter          File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/data/data/com.termux/files/home/pio/lib/python2.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
from autobahn.twisted.wamp import ApplicationSession
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: dlopen failed: library "libpthread.so.0" not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

(pio) $

StephanBeer, 2019-12-28:
~~~
(pio)  openhabian@openHABianPi  ~  pio home              Error: Traceback (most recent call last):
  File "/home/openhabian/.local/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/openhabian/.local/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/openhabian/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/openhabian/.local/lib/python2.7/site-packages/platformio/commands/home/command.py", line 52, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/home/openhabian/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "/home/openhabian/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/home/openhabian/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
File "/home/openhabian/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/home/openhabian/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: /home/openhabian/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.so: cannot open shared object file: No such file or directory

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
~~~

StephanBeer, 2019-12-28:
Two completely different systems (Rasp Stretch on Raspberry Pi, Termux terminal emulator on Android), both confirmed working with platformio (home) before. The error is present with or without installing in an virtualenv on both systems

VishalDhayalan, 2020-01-09:
Having the same issue except that I'm installing it through remote SSH on VS Code. Issue post: https://github.com/platformio/platformio-vscode-ide/issues/942#issuecomment-572801030

Any fix yet?

StephanBeer, 2020-02-13:
New platformio update...still not working pio home 
~~~
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 4.2.0!

***********************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
***********************************************************

Error: Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/data/data/com.termux/files/usr/lib/python3.8/site-packages/platformio/commands/home/command.py", line 53, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: dlopen failed: library "libpthread.so.0" not found

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
~~~

StephanBeer, 2020-02-13:
The same on Raspberry Pi 3
~~~
sudo pio home               Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 4.2.0!

***********************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
***********************************************************

Error: Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python2.7/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/platformio/commands/home/command.py", line 53, in cli
    from autobahn.twisted.resource import WebSocketResource
  File "/root/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession
  File "/root/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>
    from autobahn.wamp import protocol, auth
  File "/root/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>
    from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC
  File "/root/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>
    from cryptography.hazmat.primitives import constant_time
  File "/root/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>
    from cryptography.hazmat.bindings._constant_time import lib
ImportError: /root/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.so: cannot open shared object file: No such file or directory

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
~~~

ivankravets, 2020-02-13:
I did some improvement in https://github.com/platformio/platformio-core/commit/a10625a0529d57c3fc385898e362b4239669b2d6

Could you run `pio upgrade --dev` and restart PIO Home? 

P.S: You will need to have GCC compiuer on your machine.

StephanBeer, 2020-02-13:
Ok now its working in termux after compiling with the --dev argument in Termux, only got one complain error
~~~
....Building wheel for cryptography (PEP 517) ... done
  Created wheel for cryptography: filename=cryptography-2.8-cp38-cp38-linux_aarch64.whl size=896432 sha256=e24a8f30837b740e8f74ee1cd6a2e5e82a05f8e9e67c510619546d0396faef96
  Stored in directory: /data/data/com.termux/files/usr/tmp/pip-ephem-wheel-cache-rwxmhz7p/wheels/17/44/d0/4b5d45f432c78049ac766f5e4a13af07e4a72768055137312e
Successfully built cryptography
Installing collected packages: pyasn1, pyasn1-modules, six, pycparser, cffi, cryptography, attrs, service-identity
    Running setup.py install for pycparser ... done
    Running setup.py install for cffi ... done
Successfully installed attrs-19.3.0 cffi-1.14.0 cryptography-2.8 pyasn1-0.4.8 pyasn1-modules-0.2.8 pycparser-2.19 service-identity-18.1.0 six-1.14.0
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cffi-1.14.0-py3.8.egg-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/pycparser already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/attr already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cryptography-2.8.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/_cffi_backend.cpython-38.so already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/six.py already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/attrs-19.3.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/six-1.14.0.dist-info already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/cffi already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/pycparser-2.19-py3.8.egg-info already exists. Specify --upgrade to force replacement.
Collecting idna!=2.3,>=0.6
  Downloading idna-2.8-py2.py3-none-any.whl (58 kB)
     |█████▋                          | 10 kB 1.7 MB/s eta      |███████████▏                    | 20 kB 1.1 MB/s eta      |████████████████▊               | 30 kB 1.6 MB/s eta      |██████████████████████▍         | 40 kB 898 kB/s eta      |████████████████████████████    | 51 kB 1.1 MB/s eta      |████████████████████████████████| 58 kB 1.3 MB/s
ERROR: hdl-checker 0.6.12 has requirement requests==2.20.0, but you'll have requests 2.22.0 which is incompatible.
Installing collected packages: idna
Successfully installed idna-2.8
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/idna already exists. Specify --upgrade to force replacement.
WARNING: Target directory /data/data/com.termux/files/home/.platformio/packages/contrib-pysite/idna-2.8.dist-info already exists. Specify --upgrade to force replacement.
Collecting zeroconf==0.23.0
  Downloading zeroconf-0.23.0-py3-none-any.whl (24 kB)
Collecting ifaddr
  Downloading ifaddr-0.1.6.tar.gz (6.5 kB)
Installing collected packages: ifaddr, zeroconf
    Running setup.py install for ifaddr ... done
Successfully installed ifaddr-0.1.6 zeroconf-0.23.0
CorePackageManager: Installing contrib-piohome @ ~3.1.0
Downloading  [####################################]  100%
contrib-piohome @ 3.1.0 has been successfully installed!

  ___I_
 /\-_--\   PlatformIO Home
/  \_-__\
|[]| [] |  http://127.0.0.1:8008
|__|____|_______________________

Open PlatformIO Home in your browser by this URL => http://127.0.0.1:8008
PIO Home has been started. Press Ctrl+C to shutdown.
~~~

StephanBeer, 2020-02-13:
On the Raspberry Pi doing the same steps updates to a new version but there is no compiling action, its seems the process doesnt get startet
~~~
sudo pio upgrade --dev    Please wait while upgrading PlatformIO ...                 PlatformIO has been successfully upgraded to 4.2.1a1       Release notes: https://docs.platformio.org/en/latest/history.html                                                      openhabian@openHABianPi  ~  sudo pio home               Please wait while upgrading PlatformIO...                  PlatformIO has been successfully upgraded to 4.2.1a1!                                                                                                                            ***********************************************************If you like PlatformIO, please:                            - follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org             - star it on GitHub > https://github.com/platformio/platformio                                                        - try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide                                 - support us with PlatformIO Plus > https://pioplus.com    ***********************************************************                                                           Error: Traceback (most recent call last):                    File "/usr/local/lib/python3.5/dist-packages/platformio/commands/home/command.py", line 59, in cli                      from autobahn.twisted.resource import WebSocketResource
  File "/root/.platformio/packages/contrib-pysite/autobahn/twisted/__init__.py", line 58, in <module>
    from autobahn.twisted.wamp import ApplicationSession     File "/root/.platformio/packages/contrib-pysite/autobahn/twisted/wamp.py", line 50, in <module>                         from autobahn.wamp import protocol, auth                 File "/root/.platformio/packages/contrib-pysite/autobahn/wamp/auth.py", line 43, in <module>                            from cryptography.hazmat.primitives.kdf.pbkdf2 import PBKDF2HMAC                                                    File "/root/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/kdf/pbkdf2.py", line 12, in <module>     from cryptography.hazmat.primitives import constant_time                                                            File "/root/.platformio/packages/contrib-pysite/cryptography/hazmat/primitives/constant_time.py", line 11, in <module>                                                             from cryptography.hazmat.bindings._constant_time import lib
ImportError: /root/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.abi3.so: cannot open shared object file: No such file or directory            
During handling of the above exception, another exception occurred:                                                   
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 764, in __call__                                      return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 717, in main                                          rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/dist-packages/platformio/commands/__init__.py", line 44, in invoke                       return super(PlatformioCLI, self).invoke(ctx)            File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 1137, in invoke                                       return _process_result(sub_ctx.command.invoke(sub_ctx))  File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 956, in invoke                                        return ctx.invoke(self.callback, **ctx.params)           File "/usr/local/lib/python3.5/dist-packages/click/core.py", line 555, in invoke                                        return callback(*args, **kwargs)                         File "/usr/local/lib/python3.5/dist-packages/platformio/commands/home/command.py", line 60, in cli                      except (ImportError, ModuleNotFoundError):             NameError: name 'ModuleNotFoundError' is not defined       
============================================================                                                          
An unexpected error occurred. Further steps:                                                                          * Verify that you have the latest version of PlatformIO using                                                           `pip install -U platformio` command                      
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html                
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues     
============================================================
~~~

StephanBeer, 2020-02-13:
GCC is installed on the Pi
~~~
gcc --version             gcc (Raspbian 6.3.0-18+rpi1+deb9u1) 6.3.0 20170516         Copyright (C) 2016 Free Software Foundation, Inc.          This is free software; see the source for copying conditions.  There is NO                                            warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
~~~

ivankravets, 2020-02-13:
Please re-run `pio upgrade --dev`. It should work now.

closed 2020-02-13 by ivankravets

kgyrtkirk, 2020-04-28:
I have today was forced to upgrade to 4.3.2 (some kind of account change stuff) which reintroduced this issue on my remote agent - but this seems to be fixed in 4.2.1...

ivankravets, 2020-04-28:
@kgyrtkirk which issues do you have with the latest PIO Core 4.3.3?

kgyrtkirk, 2020-04-28:
I've just upgraded to 4.3.3 ; but this issue still persists; it's basically:
```
ImportError: /home/pi/.platformio/packages/contrib-pysite/cryptography/hazmat/bindings/_constant_time.abi3.so: cannot open shared object file: No such file or directory
```
OS is running Raspbian 10 on a raspberry pi1

ivankravets, 2020-04-28:
Please do the next:
1. Install `gcc`
2. Re-run `pio remote agent start`.

kgyrtkirk, 2020-04-28:
not really; it's a stock raspbian image...it already has gcc as well

ivankravets, 2020-04-28:
How did you install PlatformIO Core? Could you try to remove ALL files and folder from `/home/pi/.platformio/packages/contrib-pysite` but KEEP `package.json`. Try `pio remote agent start`. it should build dependencies on your side.

kgyrtkirk, 2020-04-28:
removed `.platformio` a few times so far :)
I've used the os level `pip3` to install it before....so that might be problematic
I've done a fresh install using:
```
python3 -c "$(curl -fsSL https://raw.githubusercontent.com/platformio/platformio/develop/scripts/get-platformio.py)"
```
which have install everything into a `venv`
because it took a long time - while it was running I was looking at the process list; and it seems like instead of using some pre-packaged `contrib-pysite` it used pip to install those packages...
Earlier I've seen this on first startup:
```
CorePackageManager: Installing contrib-pysite @ ~2.37.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
contrib-pysite @ 2.37.191017 has been successfully installed!
```
I see that there are [more recent versions of contrib-pysite](https://bintray.com/platformio/dl-packages/contrib-pysite) ...not sure why I'm getting 2.37

while I was looking into this the installation finished...and I got back the same error...(but I might have not properly removed the .platformio before this run)

now that I've seen the command...I've removed that "cryptography" and run:
```
/home/pi/.platformio/penv/bin/python -m pip install --no-cache-dir --no-compile -t /home/pi/.platformio/packages/contrib-pysite  cryptography
```
and bingo...I got back an Authentication Error which is great...after logging in/etc it started working...and I was able to flash the new fw I wanted to a few hours ago :D

ivankravets, 2020-04-28:
> not sure why I'm getting 2.37

Means, a package for Python 3.7.

>  it used pip to install those packages

This is actually a bug. It was fixed for PIO Home and presents for PIO Remote. Could you reopen a separate issue for the history? I would like to fix it and ask you to re-test. https://github.com/platformio/platformio-core/issues

kgyrtkirk, 2020-04-28:
I'm not sure if you wanted me to open a ticket for the crypto related issue or that the curl stuff installs those packages with pip :)
I've opened #3482 for the crypto library related issue - just ping me and I'll test it.

commit 22e8e02f (2020-02-13): Automatically rebuild contrib-pysite package when import fails // Resolve #3313