platformio/platformio-core #3314: Connection Error accessing api.platformio.org

jacklunn, 2019-12-23:
### Configuration

**Operating system**: MacOS Catalina 10.15

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.1.0

### Description of problem

I'm trying to initialize a new project but the ide hangs.  I dropped to the command line and it looks like api.platformio.org isn't reachable...

#### Steps to Reproduce

from the command line:

% pio boards
[API] ConnectionError: ('Connection aborted.', ConnectionResetError(54, 'Connection reset by peer')) (incremented retry: max=5, total=1)

daveschafer, 2019-12-23:
Same problem here, it seems like the servers are down?

`[API] ConnectionError: HTTPConnectionPool(host='api.platformio.org', port=80): Max retries exceeded with url: /v2/lib/search?query=name%3A%22AsyncTCP-esphome%22 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f0f73bf3cc0>: Failed to establish a new connection: [Errno 110] Connection timed out',)) (incremented retry: max=5, total=1)`

krupaldesai, 2019-12-23:
Yup, even website seems down `https://platformio.org/` :(

fabitom, 2019-12-23:
Yup, can't install libraries. Any way to install from mirror or github ?

tita1705, 2019-12-23:
[Documentation website](http://docs.platformio.org/en/latest) is working

pipe89, 2019-12-23:
It's now up and running.

lukasz-gieraltowski, 2019-12-24:
I still have a problem with other calls like `pio account show` (version 4.1.0)

```PIO Plus (https://pioplus.com) v2.6.1
Error: [API] Invalid response: <html>
<head><title>502 Bad Gateway</title></head>
<body>
<center><h1>502 Bad Gateway</h1></center>
<hr><center>nginx</center>
</body>
</html>```

It looks like a server outage. Looking at the tickets I see, it's not the first time. Maybe it would be a good idea to implement some kind of status page?

ivankravets, 2019-12-24:
Sorry for the issue. This is actually the 2nd time this year. The first was caused by planned maintenance.
Yesterday's issue was caused by our Ukrainian's dedicated server (E1410 System Fatal error). We fixed it by removing 2 banks of RAM. Let's see how will it work next week. If we have a problem again, we will move to VPS or cloud instance.

closed 2019-12-24 by ivankravets

wilmardo, 2019-12-26:
@ivankravets This issue seems back again :)

Is there a statuspage for the API? Might be a good idea to mitigate future questions like this. There is statuspage.io, also an opensource alternative:
https://github.com/CachetHQ/Cachet

bramschats, 2019-12-26:
Same problem here.

ivankravets, 2019-12-26:
It should work now. Sorry for the issue.

olssondaniel1983, 2019-12-26:
Still getting 502

ivankravets, 2019-12-26:
All services are restored, including PIO Plus.

Again, sorry for the isues.

Happy New Year and happy coding with PlatformIO! 😊