platformio/platformio-core #3316: python wouldn't be added to the PATH

aster94, 2019-12-25:
What kind of issue is this?

- [x] **PlatformIO Core**.

### Configuration

**Operating system**:
win 64 bit

**PlatformIO Version** (`platformio --version`):
4.1.0

### Description of problem

We know that Python is required to run platformio. But why is it added to the global PATH variables?
Can platformio index internally it's python without adding it to the system path?
Example for windows: `$env:USERPROFILE/.platformio/python27/python.exe`
and for pip: `$env:USERPROFILE/.platformio/python27/Scripts/pip.exe`

### Expected Results

when i run the command pyhton in my system console I won't run platformio's python

### Additional info

There are a lot of programs that needs python to run (two that cames to my mind: KiCad e FreeCAD) but these program won't add the python PATH to your system PATH. This behaviour isn't error-free the user could have multiple PATH to python and could inadvertitely run platformio's python and ruin the packages without noticing or the user could also uninstall python.

ivankravets, 2020-01-03:
It looks like you use VSCode.

1. Try to remove `.vscode` folder  from a project
2. Restart VSCode
3. Open VSCode > Manu > Terminal > New Terminal

P.S: We don't add python to your system environment path. We add PlatformIO's binaries to VSCode's terminal PATH

closed 2020-01-03 by ivankravets

aster94, 2020-01-06:
Yes i am using VS Code

> P.S: We don't add python to your system environment path. We add PlatformIO's binaries to VSCode's terminal PATH

I think that the problem is here, you add `.platformio\penv\Scripts` to path, isn't it?
but inside that path there is python and pip

```
PS C:\Users\vincenzo\.platformio\penv\Scripts> ls

    Directory: C:\Users\vincenzo\.platformio\penv\Scripts

Mode                LastWriteTime         Length Name
----                -------------         ------ ----
-a----       20/09/2019     08:18           2203 activate
-a----       20/09/2019     08:18            770 activate.bat
-a----       20/09/2019     08:18           8325 activate.ps1
-a----       20/09/2019     08:18           1143 activate_this.py
-a----       20/09/2019     08:18         154108 bottle.py
-a----       20/09/2019     08:19         175177 bottle.pyc
-a----       20/09/2019     08:18         102783 chardetect.exe
-a----       20/09/2019     08:18            512 deactivate.bat
-a----       20/09/2019     08:17         102792 easy_install-2.7.exe
-a----       20/09/2019     08:17         102792 easy_install.exe
-a----       20/09/2019     08:18          35136 miniterm.py
-a----       20/09/2019     08:18          40525 miniterm.pyc
-a----       09/11/2019     14:11         102780 pio.exe
-a----       09/11/2019     14:11         102800 piodebuggdb.exe
-a----       20/09/2019     08:18         102774 pip.exe
-a----       20/09/2019     08:18         102774 pip2.7.exe
-a----       20/09/2019     08:18         102774 pip2.exe
-a----       09/11/2019     14:11         102780 platformio.exe
-a----       20/09/2019     08:17          28160 python.exe
-a----       20/09/2019     08:17        3407360 python27.dll
-a----       20/09/2019     08:17          28160 pythonw.exe
-a----       09/11/2019     14:11          63127 readelf.py
-a----       09/11/2019     14:11          46523 readelf.pyc
-a----       20/09/2019     08:18         102771 tabulate.exe
-a----       20/09/2019     08:18         102770 wheel.exe

```