platformio/platformio-core #3317: Semantic Versioning of lib_deps broken

ModischFabrications, 2019-12-25:
- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

### Configuration

**Operating system**: Win10 Professional

**PlatformIO Version** (`platformio --version`): PlatformIO, version 4.1.0

### Description of problem
The versioning syntax described at https://docs.platformio.org/en/latest/userguide/lib/cmd_install.html#description does not work, at least not from platformio.ini.

Removing special flags like "^" or "~" compiles successfully. 

It seems like someone else already had a similar problem and found no solution: 
https://community.platformio.org/t/library-version-specification-in-platformio-broken/9764

#### Steps to Reproduce

1. Create platformio.ini like attached below
2. Hit compile

### Actual Results

### Expected Results

Any compatible version.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:d1_mini_lite]
platform = espressif8266
board = d1_mini_lite
framework = arduino

build_flags = -Wall ; -size

monitor_speed = 115200

lib_deps =
    FastLED,
    WifiManager@^0.15.0-beta

```

### Additional info
This is the stacktrace: 

```
Processing d1_mini_lite (platform: espressif8266; board: d1_mini_lite; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini_lite.html
PLATFORM: Espressif 8266 2.3.2 > WeMos D1 mini Lite
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: toolchain-xtensa 2.40802.191122 (4.8.2), tool-esptool 1.413.0 (4.13), tool-esptoolpy 1.20800.0 (2.8.0), framework-arduinoespressif8266 2.20603.191216 (2.6.3)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for FastLED, library in registry
Found: https://platformio.org/lib/show/126/FastLED
LibraryManager: Installing id=126
FastLED @ 3.3.2 is already installed       
Looking for WifiManager library in registry
Found: https://platformio.org/lib/show/567/WifiManager
LibraryManager: Installing id=567 @ ^0.15.0-beta      
UndefinedPackageVersion: Could not find a version that satisfies the requirement '^0.15.0-beta' for your system 'windows_amd64':
  File "C:\users\mmf\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\MMF\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\MMF\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\users\mmf\.platformio\platforms\espressif8266\builder\main.py", line 203:
    target_elf = env.BuildProgram()
  File "C:\Users\MMF\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\mmf\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\mmf\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\MMF\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\mmf\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\users\mmf\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\mmf\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "c:\users\mmf\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\users\mmf\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 139:
    requirements or "latest", util.get_systype()
```

eferley, 2019-12-31:
I confirm the same issue on Windows 7 x64 / VSCode-PlatformIO : 
```
Version: 1.41.1 (user setup)
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-18T14:58:56.166Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Windows_NT x64 6.1.7601
```

**`platformio.ini`** : 
```cpp
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps =
  Adafruit Unified Sensor@^1.1.1,
  DHT@^1.0.0
  Adafruit DHT Unified@^1.0.0
monitor_speed = 115200
```

**Building after Clean** : 
```
> Executing task: C:\Users\username\.platformio\penv\Scripts\platformio.exe run <

Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 1.11.1 > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Looking for Adafruit Unified Sensor library in registry
Found: https://platformio.org/lib/show/31/Adafruit Unified Sensor
LibraryManager: Installing id=31 @ ^1.1.1,
UndefinedPackageVersion: Could not find a version that satisfies the requirement '^1.1.1,' for your system 'windows_amd64':
  File "C:\users\username\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\username\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\username\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\username\.platformio\platforms\espressif32\builder\main.py", line 210:
    target_elf = env.BuildProgram()
  File "C:\Users\username\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\username\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\username\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\username\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\username\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\users\username\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\username\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "c:\users\username\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 698:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "c:\users\username\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 139:
    requirements or "latest", util.get_systype()
=================================================================== [FAILED] Took 1.32 seconds ===================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

ivankravets, 2020-01-03:
Yes, please remove the comma. See https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps

closed 2020-01-03 by ivankravets

ModischFabrications, 2020-01-04:
Oh wow, you are right, I somehow assumed a trailing comma is needed. Thanks a lot! Weird that it worked without a pinned version.

"WifiManager@^0.15.0-beta" still does not work, even though https://platformio.org/lib/show/567/WifiManager/installation recommends it. Am I missing something or is "-beta" not supported?