platformio/platformio-core #3331: Unable to update to or install u8g2 library to version 2.27.3

wisegeeksrules, 2020-01-05:
I'm having issues trying to update u8g2 library to version 2.27.3. Trying to update gives me a PIO Core Call Error and this results in the library no longer appearing under Installed libraries.

The only workaround I have at the moment is to install the previous version of the library which is version 2.27.2

Below is the message I'm getting and the screenshot:
[image]

Libraries: Could not install library
PIO Core Call Error: "Library Storage: /home/object-undefined/.platformio/lib\nLibraryManager: Installing id=942 @ ^2.27.3\n\n\nError: Traceback (most recent call last):\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/main.py", line 102, in main\n cli() # pylint: disable=no-value-for-parameter\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 764, in call\n return self.main(*args, **kwargs)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 717, in main\n rv = self.invoke(ctx)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/commands/init.py", line 44, in invoke\n return super(PlatformioCLI, self).invoke(ctx)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 1137, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 1137, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 956, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 555, in invoke\n return callback(*args, **kwargs)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func\n return f(get_current_context(), args, kwargs)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/commands/lib.py", line 151, in lib_install\n library, silent=silent, interactive=interactive, force=force\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/managers/lib.py", line 301, in install\n force=force,\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 698, in install\n pkg_dir = self._install_from_piorepo(name, requirements)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/managers/lib.py", line 151, in _install_from_piorepo\n requirements,\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 478, in _install_from_url\n dlpath = self.download(url, tmp_dir, sha1)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 182, in download\n fd.start(with_progress=with_progress)\n File "/home/object-undefined/.platformio/penv/lib/python3.5/site-packages/platformio/downloader.py", line 91, in start\n f.write(next(itercontent))\nStopIteration\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n Verify that you have the latest version of PlatformIO using\n pip install -U platformio command\n\n Try to find answer in FAQ Troubleshooting section\n https://docs.platformio.org/page/faq.html\n\n Report this problem to the developers\n https://github.com/platformio/platformio-core/issues

ivankravets, 2020-01-06:
I can't reproduce this issue. Could you try to run in VSCode terminal `platformio lib -g install "U8g2"`

See my log

```
platformio lib -g install "U8g2"
Library Storage: /Users/ikravets/.platformio/lib
Looking for U8g2 library in registry
Found: https://platformio.org/lib/show/942/U8g2
LibraryManager: Installing id=942
Downloading  [####################################]  100%
U8g2 @ 2.27.3 has been successfully installed!
```

wisegeeksrules, 2020-01-12:
I tried that method but it failed for me.

Here's the message log on my terminal:

 platformio lib --global update
/home/object-undefined/.local/lib/python2.7/site-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
Library Storage: /home/object-undefined/.platformio/lib
Updating AsyncTCP                        @ 1.1.1          [Up-to-date]
Updating ESP Async WebServer             @ 1.2.3          [Up-to-date]
Updating ESPAsyncTCP                     @ 1.2.2          [Up-to-date]
Updating LiquidCrystal_I2C               @ 1.2.4          [Up-to-date]
Updating TimerOne                        @ 1.1            [Up-to-date]
Updating U8g2                            @ 2.27.1         [2.27.3]
Uninstalling U8g2 @ 2.27.1:     [OK]
Looking for U8g2 library in registry
Found: https://platformio.org/lib/show/942/U8g2
LibraryManager: Installing id=942 @ 2.27.3
Downloading  [####--------------------------------]   11%  00:01:25
Error: Traceback (most recent call last):
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/commands/lib.py", line 253, in lib_update
    lm.update(library, only_check=only_check)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 816, in update
    self.install(name, latest, after_update=True)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/managers/lib.py", line 301, in install
    force=force,
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 698, in install
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/managers/lib.py", line 151, in _install_from_piorepo
    requirements,
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 478, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/managers/package.py", line 182, in download
    fd.start(with_progress=with_progress)
  File "/home/object-undefined/.local/lib/python2.7/site-packages/platformio/downloader.py", line 91, in start
    f.write(next(itercontent))
StopIteration

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

ivankravets, 2020-01-22:
Could you run for example?
```
platformio lib -g install TMCStepper
```

wisegeeksrules, 2020-01-26:
> Could you run for example?
> 
> ```
> platformio lib -g install TMCStepper
> ```

Sorry for the late response. Yes, that example worked well.

ivankravets, 2020-01-30:
It should work now, try:
```
platformio lib -g install 942
```

closed 2020-01-30 by ivankravets

wisegeeksrules, 2020-02-02:
> It should work now, try:
> 
> ```
> platformio lib -g install 942
> ```

Unfortunately it didn't solve the issue with the u8g2 library, it worked only for the TMCStepper library.

ivankravets, 2020-02-05:
Could you re-test with the latest PIO Core? Open PlatformIo IDe Terminal and type
`pio upgrade --dev`

wisegeeksrules, 2020-02-09:
I could not upgrade using that command, I got the following error:

$ pio upgrade --dev
/home/object-undefined/.local/lib/python2.7/site-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
Please wait while upgrading PlatformIO ...

-----------------
Permission denied
-----------------
You need the `sudo` permission to install Python packages. Try

> sudo pip install -U platformio

WARNING! Don't use `sudo` for the rest PlatformIO commands.

I'm on PIO Core v 4.1.0

afbayonac, 2020-05-31:
I have the same error in the installation of u8g2, for now update PIO no works me.