platformio/platformio-core #3333: [Feature] PlatformIO Offline devepment

fanghuaqi, 2020-01-06:
Hi genius，

      I am newbie to use platformio, but known it long long ago since you create this project a few years a log, and I found it is really a good iot development platform to cover so many great opensource embedded software platform, such as Zephyr, ARMMbed, STM32, Arduino, and also provided good GUI experiences using VSCode and other IDEs.

     And when I use it I found the following issues which make me quite confused:
     * If I want to do development totally offline, it seems that it is really hard to achieve it, even I have VSCode installed with platformio plugins, I still need internet to download SDK and toolchains,  but it didn't provided a way for me to install already downloaded packages, such as platform-zephyr, and framework-zephyr, and use existing toolchains installed via vendor provided. 
     * And when I want to modify the already installed framework-zephyr code to match my requirement, I found it is hard for me, I need to go to framework-zephyr installed directory, and modify it, but I am afraid when I upgrade the framework my changes will be swaped out, is there any to point to my local path, without copy it to your platformio managed directory?
      * For the toolchains, sometimes the network is not good enough, or I already downloaded the tools by myself, and I want to use the installed toolchains, and other tools, it seems that it is not possible now, or is it a way to achive it?
       * And now we can only use existing board supported in some framework, if we add new board supported, we need to modify the framework, and maintain our own version, it seems too hard to achive it for me in current version, or is there a simple way?
       * For the debug feature, I need to register an account to achive this, if I can't access internet during working time, it seems too hard for me to debug.

Thanks
Huaqi

fanghuaqi, 2020-01-07:
Hi, I found in this link https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages, there is an option called `platform_packages` seems to provide method to customize your own framework, toolchain, tools, but the packages need to copied to the [packages_dir](https://docs.platformio.org/en/latest/projectconf/section_platformio.html#projectconf-pio-packages-dir). 

Is there any way to not copy it to the platform managed directories? So I can develop the code at the same time without useless copy, or you can expose an option to control it? And it also require the folder must provide a `package.json` file to make sure the package is correct?

Thanks
Huaqi

ivankravets, 2020-01-16:
Hi! :)

> For the debug feature, I need to register an account to achieve this, if I can't access internet during working time, it seems too hard for me to debug.

No, you need PIO Account the only if you use [PIO Remote](https://docs.platformio.org/en/latest/plus/pio-remote.html) which depends on Internet.

> but it didn't provided a way for me to install already downloaded packages, such as platform-zephyr, and framework-zephyr, and use existing toolchains installed via vendor provided.

PlatformIO does not depend on the Internet at all. It works fully off-line the only if all dependencies are downloaded (libraries, toolchains, platforms, etc). A temporary solution is to use https://docs.platformio.org/en/latest/userguide/platforms/cmd_install.html#cmdoption-platformio-platform-install-with-package and manually specify which packages to install. The list of packages is declared in each dev-platform.

In any case, I've just filed a separate feature request https://github.com/platformio/platformio-core/issues/3345
So, `pio platform install nuclei --with-all-packages` will install ALL dependencies and PlatformIO can work in full off-line mode.

> And when I want to modify the already installed framework-zephyr code to match my requirement

We don't recommend to modify internal packages. It's the same if you install some packages using Linux package managers and then will try to modify them in your system. A package manager is responsible to maintain official packages from registry.

If you need modified version of vendor package, just clone/copy and modify it. Then use `platform_packages` option. 

Also, pleaser read this https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-package-files

> , and I want to use the installed toolchains, and other tools

You can, just download them from our CDN https://bintray.com/platformio/dl-packages and unpack to `~/.platformio/packages`.

> And now we can only use existing board supported in some framework, if we add new board supported, we need to modify the framework, and maintain our own version, it seems too hard to achive it for me in current version, or is there a simple way?

Could you provide an example? Have you seen this https://docs.platformio.org/en/latest/platforms/creating_board.html ?

> Is there any way to not copy it to the platform managed directories? So I can develop the code at the same time without useless copy, or you can expose an option to control it

Do you mean behavior similar to "symlinking"?

fanghuaqi, 2020-01-17:
Hi, @ivankravets  thanks for your detailed explainations. I think I have get most of the answers.

* Do you mean behavior similar to "symlinking"?
> In linux, I think symlinking will work, but in windows, I tried to create a link, but it doesn't work, I also tried to specify my modified package or framework using `platform_packages` option, but it will just copy my modified version of package or framework into the `PLATFORMIO_CORE_DIR`, is it possible to achieve it without copy the framework if the `platform_packages` is a folder?

* If I want to use existing toolchains installed via vendor provided, I need to manually create a `package.json` file and put it into the existing toolchain folder, is that correct?

Thanks
Huaqi

ivankravets, 2020-01-20:
You can use [platform_packages](https://docs.platformio.org/en/latest/projectconf/section_env_platform.html#platform-packages). However, the current implementation will copy all files. I opened https://github.com/platformio/platformio-core/issues/3348 for this case.

closed 2020-01-20 by ivankravets

TCB13, 2023-10-07:
> You can, just download them from our CDN https://bintray.com/platformio/dl-packages and unpack to `~/.platformio/packages`.

Dead link.

> A temporary solution is to use https://docs.platformio.org/en/latest/userguide/platforms/cmd_install.html#cmdoption-platformio-platform-install-with-package and manually specify which packages to install. The list of packages is declared in each dev-platform.

From the link I can read "This command is deprecated and will be removed in the next releases."

I believe there some confusion about the OP's request. Platform IO is great however it still and always depends on a working internet connection at some point. There are people who are required to code / build stuff on airgapped machines that can't have internet access even for setup.

It would be nice to see a **simple way to download / store dependencies and install them manually** / with a command so the local development platform doesn't require internet access.

Thank you.