platformio/platformio-core #3334: Debugging support for "SparkFun Pro Micro 5V/16MHz"

CiriousJoker, 2020-01-08:
### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`): 4.1.0

### Description of problem
Title says it all. I can't debug code on my "SparkFun Pro Micro 5V/16MHz" and the error message told me to create an issue in this repository.

I'm kind of confused by this big issue template, so I deleted most of it. I hope that's ok for you.

maxgerhardt, 2020-01-09:
AVR Debugger support is difficult for a number of reasons pointed on in the existing issue https://github.com/platformio/platform-atmelavr/issues/95 and e.g. https://community.platformio.org/t/debugger-for-arduino-uno-mega-and-esp8266/5307/8 and https://community.platformio.org/t/atmel-ice-debugger/4396/4.

ivankravets, 2020-01-22:
Duplicate of https://github.com/platformio/platform-atmelavr/issues/95

closed 2020-01-22 by ivankravets