platformio/platformio-core #3339: compiler STILL not including correct MCU / ARCH defines

philbowles, 2020-01-13:
PIO does not put the correct Arduino-compatible mcu and/or architecture defines on the command line during the build.

Any valid Arduino library that has for example #ifdef ARDUINO_ARCH_ESP32 (maybe not that specific one, but you get the picture) fails the build.  I raised this issue so long ago I can't remember when - close on 2 years probably. For that reason, I'm not going to waste any more of MY time tracking down whether its system-wide or board-specific, but it just failed again for one of my users.

It is THE single reason that a) I have not used it since then b) I argue very firmly in other support groups who claims that PIO is "Arduino-compatible" since is clearly most certainly is NOT. and c) I tell anyone who will listen to steer well clear of it until this bug is fixed and use something that actually works. 

Seriously guys, isn't it about time this got fixed? How can anybody recommend this as a serious alternative when it is russian roulette whether any 100% perfectly valid Arduino library you want to use may fail to compile?

valeros, 2020-01-22:
Hi @philbowles ! Sorry for the inconvenience. Could you please be more specific? What `mcu and/or architecture defines` are missing or what Arduino libraries fail to compile? Maybe you can put together a simple project to reproduce the issue?
Thanks!

ivankravets, 2020-01-29:
See https://github.com/espressif/arduino-esp32/blob/master/tools/platformio-build.py#L96

closed 2020-01-29 by ivankravets