platformio/platformio-core #3340: Add support for Nuclei RISC-V Processor development platform

fanghuaqi, 2020-01-15:
# Feature Request

Hi, we have developed several RISC-V processors called [Nuclei](https://translate.google.com/translate?sl=auto&tl=en&u=https://nucleisys.com/), covering from low-end to high-end use cases.

We have supported three boards:
* HummingBird FPGA evaluation board which will run Nuclei N/NX processor FPGA bitstream file
* Nuclei RV-Star Evaluation Board
* GD32VF103V Evaluation Board which this MCU on board is based on our Nuclei RISC-V Processor Core

Our support for Nuclei Platform currently is using our [Nuclei Embedded Software Development Kit](https://github.com/Nuclei-Software/nuclei-sdk), and you can find our support repo here: https://github.com/Nuclei-Software/platform-nuclei.

Currently, our software development tools are kept in github release here: https://github.com/Nuclei-Software/platform-nuclei/releases/tag/nuclei_tools_v0.0.1-alpha.

I think we still missing the following tasks need to be done:
* Add information about `nuclei` platform and boards to main documentation
* Add information about `nuclei` platform and boards to PlatformIO Web Site

And during the support, I also found that debugging in VSCode is slower than debugging in GNU MCU Eclipse with RISC-V support, even we are using the same openocd and riscv-gdb tool, and also open all the windows like registers, call stacks, peripherals, variables, etc, VSCode is much slower than Eclipse, do you have any experiences in these?

I also found that in these line of code:
https://github.com/platformio/platformio-core/blob/develop/platformio/commands/debug/server.py#L73-L75
~~~py
            args.extend(
                ["-c", "gdb_port pipe; tcl_port disabled; telnet_port disabled"]
            )
~~~
In openocd manual, it suggest to write the openocd file into log file in order to ensure that the pipe is not saturated when using higher debug level outputs, I tried to modify it like this:
~~~py
            args.extend(
                ["-c", "gdb_port pipe; log_output openocd.log; tcl_port disabled; telnet_port disabled"]
            )
~~~
Then the debug server connection speed and download speed will be faster than before, if you use load command, with previous version, the download speed will be 1 KB/s, but after my modification, it will be 7KB/s.
* Before
[image]

* Now 
[image]

Thanks
Huaqi

ivankravets, 2020-01-16:
Hi Huaqi,

Are you from Nuclei company?

ivankravets, 2020-01-16:
Thanks! We will back soon with updates for this issue.

fanghuaqi, 2020-01-16:
Hello, could you take a look at another issue post by me, we are also consider about using platformio offline, but currently I didn't find a good solution for it?

fanghuaqi, 2020-01-19:
Hi @valeros @ivankravets , if you need any information from our company or me, please let me know.

ivankravets, 2020-01-19:
@valeros please review https://github.com/Nuclei-Software/platform-nuclei

fanghuaqi, 2020-01-20:
BTW, here is our company's logo, could you also add it into https://platformio.org/ when you accept this feature?

[image]

valeros, 2020-01-20:
Hi @fanghuaqi ! I've opened a few issues in your platform repository, do you mind checking them out?

fanghuaqi, 2020-01-21:
> 
> 
> Hi @fanghuaqi ! I've opened a few issues in your platform repository, do you mind checking them out?

Thanks @valeros , I will check them out, and review it, do you mind I add you into this platform repository, I will send invite to you.

fanghuaqi, 2020-01-31:
Hi @valeros , I have done some updates in the our support, could you take a look at it.

fanghuaqi, 2020-02-10:
Hi all , any updates or what help do I need to provide? We want to tell our customer that we can use platformio ide, so they can go to your website, and find the documentation there?

valeros, 2020-02-10:
Hi @fanghuaqi I see all issues I submitted to your repository are still open, do you plan to resolve them?

fanghuaqi, 2020-02-10:
@valeros  I have pushed changes to solve them, could you take a look at it, if ok, I will close the issues, thanks.

valeros, 2020-02-10:
@fanghuaqi Great! Do you mind if I submit a PR with minor fixes?

fanghuaqi, 2020-02-10:
@valeros That will be great, thanks!

ivankravets, 2020-02-10:
@fanghuaqi please pull the latest change into the https://github.com/Nuclei-Software/platform-nuclei repository and re-test uploading+debugging. We renamed "nuclei-rv-debugger" to "rv-link" to avoid duplicate items in our documentation.

I can't test it. There is no support for macOS. If everything works, we will make the first public release and add dev-platform to our ecosystem.

closed 2020-02-10 by ivankravets

fanghuaqi, 2020-02-11:
Thank you a lot @ivankravets  @valeros, I will test it later, I have submit some issues I found it about documentation, could you take a look at it.

fanghuaqi, 2020-02-11:
If possible, could you also add our logo into this page?

[image]

ivankravets, 2020-02-11:
Done! https://platformio.org/

Thanks for using PlatformIO! Could you add PlatformIO at https://nucleisys.com/product.php (Partners list)?
The horizontal logo is attached.

[image]

fanghuaqi, 2020-02-12:
Hi @ivankravets , Thank you very much, of course, we are redesign our website, will add your logo when it is ready. 

Thanks
Huaqi

fanghuaqi, 2020-02-18:
Done with our website, now we also have english version website.

[image]

Thanks
Huaqi

ivankravets, 2020-02-18:
Thanks!
1. Is it possible to add PlatformIO IDE here https://nucleisys.com/download.php ?
2. Do you plan to add support for macOS (toolchain, openOCD)?

fanghuaqi, 2020-02-20:
Hi @ivankravets, I will check with team about these items.

* For item 1, do you mean, add the [PlatformIO IDE](https://platformio.org/platformio-ide) link to our tools section? Do you have a page for Nuclei, and user can follow it to install PlatformIO IDE?
Currently I am writing documentation for Nuclei SDK, I plan to add a link to PlatformIO IDE, if user want to work with it.

2. Currently we didn't provide support for macOS, I think of course in future we would like to do it.

Thanks
Huaqi

ivankravets, 2020-02-21:
Hi,

1) Yes, we have these links:
-  https://platformio.org/platformio-ide
- https://platformio.org/platforms/nuclei
- https://docs.platformio.org/en/latest/platforms/nuclei.html

Nevertheless, if you work on own docs it is even better. Please share a link with us later.

2) Thanks! I see a lot of requests for macOS.

commit c0aefe4c (2020-02-10): Add support for Nuclei RISC-V Processor development platform // Resolve #3340