platformio/platformio-core #3341: uint64 limit error？

0x587, 2020-01-15:
What kind of issue is this?

- [true] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [ ] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:
WIndows10 1903
**PlatformIO Version** (`platformio --version`):
4.1.0
### Description of problem
I have a time counter.
It use an uint64_t.
By the time it exceeds 10^10,it change to about 5.6*10^9.

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
> [env:esp12e]
> platform = espressif8266
> board = esp12e
> framework = arduino
> monitor_speed= 74880
> upload_speed = 921600
```

**Source file to reproduce issue:**
```cpp
It's ouput code.
>   rtcTimer.refresh();
>   uint64_t currentTime = rtcTimer.getStandTime();
>   Serial.printf("%llu: ", currentTime);
>   Serial.printf("Time in %llu min %llu s.\n",
>                 currentTime / 1000 / 1000 / 60, currentTime / 1000 / 1000 % 60);
>   delay(1000);
```

valeros, 2020-01-24:
Hi @0x587 ! Have you tried the same code in Arduino IDE?

ivankravets, 2020-01-29:
Please reopen if you still need help

closed 2020-01-29 by ivankravets

0x587, 2020-02-04:
### I'm very sorry.
**I am a Chinese, and as everyone knows, my country is facing a terrible virus. So I have not been able to reply to you before.**
The problem no longer needs to be resolved.
Thank you for your help and enthusiasm.