platformio/platformio-core #3342: MBed tests do not print to serial

ivan1993br, 2020-01-15:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**:

PlatformIO, version 4.1.0

### Description of problem

Tests does not output to serial port as expected on mbed os.

#### Steps to Reproduce

Create a embedded test with mbed os using the UNITY framework provided by platformio

### Actual Results

The test compile and runs on the board but does not output anything on serial port

### Expected Results

The test compile and runs on the board and output the results on the serial port

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
[env:disco_f407vg]
framework = mbed
platform = ststm32
board = disco_f407vg
build_flags = -DPIO_FRAMEWORK_MBED_RTOS_PRESENT -std=gnu++11
build_unflags = -std=gnu++98
test_transport = custom
```

**Source file to reproduce issue:**
```cpp
#ifdef PIO_FRAMEWORK_MBED_RTOS_PRESENT
#include "mbed.h"
#endif

#include "unity.h"

void test_function_calculator_addition(void) {
  TEST_ASSERT_EQUAL(32, 32);
}

void test_function_calculator_subtraction(void) {
  TEST_ASSERT_EQUAL(20, 20);
}

void test_function_calculator_multiplication(void) {
  TEST_ASSERT_EQUAL(50, 50);
}

void test_function_calculator_division(void) {
  TEST_ASSERT_EQUAL(32, 32);
}

int main(int argc, char **argv) {
  #ifdef PIO_FRAMEWORK_MBED_RTOS_PRESENT
  wait(5);     // for ARM mbed framework
  #endif
  UNITY_BEGIN();
  RUN_TEST(test_function_calculator_addition);
  RUN_TEST(test_function_calculator_subtraction);
  RUN_TEST(test_function_calculator_multiplication);
  RUN_TEST(test_function_calculator_division);
  UNITY_END();
}
```

### Additional info

I was able to fix this issue by creating a .mbedignore on the mbed folder for the unity framework (features/framework/unity). 

When I include the unity.h file, the one that is included is from the mbed framework when I do not ignore it, so the modifications done by platformio to the macros variables that allow me to print to the serial port are not used.

ivan1993br, 2020-02-06:
Do you have some idea on how you want to proceed to solve this issue? I mean, remove the unity from the mbed framework, or just remove when it is compiling the tests? - Do you think to add an option on the `.json` file?

ivankravets, 2022-04-09:
Duplicate of https://github.com/platformio/platformio-core/issues/3980

closed 2022-04-09 by ivankravets

pratiksurgeonslab98, 2022-09-03:
@ivan1993br 
Referring to #3342

> I was able to fix this issue by creating a .mbedignore on the mbed folder for the unity framework (features/framework/unity).
When I include the unity.h file, the one that is included is from the mbed framework when I do not ignore it, so the modifications done by platformio to the macros variables that allow me to print to the serial port are not used.

What was the content of your .mbedignore file and the directory in which it is kept?

for me  the file content of .mbedignore is
`unity/*`

the .mbedignore is kept in both these directories
/home/roy/.platformio/packages/framework-mbed/features/.mbedignore
and
/home/roy/.platformio/packages/framework-mbed/features/frameworks/unity/.mbedignore

I am unable to print the test results after running pio test.
Please help.

Thanks in advance