platformio/platformio-core #3344: Platform IO Remote Monitor

fabltd, 2020-01-16:
BUG:

Platform IO Remote: OSX 10.15.2

When trying to connect to a remote target after upload the following error occurs:

Starting Serial Monitor on mini.local:/dev/cu.usbmodemFA1301
b'dictionary changed size during iteration'

Terminal will be reused by tasks, press any key to close it.

Closing and and trying again will then work.

lackdaz, 2020-02-07:
@fabltd I ran into this when I didn't install the remote agent in a virtual environment, refer to #3286

ivankravets, 2020-02-14:
@fabltd could you provide a step by step instrution how to reproduce it? The list of comamnds could help.

fabltd, 2020-02-26:
Hi Ivan

I simply start the remote agent and then try and connect to the remote terminal.

Today I got the error again but with more detail:

Peer will receive following PB traceback:
Unhandled Error
Traceback (most recent call last):
  File "/Users/jeremyturner/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 173, in gotItem
    self.callExpressionReceived(item)
  File "/Users/jeremyturner/.platformio/packages/contrib-pysite/twisted/spread/banana.py", line 136, in callExpressionReceived
    self.expressionReceived(obj)
  File "/Users/jeremyturner/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 621, in expressionReceived
    method(*sexp[1:])
  File "/Users/jeremyturner/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1025, in proto_message
    self._recvMessage(self.localObjectForID, requestID, objectID, message, answerRequired, netArgs, netKw)
--- <exception caught here> ---
  File "/Users/jeremyturner/.platformio/packages/contrib-pysite/twisted/spread/pb.py", line 1054, in _recvMessage
    netResult = object.remoteMessageReceived(self, message, netArgs, netKw)
  File "/Users/jeremyturner/.platformio/packages/contrib-pysite/twisted/spread/flavors.py", line 125, in remoteMessageReceived
    state = method(*args, **kw)
  File "<string>", line 60, in remote_cmd
    
  File "<string>", line 78, in _process_cmd_device_monitor
    
builtins.RuntimeError: dictionary changed size during iteration

fabltd, 2020-02-26:
I am using VS code. PIO 3.1.0 with Core 4.2.1

ivankravets, 2020-03-16:
@valeros could you reproduce this issue?

valeros, 2020-03-16:
Hi @fabltd ! Could you please specify what platform/board you're using? What is your Python version?

fabltd, 2020-03-17:
Hi

ATSAMD21 - Arduino Core

Python 3.8.0

Thanks

-- 

Jeremy Turner

Google Cloud Platform Authorised Trainer

Director @fabcontrols

FAB Controls Ltd

Unit 6,St Albans Enterprise Centre

Long Spring, Porters Wood

St Albans, Hertfordshire

AL3 6EN

*Phone:* +44(0) 1727 238 894

Cloud implementation & ‘Internet of Things’ (IoT) solutions provider

ivankravets, 2020-04-19:
Should be fixed in the latest version

closed 2020-04-19 by ivankravets