platformio/platformio-core #3346: Can not build on ESP - Python

adri94, 2020-01-20:
Here is the errors I get : 

> Executing task in folder ESPEasy-mega: platformio run <

Processing custom_ESP32_4M316k (platform: None; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/command.py", line 138, in cli
    is_test_running,
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/command.py", line 170, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/platformio/commands/run/processor.py", line 71, in process
    p = PlatformFactory.newPlatform(self.options["platform"])
  File "/Users/Adri/.platformio/penv/lib/python2.7/site-packages/platformio/managers/platform.py", line 232, in newPlatform
    if isdir(name):
  File "/Users/Adri/.platformio/penv/lib/python2.7/genericpath.py", line 49, in isdir
    st = os.stat(s)
TypeError: coercing to Unicode: need string or buffer, NoneType found

Please help :)

ivankravets, 2020-01-20:
What is your locale?
Does your project path contain non-ascii chars?

adri94, 2020-01-20:
Hello,

Thanks for the answer.
I a m in France on a Mac (10.14.6)

pio settings get projects_dir
Name          Current value [Default]                    Description
------------  -----------------------------------------  ---------------------------------------------------
projects_dir  /Users/Adri/Documents/PlatformIO/Projects  Default location for PlatformIO projects (PIO Home)

ivankravets, 2020-01-20:
1. Could you try to install the latest Python 3? See insctruction => https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

2. Remove `/Users/Adri/.platformio/penv` folder and restart VSCode.

Does it work now?

adri94, 2020-01-20:
I installed python 3.8,  removed the folder and I also changed the path for python on VS settings. 

I still get the same errors with python 2.7 involved. 

What to do next ?

adri94, 2020-01-20:
No matter what I do the lib folder always comes "with python2.7"

ivankravets, 2020-01-20:
Did you follow https://docs.platformio.org/en/latest/faq.html#install-python-interpreter? Did you install SSL certificates?

If no, pelase repeat installation and remove `/Users/Adri/.platformio/penv`

adri94, 2020-01-20:
Ok. Same errors with a different python version. There is progress...

Processing custom_ESP32_4M316k (platform: None; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 128, in cli
    process_env(
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 170, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/processor.py", line 71, in process
    p = PlatformFactory.newPlatform(self.options["platform"])
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/managers/platform.py", line 232, in newPlatform
    if isdir(name):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/genericpath.py", line 42, in isdir
    st = os.stat(s)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType

ivankravets, 2020-01-20:
Ok, what you did with your PlatformIO installation? What is inside `~/.platformio/platforms`? Please remove `~/.platformio` folder and use OFFICIAL dev/platforms and do not do manual hacking.

adri94, 2020-01-22:
Thanks for your help.
I removed the folder. The IDE re-installed but still the same problem. 

Processing custom_ESP32_4M316k (platform: None; board: esp32dev; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 128, in cli
    process_env(
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/command.py", line 170, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/commands/run/processor.py", line 71, in process
    p = PlatformFactory.newPlatform(self.options["platform"])
  File "/Users/Adri/.platformio/penv/lib/python3.8/site-packages/platformio/managers/platform.py", line 232, in newPlatform
    if isdir(name):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/genericpath.py", line 42, in isdir
    st = os.stat(s)
TypeError: stat: path should be string, bytes, os.PathLike or integer, not NoneType

ivankravets, 2020-01-22:
Please provide an output from this command
```
ls -al ~/.platformio/platforms
```

adri94, 2020-01-22:
$ ls -al ~/.platformio/platforms
total 0
drwxr-xr-x+ 2 Adri  staff   64 22 jan 14:04 .
drwxr-xr-x+ 9 Adri  staff  288 22 jan 14:04 ..

ivankravets, 2020-01-22:
```
Processing custom_ESP32_4M316k (platform: None; board: esp32dev; framework: arduino)
```

> platform: None

PLEASE FIX platformio.ini

ivankravets, 2020-01-22:
Could you share here your `platformio.ini`?

adri94, 2020-01-22:
I wanted to change the default board from esp8266 to esp32. I did it the wrong way. It works with the default board (esp8266). 
I can't figure out where to choose the board. Is it in platformio_core_defs.ini ?
Thanks for helping a newbie.

ivankravets, 2020-01-22:
You need to set `platform = espressif32`

adri94, 2020-01-23:
I am struggling. Where do I set platform to espressif32 ?

ivankravets, 2020-01-23:
https://github.com/platformio/platform-espressif32/blob/develop/examples/arduino-blink/platformio.ini

closed 2020-01-23 by ivankravets