platformio/platformio-core #3347: Support custom boards for existing platform

binloan, 2020-01-20:
What kind of issue is this?

- [ ] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

- [ ] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Description of problem

Today only custom platforms are supported. Creating a custom board for an existing platform is breaking the source.

It should be possible to define your own custom board reusing the existing platform without breaking the source. E.g. a new board definition for the nrf52832 that maybe doesn't have a button or led, etc.

ivankravets, 2020-02-14:
You can create this. Just add to your customer board JSON manifest
```
{
  "platform": "nordicnrf52"
}
```

or, if you custom board is compatible with more dev-platforms, use `platforms: ["platformname1", "platformname2"]`

closed 2020-02-14 by ivankravets

binloan, 2020-02-19:
Thanks. Documentation about this was not clear to me.