platformio/platformio-core #3349: Serial Port Auto-Detection Issues

thomastech, 2020-01-21:
### Configuration
**Operating system**:
 WIN10 Pro/64

**PlatformIO Version**
 Platformio IDE 1.10.0 installed on VSCode 
 Platform is espressif32 V1.11.1
 lolin_d32_pro board
 USB is assigned to COM4

```
PIO Home=>Devices:
 COM1 	Communications Port (COM1)	ACPI\PNP0501\0
 COM2 	High-Speed USB Serial Port (COM2)	QUADPORT\QUAD_SERIAL_INTERFACE\7&29C3DB4D&0&0000
 COM3 	High-Speed USB Serial Port (COM3)	QUADPORT\QUAD_SERIAL_INTERFACE\7&29C3DB4D&0&0001
 COM4 	USB-SERIAL CH340 (COM4)	USB VID:PID=1A86:7523 SER=5 LOCATION=1-9
 COM5 	USB Serial Port (COM5)	USB VID:PID=0403:6001 SER=PXF3PKCPA
```

### Description of problem
Serial port Auto detection chooses the wrong USB COMx port. Success can occur too, but it depends on the USB serial devices that are currently installed. 

I originally thought that only the Monitor port was affected. But after some experimentation (noted further below) I found that Upload can also choose the wrong COMx port.

Initially I found that the Upload Port auto-detect was working correctly (ESP32 Upload flash worked normally). But Serial Monitor would not auto-detect. Instead it lists the available ports and waits for me to enter the COMx port name. As expected, typing "COM4" (or "4") launches the serial monitor. 

This is the list I see in the terminal window when Serial Monitor is launched from the IDE:
```
--- Available ports:
---  1: COM1                 'Communications Port (COM1)'
---  2: COM2                 'High-Speed USB Serial Port (COM2)'
---  3: COM3                 'High-Speed USB Serial Port (COM3)'
---  4: COM4                 'USB-SERIAL CH340 (COM4)'
---  5: COM5                 'USB Serial Port (COM5)'
--- Enter port index or full name:
```

#### Steps to Reproduce

**No. 1**
I tried specifying the COM port with a wildcard in platformio.ini. Like this: _monitor_port = COM*_

The good news is that this automatically launches the monitor terminal after the flash Upload, but it chooses the wrong port for the serial monitor. For example, the ESP32 board is on COM4, but the monitor_port wildcard is picking COM5.

**No. 2**
I found that specifying the monitor_port COM4 value in platformio.ini works correctly. Like this: _monitor_port = COM4_
After a flash Upload this automatically opens the serial monitor for my serial debugging. But specifying a fixed COMx is problematic since port usage can vary with the other project members.

**No. 3**
I removed the COM5 USB adapter and everything works perfectly; Both upload and serial monitor are correctly auto-detected. I wondered if it could it be a bad USB device.

So I installed a randomly chosen FT232RL USB adapter. It was assigned COM10. I rebooted the PC and got back into the Platformio IDE. As before, when I launched the Serial Monitor it did not auto-detect. The available COM ports list appeared, like this:

--- Available ports:
---  1: COM1                 'Communications Port (COM1)'
---  2: COM2                 'High-Speed USB Serial Port (COM2)'
---  3: COM3                 'High-Speed USB Serial Port (COM3)'
---  4: COM4                 'USB-SERIAL CH340 (COM4)'
---  5: COM10                'USB Serial Port (COM10)'
--- Enter port index or full name:

Choosing COM4 enables the ESP32’s serial monitor. So at this point the behavior follows what I saw when the COM5 device was present. 

But now when I do a flash upload it auto-detects COM10! I removed the COM10 USB device and flash Upload correctly auto-detects COM4.

**No. 4**
The Lolin D32 Pro uses a CH340 that has VID=1A86 and PID=7523. As an experiment, I edited the build section of the d32_pro board.json file and added the USB ID's as follows:
```
    "hwids": [
      [
        "0x1A86",
        "0x7523"
      ]
    ],
    "ldscript": "esp32_out.ld",
```
This did not change anything, all observed symptoms remained the same.

### Expected Results
The intended action is that Upload and Serial Monitor should auto-detect the COMx and not require any user intervention or user specific edits to platformio.ini.

ivankravets, 2020-01-22:
I don't see any problems with PlatformIO Core. Please provide an output from 
```
pio device list
```

thomastech, 2020-01-22:
@ivankravets:  Thanks for the helping hand. 

My ESP32 is on COM4. Here is the device list, including the COM5 and COM10 USB serial devices used in the testing scenarios described earlier.

```
COM1 | Communications Port (COM1) | ACPI\PNP0501\0
COM10 | USB Serial Port (COM10) | USB VID:PID=0403:6001 SER=A50285BIA
COM2 | High-Speed USB Serial Port (COM2) | QUADPORT\QUAD_SERIAL_INTERFACE\7&29C3DB4D&0&0000
COM3 | High-Speed USB Serial Port (COM3) | QUADPORT\QUAD_SERIAL_INTERFACE\7&29C3DB4D&0&0001
COM4 | USB-SERIAL CH340 (COM4) | USB VID:PID=1A86:7523 SER=5 LOCATION=1-9
COM5 | USB Serial Port (COM5) | USB VID:PID=0403:6001 SER=PXF3PKCPA
```
**Partial good news:**
I found the reason for the failure with upload port auto-detection. That is to say, upload port would incorrectly choose COM10 even when I used the build.hwids script that explicitly called out the required VIDS & PIDS.

The upload port bug is in the _pioupload.py_ file at the __look_for_serial_port()_ function. Instead of using the hwids string array vars, it incorrectly calls out the char array vars. So build.hwid port match detection always fails. 

Here is the code snippet that shows the corrected function. The old code line has been commented out so you can see the code revision more distinctly. 

See File:   .platformio\penv\Lib\site-packages\platformio\builder\tools\pioupload.py
```
            for hwid in board_hwids:
                #hwid_str = ("%s:%s" % (hwid[0], hwid[1])).replace("0x", "")
                hwid_str = ("%s:%s" % (board_hwids[0], board_hwids[1])).replace("0x", "")
                if hwid_str in item["hwid"]:
                    print(env.subst("Found build.hwids match at %s" % (port)))
                    return port
```
This patch also includes a user message that reports when a build.hwids port match has occurred, which is a useful confirmation. 

Although this patch fixes upload port, it doesn't help solve the monitor port's auto-detection issue.  It would be great to get some help with that.

- Thomas

ivankravets, 2020-01-22:
https://github.com/platformio/platformio-core/issues/3349#issuecomment-577342300

thomastech, 2020-01-22:
> #3349 (comment)
```
PS C:\users\tom\.platformio\penv\Scripts> pio device list
COM5
----
Hardware ID: USB VID:PID=0403:6001 SER=PXF3PKCPA
Description: USB Serial Port (COM5)

COM1
----
Hardware ID: ACPI\PNP0501\0
Description: Communications Port (COM1)

COM4
----
Hardware ID: USB VID:PID=1A86:7523 SER=5 LOCATION=1-9
Description: USB-SERIAL CH340 (COM4)

COM10
-----
Hardware ID: USB VID:PID=0403:6001 SER=A50285BIA
Description: USB Serial Port (COM10)

COM2
----
Hardware ID: QUADPORT\QUAD_SERIAL_INTERFACE\7&29C3DB4D&0&0000
Description: High-Speed USB Serial Port (COM2)

COM3
----
Hardware ID: QUADPORT\QUAD_SERIAL_INTERFACE\7&29C3DB4D&0&0001
Description: High-Speed USB Serial Port (COM3)
````

ivankravets, 2020-01-23:
PlatformIO depends on VID:PID which declared in board manifest. You board has generic PID/VID. Million of other boards use the same UART chip. In this case, we recommend specifing uploader port manually.

thomastech, 2020-01-23:
@ivankravets. Thanks for the help. I dug into the platformio code this morning and saw that the "VID:PID" string matching was the method used to choose the monitor port.  I learned that if only one of these type of COM ports exists then it is chosen, else the user has to manually choose it. That is exactly what I am seeing, so monitor port it is working as designed. 

I was under the impression that specifying the build.hwids would be used in all COMx port selection. But now I know only applies to the Upload port.

But as I reported earlier, the hwids matching is broken in pioupload.py.  As-is, the build.hwids directives are ignored due to incorrect vars (hwid[ ] should be board_hwids[ ]).

Please see this revision:
See File: .platformio\penv\Lib\site-packages\platformio\builder\tools\pioupload.py
```
            for hwid in board_hwids:
                #hwid_str = ("%s:%s" % (hwid[0], hwid[1])).replace("0x", "")
                hwid_str = ("%s:%s" % (board_hwids[0], board_hwids[1])).replace("0x", "")
                if hwid_str in item["hwid"]:
                    print(env.subst("Found build.hwids match at %s" % (port)))
                    return port
```
- Thomas

ivankravets, 2020-01-23:
```
board_config.update("build.hwids", [
   "0x1A86", #VIDS
   "0x7523" #PIDS
])
```

This should be array of array

Fixed code

```
board_config.update("build.hwids", [
   ["0x1A86", #VIDS
   "0x7523" #PIDS
]])
```

thomastech, 2020-01-23:
@ivankravets, thank you! That solves the problem. My array's construction was taken from a example I found on the web.  So it seems I got what I paid for. :)

BTW, my proposed idea for monitor port "auto-detect" would be a new platformio.ini directive like this:
  monitor_port = upload_port
or
  monitor_port = build.hwids

- Thomas

ivankravets, 2020-01-24:
See docs https://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#override-board-configuration

thomastech, 2020-01-24:
Thanks. 

From what I saw in the core code the hwids's manifest override is for the Upload, Debug, and Test ports. So I think it would be a good idea to add a note to the Override Board Configuration text that states it does not apply to the Monitor port. 

- Thomas

ivankravets, 2020-01-24:
Thanks! I renamed this issue and reopened. We will implement in 4.1.1.

thomastech, 2020-01-24:
Nice. Adding Monitor port to the hwids override is fantastic. Looking forward to V4.1.1.

- Thomas

closed 2020-02-05 by ivankravets

thomastech, 2020-02-07:
@ivankravets : Thanks for adding the monitor port auto-detect, I appreciate it.

If you desire any user feedback then I'm available to test the new feature. To do it I'll need advice on how to temporarily install the platformio-core's develop version (will revert back after the validation test). I'm running VS Code Platformio IDE on a WIN10/64 machine. 

- Thomas

ivankravets, 2020-02-08:
All kudos goes to @valeros, he resolved this issue 😊 

Thanks for the willingness to help with testing. You can easily switch to PlatformIO Core dev-version running the next command:
```
pio upgrade --dev
```

thomastech, 2020-02-08:
> All kudos goes to @valeros, he resolved this issue
Thanks @valeros!

I've switched to the dev version and tested the new feature. Unfortunately Serial Monitor port detection is still not working for me (same behavior as before).

I've rebooted and re-verified the dev version using terminal:
`> pio --version
PlatformIO, version 4.2.0rc1
`

ivankravets, 2020-02-08:
I see the problem and I'm not sure that it could be fixed. The extra scripting is a part of a build process. The device monitor works in another way. You have 2 options to resolve this issue without extra script (yes, you can remove it):

1. Edit board JSON directly in `~/.platformio/platforms/espressif23/boards/your_board.json`. See example https://github.com/platformio/platform-atmelavr/blob/develop/boards/uno.json#L6
2. Clone this `your_board.json` to `project-dir/boards/your_board.json` and edit here. Please do not forget to add `"platform": "espressif32",` to new JSON to instruct PlatformIO to override  only for ESP32 dev-platform.

thomastech, 2020-02-08:
Oh, that's not the outcome I was hoping for. 

Early on I tried Choice #1. But abandoned it because it requires custom edits outside the project folder. The goal is to keep it all in the project.

Choice #2 seems reasonable since it allows the workaround to be included in the project folder. I'll be able to try it out in a couple days and will report back on my success. 

Thanks again for your advice.

- Thomas

thomastech, 2020-02-08:
@ivankravets, Sorry for being a persistent PITA. 
I implemented choice # 2. Unfortunately the Serial Monitor still requires me to manually choose the serial port.  

I confirmed that the customize /board json file is being used by observing the log messages seen during compile.

Here's the related files:
**platformio.ini:**
```
; PlatformIO Project Configuration File
; Revised Feb-08-2020 by thomastech
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
data_dir    = ./data

;[env:lolin_d32_pro]
[env:lolin_d32_pro_sparky]
platform = espressif32@>=1.11.1
framework = arduino
;board = lolin_d32_pro
board = lolin_d32_pro_sparky
monitor_speed = 115200
;monitor_port = COM[4]
monitor_rts = 0
monitor_dtr = 0
upload_speed = 921600
board_build.f_cpu = 240000000L ;See https://github.com/espressif/arduino-esp32/issues/487
board_upload.flash_size = 16MB
board_upload.maximum_ram_size = 4096000
board_upload.maximum_size = 6553600
build_flags =
    -DCORE_DEBUG_LEVEL=0
board_build.embed_files =
    src/test1.img
    src/test2.bin
    src/wav/beep.wav
    src/wav/bleep.wav
    src/wav/blip.wav
    src/wav/bloop.wav
    src/wav/currentOn.wav
    src/wav/decreaseMsg.wav
    src/wav/ding.wav
    src/wav/increaseMsg.wav
    src/wav/overheat.wav
    src/wav/promo.wav
    src/wav/silence100ms.wav
    src/wav/0000.wav
    src/wav/0001.wav
    src/wav/0002.wav
    src/wav/0003.wav
    src/wav/0004.wav
    src/wav/0005.wav
    src/wav/0006.wav
    src/wav/0007.wav
    src/wav/0008.wav
    src/wav/0009.wav
    src/wav/0010.wav
board_build.partitions = default_16MB.csv
lib_deps =
	Adafruit GFX Library@=1.7.3
	Adafruit ILI9341@=1.5.3
	XPT2046_Touchscreen@26b691b2c8
;extra_scripts = pre:custom_hwids.py
```

**/boards/lolin_d32_pro_sparky.json:**
```
{
  "platforms": "espressif32",
  "build": {
    "core": "esp32",
    "extra_flags": "-DARDUINO_LOLIN_D32_PRO",
    "f_cpu": "240000000L",
    "f_flash": "40000000L",
    "flash_mode": "dio",
    "hwids": [
      [
        "0x1A86",
        "0x7523"
      ]
    ],
    "ldscript": "esp32_out.ld",
    "mcu": "esp32",
    "variant": "d32_pro"
  },
  "connectivity": [
    "wifi",
    "bluetooth",
    "ethernet",
    "can"
  ],
  "debug": {
    "openocd_board": "esp-wroom-32.cfg"
  },
  "frameworks": [
    "arduino",
    "espidf"
  ],
  "name": "WEMOS LOLIN D32 PRO Sparky",
  "upload": {
    "flash_size": "4MB",
    "maximum_ram_size": 327680,
    "maximum_size": 4194304,
    "require_upload_port": true,
    "speed": 460800
  },
  "url": "https://wiki.wemos.cc/products:d32:d32_pro",
  "vendor": "WEMOS"
}

```
I am uncertain about the board's json _"platform":_ syntax. I originally tried your example that was given as:
"platform": "espressif32"
But after some research I it seems that https://docs.platformio.org/en/latest/librarymanager/config.html#platforms suggests it should be:
"platforms": "espressif32"
But no Serial Monitor port success with either syntax. 

Do you see any issues with my configuration?

- Thomas

ivankravets, 2020-02-12:
Thanks! Docs is updated https://github.com/platformio/platformio-core/commit/b87048020d8d1a8f1bf83c41d551eb5cc4bedb9a

thomastech, 2020-02-12:
@ivankravets, thank you for all the hands-on advice.
@valeros, thank you for coding the new feature. 

Looking forward to the official release (4.1.1). In the meantime I'll continue to use the Dev release.

- Thomas

pfeerick, 2020-02-13:
@thomastech You don't have to wait long... it's already been released! ;) https://community.platformio.org/t/platformio-core-4-2/11980

thomastech, 2020-02-13:
Thanks for the heads-up!

- Thomas

piersh-aetheros, 2021-12-12:
I see the same problem. pressing the 'Upload' button runs `platformio.exe run --target upload --target monitor` and after the upload completes it starts the monitor immediately, the OS hasn't re-enumerated the device yet and it fails to find the serial port. 

if i then close the terminal window and manually run the monitor, then it works fine.

commit 8fce660a (2020-01-24): Add notice about VID:PID pairs // Resolve #3349
commit 7716fe1d (2020-02-05): Autodetect monitor port for boards with specified HWIDs // Resolves #3349 (#3369)

* Autodetect serial port for boards with specified hwids

* Add new record to history log