platformio/platformio-core #3354: cannot open source file "Arduino.h"

sekt1953, 2020-01-22:
What kind of issue is this?

- [ ] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to appropriate repository:
      [PlatformIO IDE for Atom](https://github.com/platformio/platformio-atom-ide/issues) or
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues)

I get this error after running today's update 22/1/2020
#include errors detected. Please update your includePath. Squiggles are disabled for this translation unit (/home/sekt/Documents/PlatformIO/Projects/ESP32_UltraSonic/src/main.cpp).
cannot open source file "Arduino.h"
cannot open source file "Wire.h"
cannot open source file "Adafruit_GFX.h"
cannot open source file "Adafruit_SSD1306.h"
cannot open source file "Sonar.h"

the code had no problim befor running platformio update today 22/1/2020

ivankravets, 2020-01-22:
Please remove `.vscode` folder from a project and restart VSCode.

sekt1953, 2020-01-22:
it do not help to remove .vscode
I am running ubuntu 18.04, VSC & PlatformIO, programming,  ESP32 with
Arduino framework.
 it failed on mine and  my students' machines the moment vi did the
update ve all try to remove .vscode it did not help

SEK. Thomsen