platformio/platformio-core #3365: Platform IO will not build (i keep getting errors of all sort)

mal136, 2020-02-04:
Im on Mac Os Catalina 10.15.2 with slow internet connection 

Running Platform IO extension on Vscode 

I tried updgrading PlatformIO core and get this error everything no matter(yes i am connected to internet)
Error: Can not retrieve the latest PlatformIO version

My main complaint is not being able to build and upload marlin 2.0 for my 3d printer, i figured i have to update everything but i can not get it to update . please help

i tried updating python via the terminal but i cant get it to update, im on python version 2.7.16, but i download python 3.8(latest one) and i dont understand why it doesnt say 3.8 when i type in "python --version" in the terminal?

I tried the "pip update python" and terminal says "pip" not understood or something like that and won't update

Here the errors i get when i try to build(minus my name and info, which is replaced with " ***** ")

Error: Please read http://bit.ly/package-manager-ioerror
Error: Traceback (most recent call last):
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/commands/run/processor.py", line 71, in process
    p = PlatformFactory.newPlatform(self.options["platform"])
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/managers/platform.py", line 246, in newPlatform
    name if not requirements else "%s@%s" % (name, requirements)
platformio.exception.UnknownPlatform: Unknown development platform '0.1.2'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 376, in _make_request
    self._validate_conn(conn)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 994, in _validate_conn
    conn.connect()
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/connection.py", line 360, in connect
    ssl_context=context,
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/util/ssl_.py", line 370, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 376, in wrap_socket
    _context=self)
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 747, in __init__
    self.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 983, in do_handshake
    self._sslobj.do_handshake()
  File "/Library/Frameworks/Python.framework/Versions/3.5/lib/python3.5/ssl.py", line 628, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/connectionpool.py", line 720, in urlopen
    method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/urllib3/util/retry.py", line 436, in increment
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip (Caused by SSLError(SSLError(1, '[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645)'),))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/commands/run/command.py", line 138, in cli
    is_test_running,
  File "/Users/*****.platformio/penv/lib/python3.5/site-packages/platformio/commands/run/command.py", line 170, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/commands/run/processor.py", line 76, in process
    skip_default_package=True,
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/commands/platform.py", line 317, in platform_install
    force=force,
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/managers/platform.py", line 78, in install
    self, name, requirements, silent=silent, force=force
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 696, in install
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 478, in _install_from_url
    dlpath = self.download(url, tmp_dir, sha1)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 197, in download
    raise e
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/managers/package.py", line 181, in download
    fd = FileDownloader(url, dest_dir)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/platformio/downloader.py", line 44, in __init__
    verify=version_info >= (2, 7, 9),
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/requests/sessions.py", line 533, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/Users/*****/.platformio/penv/lib/python3.5/site-packages/requests/adapters.py", line 514, in send
    raise SSLError(e, request=request)
requests.exceptions.SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /p3p/pio-nxplpc-arduino-lpc176x/archive/0.1.2.zip (Caused by SSLError(SSLError(1, '[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:645)'),))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

mal136, 2020-02-04:
I'm sorry i'm not a computer science major, i know how to program and mess with Arduino code to make practically anything(robots, drones etc) but i do not understand software/terminal stuff that well, i would appreciate any links and good sources for learning this information so i can be self-sufficient and learn. Thank you in advanced!

ivankravets, 2020-02-04:
You have a problem with outdated Python 3 in your system. Please install the latest Python 3.8 and install SSL certificates. See guide https://docs.platformio.org/en/latest/faq.html#install-python-interpreter

After this step, please remove `~/.platformio/penv` folder and restart VSCode.

closed 2020-02-04 by ivankravets

mal136, 2020-02-04:
I did install the latest python 3.8 & installed the SSL certificates via the link   , however i get a “warning: you are using pip version 19.2.3, however 20.0.2 is avaiable. You Should consider upgrading via the ‘pip install —upgrade pip’ command” 

And when i type it into terminal , and hit enter, i get “zsh: command not found: pip “ 

So im not sure how to proceed... and for some reason after i installed python 3.8, terminal still tells me i have python version 2.7.16... do i have to uninstall 2.7.16 in a special way , becauze i already moved it to trash and deleted python 2.7.16

ivankravets, 2020-02-04:
`python3 --version`.
Run in terminal `rm rf ~/.platformio/penv`

mal136, 2020-02-04:
I just ran that in terminal and it says”rm: rf: no such file or directory”

I also just ran it in VScode and it said python was 3.6.4 , but my terminal says 3.8.1 

Is there a manual way to do this? Instead of using terminal? Should i uninstall vscode and python and re-download everything?

ivankravets, 2020-02-04:
```
rm -rf ~/.platformio/penv
```

mal136, 2020-02-04:
That seemed to do something, i typed it into terminal & now my python - -version is 2.7.16 .... so i tried re-installing python 3.8 & it still says 2.7.16 & it gave me the warning again

“warning: you are using pip version 19.2.3, however 20.0.2 is avaiable. You Should consider upgrading via the ‘pip install —upgrade pip’ command”

And when i type it into terminal , and hit enter, i get “zsh: command not found: pip “
  
Thank you for your help, but nothing seems to work. It’s so weird, i dont understand why none of the methods are working. I tried most of what you said already but nothing happens. Still doesnt work