platformio/platformio-core #3366: Unable to do a library simple search

cailloumajor, 2020-02-04:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

As stated in https://docs.platformio.org/en/latest/userguide/lib/cmd_search.html#examples, the command `platformio lib search "1-wire"` should return some results. For now, the CLI returns:
> Nothing has been found by your request
> Try a less-specific search or use truncation (or wildcard) operator *
> For example: DS*, PCA*, DHT* and etc.
>
> For more examples and advanced search syntax, please use documentation:
> https://docs.platformio.org/page/userguide/lib/cmd_search.html

Searching `1-wire` in PlatformIO home (Visual Studio Code) returns `No Results`.

ivankravets, 2020-02-05:
Please re-run `pio update` command and try search again. It was known issue and we fixed quickly.

closed 2020-02-05 by ivankravets

cailloumajor, 2020-02-05:
OK now following your advice, thanks ! :+1: