platformio/platformio-core #3367: Build error when parameter is overrided in mbed_app.json (for mbed libraries into project/lib)

edgrin, 2020-02-05:
What kind of issue is this?

- [ ] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.
------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`): version 4.1.1b9

### Description of problem

Currently it is possible to build mbed libraries in project/lib directory as PlatformIO can parse mbed_lib.json for each library and add macroses into mbed_config.h. The issue appears when you try to override parameters of those libraries through mbed_app.json. Build fails with error "mbed build API internal error. Attempt to override undefined parameter". It looks like currently it is only possible to override parameters of libraries which come with mbed by default.

#### Steps to Reproduce

1. clone https://github.com/edgrin/pio-mbed-lib-target-overrides
2. try to build
3. see error Attempt to override undefined parameter 'newlib.param' in 'application[*]'
4. delete line 4 with '"newlib.param": 2' in mbed_app.json
5. try to build
6. see successful build with newlib in Dependency Graph

### Actual Results

mbed build API internal error

### Expected Results

Successful build

valeros, 2020-02-05:
Hi @edgrin! Thanks for the report. Indeed, there is a limitation which relates to configuration of additional libraries from `mbed_app.json` file. The reason is that our build logic is slightly differs from the build logic implemented in mbed tools. In a nutshell, we process libraries after processing the main parts of the framework. Since this configs eventually end up as defines in `mbed_config.h` file, you can try to create [an extra script](http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#construction-environments) for your library and add these defines to the global environment. 
Sorry for the inconvenience.

edgrin, 2020-02-06:
Hi @valeros, thanks for the quick reply! Yeah, I noticed that mbed framework complains before platformio scans for library dependencies. I guess there is no easy solution to fix this issue and I would be happy with some workaround like you offered. I tried it and unfortunately it doesn't work.

You can check the repo pio-mbed-lib-target-overrides I mentioned in the issue description. Now it has pre script to override MBED_CONF_NEWLIB_PARAM and main.c has #pragma to print the value of this define at compilation.

When main.c is compiled you will see the following:
In file included from <command-line>:0:0:
.pio/build/debug/mbed_config.h:70:0: warning: "MBED_CONF_NEWLIB_PARAM" redefined
 #define MBED_CONF_NEWLIB_PARAM 1
<command-line>:0:0: note: this is the location of the previous definition
src/main.cpp:7:55: note: #pragma message: MBED_CONF_NEWLIB_PARAM = 1
 #pragma message(VAR_NAME_VALUE(MBED_CONF_NEWLIB_PARAM))

The reason of this behaviour is because prescript defines new value via command line and it is redefined when main.c includes initial value from mbed_config.h via mbed.h

Is there a way to modify mbed_config.h and insert at the end:
#undef MBED_CONF_NEWLIB_PARAM
#define MBED_CONF_NEWLIB_PARAM 2

edgrin, 2020-02-06:
@valeros , I ended up with post script which makes replacement for desired defines in mbed_config.h. You can take a look in the [third commit](https://github.com/edgrin/pio-mbed-lib-target-overrides/tree/ff0bf6059bc5b4356a4acc2a9cd19995105625df).

After that I couldn't stop and implemented solution I'm going to use in my current project. It is the [4th commit](https://github.com/edgrin/pio-mbed-lib-target-overrides/tree/30fb1639a74a7f0a181e357d03be086066500aee). Now it is possible for me with using one script to override params for local libs, and those overrides could be different for different environments.

So, now I can wait when you guys can fix this issue :) Btw, thanks a lot for Platform IO, it is awesome!

edgrin, 2020-02-06:
I started to think that grouping overrides by environments maybe is not good idea, because in mbed_app.json overrides are grouped by targets, i.e. boards. Most likely I will update current solution end replace environments with boards.

edgrin, 2020-02-07:
Now script uses board id instead of environment id to separate different overrides. Also the script is simplified and instead of replacing defines with new values, it is just add the content of all found header files in the end of mbed_config.h. All inserted defines come with their #undef first. Maybe this temporary solution would be useful for someone else who also come across this issue.

horeich, 2020-10-09:
I love PlatformIO but this is a real problem for us having large external libraries. Would be great if you could be able to replace lib_json in target_overrides in mbed_app.json.

@edgrin Thanks for adressing this issue, your script wouldn't work for me, unfortunately.