platformio/platformio-core #3368: Project Inspector Crashes when Flash Use > 100%

shiyouganai, 2020-02-05:
Versions:
Home 3.1.0-rc.2
Core 4.1.1b9

Platform: atmel avr 2.0.0
Environment: arduino
Board: arduboy

Output:
Errors
PIO Core Call Error: "Processing arduboy (platform: atmelavr; board: arduboy; framework: arduino)\r\n--------------------------------------------------------------------------------\r\n\n\n========================== [FAILED] Took 4.17 seconds =========================="

Steps to recreate:
Create a project for a board, set the board's upload "maximum_size" to less than the size of the firmware compiled without any optimizations (Debug Mode). Press the inspect button in PIO Home with this project selected and "Inspect Memory" selected. The inspection will fail. Set the board's upload "maximum_size" to a very large value, and run the inspection again. The inspection will pass.

I do not know if this is intended behavior, but in my specific use case, my project is under the max flash size with compiler optimizations on (Release Mode), but with optimizations off, it's size increases by approx 30% and goes way over the flash size, causing Project Inspector to no longer work. Especially since the goal of project inspector is to inspect the use of flash for high use symbols, I think that >100% flash use should be a valid inspection case.

Thanks for your work on this great software!

ivankravets, 2020-02-05:
Please open VSCode Terminal and type
```
pio upgrade --dev
```

Restart VSCode. Does it work?

closed 2020-02-05 by ivankravets

shiyouganai, 2020-02-07:
It works. Thank you very much for the quick fix!

JonnyHaystack, 2023-10-21:
Has there possibly been a regression here? I seem to be having this same problem with the inspector where it fails due to exceeding 100% flash usage, and increasing `board_upload.maximum_size` gets it to pass for me as well.

commit ee2e4896 (2020-02-05): Fixed an issue when Project Inspector crashes when flash use > 100% // Resolve #3368