platformio/platformio-core #3375: Include wrong library

meditant, 2020-02-10:
Hello,
Since the last update to Home 3.1.0 RC4 and Core 4.2.0RC2 i have a compilation error on one project is load the good library but an other wrong library (lib319).

Compiling .pio/build/debug/lib60f/AsyncTCP_ID1826/AsyncTCP.cpp.o
Compiling .pio/build/debug/lib319/ESPAsyncTCP_ID305/AsyncPrinter.cpp.o

My PIO ini :

platform = espressif32
board = lolin32
lib_deps =
  AsyncTCP
  AsyncMqttClient
  ESP Async WebServer
  ArduinoJson
framework = arduino

Mac OS 10.15.3

Best

meditant, 2020-02-10:
I deleted the library in libdeps in .pio and now it's work but i nevers add this 8266 library...

Strange !

ivankravets, 2020-02-10:
Which library is incompatible?

closed 2020-02-12 by ivankravets